thesofproject/sof #2217: [BUG] kpb_prepare() error: wrong host params

wwittbrx, 2019-12-17:
**Describe the bug**
KD pipeline is unable to start after merging https://github.com/thesofproject/sof/pull/2038 on master due to "kpb_prepare() error: wrong host params".

**To Reproduce**
arecord -M -N -D hw:0,8 -f s24_le -c 2 -r 16000 -d 10 record_kd.wav

**Reproduction Rate**
100%

**Expected behavior**
KD pipeline starts normally.

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: c985ebd27803f695626c38ea5ef18821be142f85
    * SOF: 2796310f89a59e6c3c7c5e68f639ebe33dc398cb
2) Name of the topology file
    * Topology: sof-cml-rt5682-max98357a.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML Hatch with UbuntuOS

**Screenshots or console output**
[sof-logger-error.log](https://github.com/thesofproject/sof/files/3972437/sof-logger-error.log)

wwittbrx, 2019-12-17:
@lgirdwood Why was this pull request merged without passing CI? https://sof-ci.01.org/sof-pr-viewer/#/build/PR2038/build4547433 clearly shows problems with KD tests.

mrajwa, 2019-12-17:
This happens because test cases are outdated. More than a month ago, having feedback from customers about their KWD usage + knowing all technical aspects we decided to reject host buffers < kd buffer. Once you provide host buffer sufficiently big enough the problem is gone.

lgirdwood, 2019-12-18:
@wwittbrx that was merged in error as CI SNR was poor with SCHEDULER and some other failures.
@mrajwa does this mean we need new KD CI tests now ? Where do we document the KWD min/max buffer sizes ?

wwittbrx, 2019-12-18:
For this PR we will be running tests manually as there are some other issues with it. Once PR is ready to merge we will need to coordinate so that test repository update and merge happen at roughly the same time to avoid impact on other PRs.

mrajwa, 2019-12-18:
> @mrajwa does this mean we need new KD CI tests now ? Where do we document the KWD min/max buffer sizes ?

@lgirdwood, I am afraid we don't officially document it yet. We have agreed on it on mailing list already though. Yes, we need to slightly change test cases to use different buffer sizes. The other issue @wwittbrx mentioned is because there is one more PR which needs to be merged but I would really like this one to be merged first.
Okay, let's change test cases and I will submit this PR again.

lgirdwood, 2019-12-18:
@mrajwa ok, please resubmit. @wwittbrx pls review and approve if aligned, I will wait until we see your approval prior to merge.

closed 2020-01-21 by wwittbrx