thesofproject/sof #2219: [BUG]Unexpected sof error trace print when trigger aplay/arecord

Liviali155, 2019-12-17:
**Describe the bug**
Unexpected sof error trace print when trigger aplay/arecord, but aplay/arecord can work normally, just an error message output

**To Reproduce**
1.sudo reboot
2.aplay -Dhw:0,1 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i -d 1 
3.sudo sof-logger -l /etc/sof/sof-*.ldc

**Reproduction Rate**
10/10

**Expected behavior**
There is no sof error trace print when aplay/arecord

**Actual result**
There is sof error trace output when aplay/arecord,but function can work normally

```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA          76527513.697917  76527512.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 40493
```

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:c5e4a391
Firmware:https://github.com/thesofproject/sof/commits/master commit: 85e462ad
Can reproduce on all below platforms: TGL RT711 in HDA mode, ICL rt711 in HDA mode, APL-UP2 nocodec, APL-up2 PCM512x in I2S mode, BYT nocode, BYT 5682 in I2S mode.

[dmesg20191217.log](https://github.com/thesofproject/sof/files/3972776/dmesg20191217.log)
[sof_errortrace1217.log](https://github.com/thesofproject/sof/files/3972777/sof_errortrace1217.log)
[sof_trace1217.log](https://github.com/thesofproject/sof/files/3972778/sof_trace1217.log)

paulstelian97, 2019-12-17:
IMX8 also has this issue:

` 0      1           SA         1500815271.153153 110893208.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 1330786`

singalsu, 2019-12-17:
The file sof_trace1217.log looks empty except the header. There should definitely be something seen from the playback.

paulstelian97, 2019-12-17:
@singalsu I think this is an older problem. Should I come back with a full log dump on IMX? This one definitely isn't empty but, given the line you have noticed in my comment, takes quite a bit longer to reproduce.

singalsu, 2019-12-17:
I was wondering if this is related to audio processing and created load in pipelines. I'm seeing this quite often when I'm testing processing components. I wondered if there's SRC in hw:0,1. Seeing trace and component prepare() calls helps to figure out what's included.

paulstelian97, 2019-12-17:
@singalsu In my case there is only two HOST, two VOLUME and two DAI (the DAI components being tied to the same hardware), as seen in the sof-imx8qxp-cs42888.m4 topology, one of each per stream (playback + capture simultaneously). No SRC.

For the tested topologies I think there is at least one that doesn't have SRC (sof-tgl-rt711-rt1308.m4 doesn't seem to have any), although I believe it's not related.

tlauda, 2019-12-17:
@Liviali155 @singalsu @paulstelian97 Please try with #2220.

paulstelian97, 2019-12-17:
@tlauda If Livia reports it to be fixed (issue seems to happen much more quickly on the Intel platforms) you should be good to merge. I will chime in if the agent doesn't trigger any issue in 2 hours.

paulstelian97, 2019-12-17:
@tlauda Actually... something more interesting is happening. Your commit has the perverse effect on my system, now getting stuff like:

```
    0      1           SA         114654699.264264     14618.360352      src/lib/agent.c:55  	validate(), ll drift detected, delta = 707470
    0      1           SA         114728687.998498     73988.734375      src/lib/agent.c:55  	validate(), ll drift detected, delta = 699980
    0      1           SA         114777701.623123     49013.625000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 709194
    0      1           SA         114802687.776276     24986.152344      src/lib/agent.c:55  	validate(), ll drift detected, delta = 699956
    0      1           SA         114851691.058559     49003.281250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 702156
    0      1           SA         114900704.073574     49013.015625      src/lib/agent.c:55  	validate(), ll drift detected, delta = 710688
    0      1           SA         114925690.115616     24986.042969      src/lib/agent.c:55  	validate(), ll drift detected, delta = 701528
    0      1           SA         114974694.530030     49004.414062      src/lib/agent.c:55  	validate(), ll drift detected, delta = 704468
    0      1           SA         115048692.833333     73998.304688      src/lib/agent.c:55  	validate(), ll drift detected, delta = 703188
    0      1           SA         115097697.539039     49004.707031      src/lib/agent.c:55  	validate(), ll drift detected, delta = 706466
    0      1           SA         115171695.971471     73998.429688      src/lib/agent.c:55  	validate(), ll drift detected, delta = 705430
    0      1           SA         115220701.307808     49005.335938      src/lib/agent.c:55  	validate(), ll drift detected, delta = 708880
    0      1           SA         115294699.600601     73998.289062      src/lib/agent.c:55  	validate(), ll drift detected, delta = 707845
    0      1           SA         115368689.466967     73989.867188      src/lib/agent.c:55  	validate(), ll drift detected, delta = 701094
    0      1           SA         115417702.725225     49013.257812      src/lib/agent.c:55  	validate(), ll drift detected, delta = 709744
    0      1           SA         115442687.133634     24984.408203      src/lib/agent.c:55  	validate(), ll drift detected, delta = 699364
    0      1           SA         115491691.572072     49004.437500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 702494
    0      1           SA         115540705.722222     49014.148438      src/lib/agent.c:55  	validate(), ll drift detected, delta = 711922
    0      1           SA         115565690.665165     24984.943359      src/lib/agent.c:55  	validate(), ll drift detected, delta = 701894
    0      1           SA         115614694.055556     49003.390625      src/lib/agent.c:55  	validate(), ll drift detected, delta = 703970
    0      1           SA         115688692.605105     73998.546875      src/lib/agent.c:55  	validate(), ll drift detected, delta = 703176
    0      1           SA         115737697.515015     49004.910156      src/lib/agent.c:55  	validate(), ll drift detected, delta = 706458
    0      1           SA         115811695.548048     73998.031250      src/lib/agent.c:55  	validate(), ll drift detected, delta = 705144
    0      1           SA         115860700.067568     49004.519531      src/lib/agent.c:55  	validate(), ll drift detected, delta = 708154
    0      1           SA         115934698.521021     73998.453125      src/lib/agent.c:55  	validate(), ll drift detected, delta = 706994
    0      1           SA         116008687.959459     73989.437500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 700094
    0      1           SA         116057702.214715     49014.253906      src/lib/agent.c:55  	validate(), ll drift detected, delta = 709584
    0      1           SA         116082686.920420     24984.705078      src/lib/agent.c:55  	validate(), ll drift detected, delta = 699398
    0      1           SA         116131691.584084     49004.664062      src/lib/agent.c:55  	validate(), ll drift detected, delta = 702500
    0      1           SA         116180704.448949     49012.863281      src/lib/agent.c:55  	validate(), ll drift detected, delta = 711052
    0      1           SA         116205689.998498     24985.548828      src/lib/agent.c:55  	validate(), ll drift detected, delta = 701350
    0      1           SA         116254694.067568     49004.070312      src/lib/agent.c:55  	validate(), ll drift detected, delta = 704022
    0      1           SA         116328692.127628     73998.062500      src/lib/agent.c:55  	validate(), ll drift detected, delta = 702870
    0      1           SA         116377696.809309     49004.683594      src/lib/agent.c:55  	validate(), ll drift detected, delta = 705974
    0      1           SA         116451695.722222     73998.914062      src/lib/agent.c:55  	validate(), ll drift detected, delta = 705258
    0      1           SA         116500700.172673     49004.449219      src/lib/agent.c:55  	validate(), ll drift detected, delta = 708088
    0      1           SA         116574699.025526     73998.851562      src/lib/agent.c:55  	validate(), ll drift detected, delta = 707362
    0      1           SA         116648688.253754     73989.226562      src/lib/agent.c:55  	validate(), ll drift detected, delta = 700286
[... cut ...]
    0      1           SA         136291695.944444     73999.007812      src/lib/agent.c:55  	validate(), ll drift detected, delta = 705224
    0      1           SA         136340700.106607     49004.164062      src/lib/agent.c:55  	validate(), ll drift detected, delta = 708180
    0      1           SA         136365687.001502     24986.894531      src/lib/agent.c:55  	validate(), ll drift detected, delta = 699354
    0      1           SA         136414699.196697     49012.195312      src/lib/agent.c:55  	validate(), ll drift detected, delta = 707570
```
(note: even more than visible here)

paulstelian97, 2019-12-17:
(all errors are happening while DSP is idle)

tlauda, 2019-12-17:
@paulstelian97 #2220 was wrong. Please try with #2221.

paulstelian97, 2019-12-17:
Trying with #2221 now, will report.

paulstelian97, 2019-12-17:
Trying. Now I've got a storm of these logs after an Xrun (as in, 2 seconds worth of such logs) with no panic. For me getting these with the small delta is better than getting a delta large enough to trigger a panic but both cases are still bad.

Timestamp of first error: 182888957.115616
Timestamp of last error in burst: 198076948.747748
Actual error message is the same.

paulstelian97, 2019-12-17:
@Liviali155 IMX remains as broken as it was before. Can you report if #2221 fixes the issue on the Intel platforms? If so, I'll allow that to be merged and I'll keep on working with the agent disabled for now on IMX.

Liviali155, 2019-12-18:
> @Liviali155 IMX remains as broken as it was before. Can you report if #2221 fixes the issue on the Intel platforms? If so, I'll allow that to be merged and I'll keep on working with the agent disabled for now on IMX.

Tried with #2221 ,error not seen on below platforms:
APL-UP2 nocodec
APL-up2 PCM512x in I2S mode
TGL RT711 in HDA mode
BYT nocode, 

But can reproduce on:
BYT 5682 in I2S mode.

PR #2221 (2019-12-19): ll_schedule: fix task insertion
Fixes low latency task insertion into the priority list.
From now on tasks will be inserted in the right order based
on their priority and time of insertion.

Fixes #2219.
closed 2019-12-19 by lgirdwood