thesofproject/sof #2224: [BUG][CML] pause/release headset capture will cause pulse noise in recorded wave

aiChaoSONG, 2019-12-18:
**Describe the bug**
pause/release headset capture with SPACE will cause pulse noise in the recorded wave.
[image]

**To Reproduce**
1. run arecord via headset pipeline: `arecord -Dhw:0,0 -f dat -r 48000 -i -vvv -c2 ./headset.wav`
2. press SPACE in pair(double press)(see picture above)
3. after several pairs of press, the noise should be triggered.

***if you don't have correct UCM, the headset capture may be muted, please use the ucm file attached.***

**Reproduction Rate**
within several pairs of SPACE press, you will see the pulse noise in the terminal. should be easy to reproduce on the platform specified below.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev {98b6f21f}
    * SOF: master {3d44dead}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML with rt5682 and rt1011 codecs}

**Error log**
No obvious error seen in the log.
[dmesg_pause_release_trigger_pulse_noise.log](https://github.com/thesofproject/sof/files/3976265/dmesg_pause_release_trigger_pulse_noise.log)

[soflogger_pause_release_trigger_pulse_noise.log](https://github.com/thesofproject/sof/files/3976266/soflogger_pause_release_trigger_pulse_noise.log)

**UCM Files**
[ucm_sof-cml_rt1011_rt5682.zip](https://github.com/thesofproject/sof/files/3976267/ucm_sof-cml_rt1011_rt5682.zip)

**Recorded Wave**
[headsetwave.zip](https://github.com/thesofproject/sof/files/3976268/headsetwave.zip)

aiChaoSONG, 2019-12-18:
I also observed:
1. once the pulse is seen, it will exist along the whole recording.
2. once the pulse is seen, unplug headset, it disappears, insert jack back, it appears again.
3. once the pulse is seen,  continue double press SPACE, it may disappear and appear again.
4. Some of kcontrols which may change data route have influence on the pulse noise and the recording, such as `ADCDAT Mux` and `Stereo1 ADC L Mux`.
5. once the pulse is seen, change `Stereo1 ADC L Mux` from `ADC1 L` to `ADC1 R`,  and change it back, the pulse disappears, but press SPACE will trigger the noise again.

lgirdwood, 2019-12-18:
@bardliao any insight here, it look like a codec driver (pause/release) issue if the noise disappears when headset is removed  (2, 4, 5 and 7) ? Should be add realtek to issue ?

bardliao, 2019-12-19:
> @bardliao any insights here, it look like a codec driver (pause/release) issue if the noise disappears when 2, 4, 5 and 7 is tried ? Should be add realtek to issue ?

@lgirdwood Yes, it could be. @aiChaoSONG Could you check if MCLK is active when the recording is paused? If not, please check if MCLK starts before I2S BCLK. IIRC, MCLK starts before BCLK is a requirement of rt5682. @shumingfan can you take a look?

shumingfan, 2019-12-19:
@aiChaoSONG @bardliao I want to involve our hardware AE to debug this issue. 
He wants to know which is customer and project name to this issue?

shumingfan, 2019-12-25:
We had done some experiments like loopback the signal inside/outside of codec when pulse noise happened. We heard the sound from the headset mic to the headphone. The sound is OK. 
Another test is that AP outputs the signal to SoC instead of rt5682i codec. This issue still happened.
This issue we are discussing on Google issue tracker and please check the website below.
https://partnerissuetracker.corp.google.com/issues/145640941

aiChaoSONG, 2020-01-13:
this issue is not a sof issue nor kernel issue, close for now

aiChaoSONG, 2020-02-25:
Codecs seems fine, reopen

slawblauciak, 2020-02-25:
Is this reproducible on the high DSP clock?

bardliao, 2020-02-26:
@aiChaoSONG As I know, Realtek suspects that is a layout or HW design issue. They provided a solution to ODM and are waiting for new HW version to verify. @shumingfan Is there any update on the status of new HW version?

shumingfan, 2020-02-26:
@bardliao 
I think there is no new progress for now. Realtek HW's AE left a comment on Google issue tracker last week.
I post his commit here.
[From Realtek Albert Chen]
The issue phenomenon is completely different between ALC1011 and ALC5682.
ALC1011 request bclk output earlier than data is to prevent some sounds disappearing while start playback.
And currently issue is random noise happened during recording in CML and measurement I2S signal waveform on PCB, it’s a signal reflection on the clock can’t solved by add delay between CLK start and data.

FYI, Only for reference following my thinking.
There are two methods maybe can process signal reflection phenomenon . One is to reduce I2S driving strength and the other is to modify the I2S traces on PCB layout.
If platform can’t adjust I2S driving strength, only waiting for ODMs to modify the PCB layout for verify this issue.

bardliao, 2020-02-26:
Let's close it for now since Realtek is working with Google. We can reopen it if Realtek needs our help.

closed 2020-02-26 by bardliao

keyonjie, 2020-03-11:
Hi @aiChaoSONG can you try the method provided by Albert to check if the noise is from ADC via the 5682 A-D D-A loopback feature?

Please following sequence setup.
1. modify drivers/base/regmap/regmap-debugfs.c to enable regmap write permission
--- a/drivers/base/regmap/regmap-debugfs.c
+++ b/drivers/base/regmap/regmap-debugfs.c
@@ -274,7 +274,7 @@ static ssize_t regmap_map_read_file(struct file *file, char __user *user_buf,
                                  count, ppos);
}

-#undef REGMAP_ALLOW_WRITE_DEBUGFS
+#define REGMAP_ALLOW_WRITE_DEBUGFS
#ifdef REGMAP_ALLOW_WRITE_DEBUGFS

2. enable ADDA path in rt5682i internal while recording noise happened
# sudo -i
# cd /sys/kernel/debug/regmap/i2c-10EC5682\:00/
# echo 0029 4040 > registers

3. run aplay and check if there is noise rendered to the headset when the arecord is recording noise.

aiChaoSONG, 2020-03-11:
@keyonjie 
- trigger noise in recording
- aplay /dev/zero
- I hear nothing but very tiny white noise
so noise may not be produced by codec

keyonjie, 2020-03-11:
Thanks @aiChaoSONG 

So next step let's try if we can reproduce similar issue with nocodec SSP LBM feature.

Using any other verified inputs(e.g. input from AP, from rt5651, DMIC injector, ...) to reproduce it can help for analyzing also.

aiChaoSONG, 2020-03-12:
@keyonjie I tried nocodec kernel and tplg:
- only arecord, run arecord 30 times or pause-release
- aplay fixed level wave, and arecord, run arecord 30 times or pause-release
- aplay /dev/zero, and arecord, run arecord 30 times or pause-release
- aplay a normal music, and arecord, run arecord 30 times or pause-release
none of the above test reproduced the issue. Very wired.

lancedkhou, 2020-03-12:
@aiChaoSONG 
The noise symptom might be low reproduction rate in a certain sku of SoC.
Normally I will use test scripts to reproduce.

Let me mail you test scripts to see whether it helps.