thesofproject/sof #2225: [BUG][gcc]DSP panic in eq-test

Liviali155, 2019-12-18:
**Describe the bug**
DSP panic in eq-test, issue only seen  when build with gcc compiler, error not seen when build with xcc compiler

**To Reproduce**
1.   Firmware is build with gcc compiler
2.  cd ~/sofcitest/sof-test-suites/eq-test/eqctl
3.  ./eq_test.sh

**Reproduction Rate**
10/10

**Expected behavior**
EQ test pass

**Impact**
aplay: pcm_write:2011: write error: Input/output error
And DSP panic in dmesg.

**Environment**
Kernel: sof-dev {98b6f21f}
SOF: master {3d44dea}
Topology: sof-apl-eq-pcm512x.tplg renamed to sof-apl-pcm512x.tplg
Platform: APL UP2 pcm512x in I2S mode

<details>
 <summary>dmesgerror.log</summary>

````
[   71.765466] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   71.765473] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   71.772452] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   71.772460] sof-audio-pci 0000:00:0e.0: pcm: stream dir 0, posn mailbox offset is 790528
[   71.786357] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 1
[   71.786896] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x14001e successful
[   71.786902] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   71.787128] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   71.789264] sof-audio-pci 0000:00:0e.0: error : DSP panic!
[   71.789272] sof-audio-pci 0000:00:0e.0: panic: dsp_oops_offset 788480 offset 788480
[   71.789276] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   71.789319] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0xbe00297c trace 0x00004000
[   71.789321] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   73.504265] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[   73.504275] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   74.007194] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[   74.007203] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   74.007210] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   74.007260] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0xbe00297c trace 0x00004000
[   74.007268] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[   74.007270] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   74.007274] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   74.007277] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   74.007286] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[   74.007292]  Port5: ASoC: trigger FE cmd: 0 failed: -110
[   74.007322] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[   74.007328] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   74.511287] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
[   74.511293] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   74.511299] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   74.511342] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0xbe00297c trace 0x00004000
[   74.511348] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   74.511350] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   74.511354] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   74.511357] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   74.525350] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[   74.535043] sof-audio-pci 0000:00:0e.0: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[   75.039187] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x50030000 size 180
[   75.039194] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[   75.039200] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[   75.039242] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0xbe00297c trace 0x00004000
[   75.039248] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   75.039250] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   75.039255] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   75.039257] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[   75.039260] sof-audio-pci 0000:00:0e.0: error: set/get ctrl ipc comp 3
[   75.048437] pcm512x i2c-104C5122:00: No SCLK, using BCLK: -2
[   75.048454] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[   75.048457] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
[   75.048459] sof-audio-pci 0000:00:0e.0: period count 2 max 16
[   75.048460] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
[   75.048840] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[   75.048843] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2	
````
</details>

[1217_eq-test_dsp_panic.log](https://github.com/thesofproject/sof/files/3976599/1217_eq-test_dsp_panic.log)

singalsu, 2019-12-18:
Thanks for reporting but from where can I retrieve sofcitest?

Meanwhile could you test if PR #2223 fixes this issue.

Liviali155, 2019-12-18:
> Thanks for reporting but from where can I retrieve sofcitest?
> 
> Meanwhile could you test if PR #2223 fixes this issue.

Tried with #2223, issue can  still be reproduced
sofcitest.zip as attachment
Test step:
1.chmod +x sof-ctl (sof-ctl can be found at ~/sof/tools/build_tools/ctl after build success)
2.sudo mv sof-ctl /usr/bin/
3.get sof-apl-eq-pcm512x.tplg,then renamed to sof-apl-pcm512x.tplg
4.reboot
5.cd ~/sofcitest/sof-test-suites/eq-test/eqctl
6./eq_test.sh

[sofcitest.zip](https://github.com/thesofproject/sof/files/3977598/sofcitest.zip)

singalsu, 2019-12-18:
Thanks @Liviali155 ! I'm now able to repeat this issue. It happens after completing IIR tests with second FIR response. The first FIR test is a simple "flat" one coefficient filter and it's passed. The second test with a much longer mid-boost response fails.

I suspect it's due to insufficient execution speed with gcc build. Maybe I should limit in gcc version the max filter length to something that can execute in real time. Though the FW should not panic, only emit xruns in such case. I'll investigate what happens.

singalsu, 2019-12-18:
PR #2221 avoids the DSP panic. I tested it because the error trace contained "validate(), ll drift detected, delta = 27959" like events.

The playback is with non-default FIR still fails due to known reason but at least it fails in a more nice way with that PR included.

I think I should add for gcc build a runtime fail for exceeding know safe limit (around 30 tap filter). And make available at least one such short filter to test gcc builds.

singalsu, 2019-12-18:
With the PR applied the error trace shows "timer_report_delay(): timer 3 delayed by 221 uS 4262 ticks" and aplay fails to "aplay: pcm_write:2011: write error: Input/output error" that's a lot nicer behavior than a panic.

lgirdwood, 2019-12-18:
@singalsu supporting a very reduced filter length is best for gcc. It just needs to be functional, you could state this in Kconfig too when selecting the EQ component.

Liviali155, 2019-12-20:
On latest sof master test ,there is no DSP panic in dmesg,the issue is just "aplay: pcm_write:2011: write error: Input/output error" in eq test,because  #2221 merged

Liviali155, 2019-12-20:
Tried with PR #2230,tips"Equalizer test failure with eq_fir_mid.txt" and dmesg show "error: ipc error for 0x60010000 size 20"

[image]

<details>
 <summary>dmesgerror.log</summary>

````
[   94.287316] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   94.287331] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   94.287334] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
[   94.287335] sof-audio-pci 0000:00:0e.0: periods:4
[   94.287344] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   94.287350] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   94.287851] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   94.287859] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[   94.287863] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
[   94.287868]  Port5: ASoC: hw_params FE failed -22
[   94.288815] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[   94.290067] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[   95.271829] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   95.271843] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   96.790338] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[   96.795190] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   96.795320] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   96.795396] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[   96.795403] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[   96.795409] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[   96.795441] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000010 at  00000044
````
</details>

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1       EQ_FIR          94282637.343750  94282640.000000 udio/eq_fir/eq_fir.c:356 	Filter length 84 exceeds limitation for build.
    0      1       EQ_FIR 1.3      94282643.333333         5.989583 udio/eq_fir/eq_fir.c:813 	eq_fir_prepare() error: eq_fir_setup failed.
    0      1         PIPE          94282649.322917         5.989583 src/audio/pipeline.c:501 	pipeline_prepare() error: ret = -22,dev->comp.id = 0
    0      1          IPC          94282655.625000         6.302083    src/ipc/handler.c:307 	ipc: pipe 1 comp 0 prepare failed -22
```
[dmesg1220.log](https://github.com/thesofproject/sof/files/3987408/dmesg1220.log)
[sof_errortrace1220.log](https://github.com/thesofproject/sof/files/3987409/sof_errortrace1220.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/3987410/sof_trace.log)

singalsu, 2019-12-20:
> @singalsu supporting a very reduced filter length is best for gcc. It just needs to be functional (pls create a new minimal filter if needed), you could state this in Kconfig too when selecting the EQ component.

Yep, I'll it to PR #2230. There's also similar issue with GCC built ASRC. I'll add it with soon coming PR for it.

Liviali155, 2019-12-20:
Sorry for click a wrong menu to closed it，reopen it

singalsu, 2019-12-20:
> Tried with PR #2230,tips"Equalizer test failure with eq_fir_mid.txt" and dmesg show "error: ipc error for 0x60010000 size 20"
> 
> [image]
> 
> dmesgerror.log
> ```
>  CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
>     0      1       EQ_FIR          94282637.343750  94282640.000000 udio/eq_fir/eq_fir.c:356 	Filter length 84 exceeds limitation for build.
>     0      1       EQ_FIR 1.3      94282643.333333         5.989583 udio/eq_fir/eq_fir.c:813 	eq_fir_prepare() error: eq_fir_setup failed.
>     0      1         PIPE          94282649.322917         5.989583 src/audio/pipeline.c:501 	pipeline_prepare() error: ret = -22,dev->comp.id = 0
>     0      1          IPC          94282655.625000         6.302083    src/ipc/handler.c:307 	ipc: pipe 1 comp 0 prepare failed -22
> ```
> 
> [dmesg1220.log](https://github.com/thesofproject/sof/files/3987408/dmesg1220.log)
> [sof_errortrace1220.log](https://github.com/thesofproject/sof/files/3987409/sof_errortrace1220.log)
> [sof_trace.log](https://github.com/thesofproject/sof/files/3987410/sof_trace.log)

There's a new eq_fir_mid.txt version included in the PR. Please update sofcitest with it once PR #2230 is merged.

Liviali155, 2019-12-23:
> There's a new eq_fir_mid.txt version included in the PR. Please update sofcitest with it once PR #2230 is merged.

After update eq_fir_mid.txt and sof-apl-eq-pcm512x.tpl,Issue can still be reproduced

singalsu, 2020-01-02:
For me the new eq_fir_mid.txt passes the test. Based on screenshot above the file you are using is the old one. You can see the content of the new file below:

```
Applying configuration "eq_fir_mid.txt" into device hw:0 control numid=23.
4607827,0,128,50380800,0,0,0,0,128,65538,0,0,0,0,0,65576,0,0,0,0,1202154746,4222755303,3873958283,4024953600,4126012299,4176934804,4217764409,4250664050,4273864222,4288806716,2490349,6750285,8126582,7733371,6488174,4980824,3473472,2293804,1245211,1245197
Success.
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Control size is 304.
Applying configuration "eq_fir_pass.txt" into device hw:0 control numid=23.
4607827,0,56,50331648,0,0,0,0,56,65538,0,0,0,0,4294967295,4294901764,0,0,0,0,16384,0
Success.
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Control size is 304.
Applying configuration "eq_fir_loudness.txt" into device hw:0 control numid=23.
4607827,0,256,50331648,0,0,0,0,256,131076,0,0,0,0,65537,65537,4294901764,0,0,0,0,16384,0,88,0,0,0,0,3535366939,119016436,49938453,30147089,30343698,30409180,29426121,27328945,25428369,23921015,22479202,20906315,19267889,17826074,16646407,15663350,14745832,13762777,12845259,11862204,10879149,10027167,9175186,8388742,7602298,6946927,6291557,5636187,5046354,4522057,3997761,3539002,3145779,2752557,2424872,2162723,1900575,1638427,1441815,1245204,1048593,917519,720909,7
Success.
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
Equalizer test failure with eq_fir_loudness.txt

```

As you can see above the eq_fir_loudness.txt fails so that case need to be removed for tests for gcc build. Only the xt-xcc build can run it. Please let me know if check/fix of your eq_fir_mid.txt helped.

Liviali155, 2020-01-03:
Tried on latest sof-dev+master used commmand:
./sat -c "tplg=sof-apl-pcm512x.tplg" -s eq-test -xcheck-audio-equalizer.sh
eq test pass both on gcc and xcc
but when use ./eq_test.sh to test,both failed on gcc and xcc:
```
ubuntu@cd-apl-up2-pcm512x-07:~/sofcitest/sof-test-suites/eq-test/eqctl$ ./eq_test.sh
./eq_test.sh: line 7: sof-eqctl: command not found
Equalizer setting failure with eq_iir_bandpass.txt
```

Environment
Kernel: 52ec20e2 https://github.com/thesofproject/linux/commits/topic/sof-dev
SOF: 0194175d  https://github.com/thesofproject/sof/commits/master
Topology: sof-apl-eq-pcm512x.tplg renamed to sof-apl-pcm512x.tplg
Platform: APL UP2 pcm512x in I2S mode

singalsu, 2020-01-03:
The script should be updated, it's "sof-ctl" and not the old executable name ("sof-eqctl").

Liviali155, 2020-01-06:
> The script should be updated, it's "sof-ctl" and not the old executable name ("sof-eqctl").

After updated the script, eq test pass on xcc,test failure with eq_iir_bandpass.txt on gcc,dmesg show " error: ipc error for 0x60010000 size 20"

[image]

Environment
Kernel: 52ec20e2 https://github.com/thesofproject/linux/commits/topic/sof-dev
SOF: 373f3319 https://github.com/thesofproject/sof/commits/master
Topology: sof-apl-eq-pcm512x.tplg renamed to sof-apl-pcm512x.tplg
Platform: APL UP2 pcm512x in I2S mode

singalsu, 2020-01-08:
@Liviali155 There's need to remove eq_fir_loudness.txt for gcc build tests. So there would need to be a command line parameter passed to eq_test.sh to prevent testing it with not optimized version code.

Or alternatively make the fail an expected and required event for gcc build version test. It would be even better.

lgirdwood, 2020-01-08:
@singalsu why would installing hw_params fail with GCC ? I would expect us to fail after the stream had started with some XRUNs, not before stream start ? Could this be a ```#ifdef GCC``` logic bug in the EQ ?

singalsu, 2020-01-08:
The `#if defined FIR_MAX_LENGTH_BUILD_SPECIFIC` code causes an intentional fail when the firmware has been configured for too long filter for the GCC build to handle. In this case the example eq_fir_loudness.txt  is such.  The -EINVAIL fail in prepare() prevents the firmware into go into endless xruns mode (and scheduler related panic that happened before PR #2221).

I wanted to keep there one filter for sof-ctl usage that exercises the xt-xcc build version capability. In fact I should update it into an even longer version that exercises the multi-part IPC as I do in my own tests. The version in git master still fits single IPC message.

lgirdwood, 2020-01-08:
@singalsu thanks got you, we can ignore this test for GCC then with big FIR lengths.

Liviali155, 2020-01-13:
reference the previous  comment of lgirdwood

closed 2020-01-13 by Liviali155