thesofproject/sof #2227: [BUG] [BDW] aplay failed on bdw

xiulipan, 2019-12-18:
**Describe the bug**
aplay will got `aplay: pcm_write:2011: write error: Input/output error` error
**To Reproduce**
aplay -Dhw:0,0 -f dat /dev/zero -vvv

**Reproduction Rate**
10/10

**Expected behavior**
aplay can work

**Impact**
bdw playback is broken again

**Environment**
last kernel (98b6f21f) with nocodec
and fw with 3d44dead

**Screenshots or console output**
```
$ aplay -Dhw:0,1 -f dat /dev/zero -vvv
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-broadwell-rt286' device 1 subdevice 0
Max peak (12000 samples): 0x00000000 #                    0%
Max peak (12000 samples): 0x00000000 #                    0%
Max peak (12000 samples): 0x00000000 #                    0%
Max peak (12000 samples): 0x00000000 #                    0%
aplay: pcm_write:2011: write error: Input/output error
```
error trace:
```
$ sudo sof-logger -l /etc/sof/sof-bdw.ldc
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1           SA         13891601120.572918              nan      src/lib/agent.c:55   validate(), ll drift detected, delta = 358652
```
dma trace:
```
    0      2          IPC         13976164179.479168  76163264.000000    src/ipc/handler.c:228  ipc: comp 12 -> params
    0      2          DMA         13976164235.000000        55.520832 src/drivers/dw/dma.c:137  dw_dma_channel_get(): dma 1 request channel 0
    0      2          DMA         13976167215.156250      2980.156250 src/drivers/dw/dma.c:196  dw_dma_channel_put(): dma 1 channel 3 put
    0      2         PIPE 2.17    13976167448.072918       232.916672 src/audio/pipeline.c:385  pipeline_params()
    0      2         HOST 2.12    13976167499.427084        51.354168     src/audio/host.c:467  host_params()
    0      2       BUFFER         13976167548.958334        49.531250   src/audio/buffer.c:25   buffer_alloc()
    0      2          DMA         13976167889.583334       340.625000 src/drivers/dw/dma.c:137  dw_dma_channel_get(): dma 1 request channel 65535
    0      2       VOLUME 2.13    13976168045.677084       156.093750 udio/volume/volume.c:267  volume_params()
    0      2          DAI 2.16    13976168102.343750        56.666668      src/audio/dai.c:352  dai_params()
    0      2       BUFFER         13976168150.677084        48.333332   src/audio/buffer.c:25   buffer_alloc()
    0      2          DAI 2.16    13976168575.989584       425.312500      src/audio/dai.c:313  dai_capture_params() src_dev = 2 stream_id = 0 src_width = 4 dest_width = 4
    0      2          DAI 2.16    13976168611.302084        35.312500      src/audio/dai.c:320  dai_capture_params() fifo FFFFC010
    0      2         PIPE 2.17    13976168700.416668        89.114586 src/audio/pipeline.c:492  pipeline_prepare()
    0      2         HOST 2.12    13976168824.166668       123.750000     src/audio/host.c:618  host_prepare()
    0      2       VOLUME 2.13    13976169151.250000       327.083344 udio/volume/volume.c:621  volume_prepare()
    0      2          DAI 2.16    13976169484.843750       333.593750      src/audio/dai.c:450  dai_prepare()
    0      2          IPC         13976180275.208334     10790.364258    src/ipc/handler.c:457  ipc: comp 12 -> trigger cmd 0x40000
    0      2         PIPE 2.17    13976180325.729168        50.520832 src/audio/pipeline.c:711  pipeline_trigger()
    0      2         HOST 2.12    13976180363.645834        37.916668     src/audio/host.c:299  host_trigger()
    0      2       VOLUME 2.13    13976180464.895834       101.250000 udio/volume/volume.c:563  volume_trigger()
    0      2          DAI 2.16    13976180510.520834        45.625000      src/audio/dai.c:536  dai_comp_trigger(), command = 1
    0      2          DAI 2.16    13976180589.322918        78.802086      src/audio/dai.c:552  dai_comp_trigger(), START
    0      2          SSP         13976180624.114584        34.791668 /intel/haswell/ssp.c:481  ssp_trigger()
    0      2          SSP         13976180660.364584        36.250000 /intel/haswell/ssp.c:417  ssp_start()
    0      2  SCHEDULE_LL         13976180745.000000        84.635414 _multi_chan_domain.c:109  dma_multi_chan_domain_register()
    0      2  SCHEDULE_LL         13976180788.333334        43.333332 _multi_chan_domain.c:70   dma_multi_chan_domain_irq_register()
    0      2          IPC         14003126495.781250  26945708.000000    src/ipc/handler.c:457  ipc: comp 12 -> trigger cmd 0x50000
    0      2         PIPE 2.17    14003126546.250000        50.468750 src/audio/pipeline.c:711  pipeline_trigger()
    0      2         HOST 2.12    14003126584.166668        37.916668     src/audio/host.c:299  host_trigger()
    0      2       VOLUME 2.13    14003126692.708334       108.541664 udio/volume/volume.c:563  volume_trigger()
    0      2          DAI 2.16    14003126738.125000        45.416668      src/audio/dai.c:536  dai_comp_trigger(), command = 0
    0      2          DAI 2.16    14003126835.989584        97.864586      src/audio/dai.c:600  dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         14003126870.729168        34.739582 src/drivers/dw/dma.c:366  dw_dma_stop(): dma 1 channel 2 stop
[... 34 lines cut ...]
    0      2         HOST 3.18    25234634204.114586        37.031250     src/audio/host.c:618  host_prepare()
    0      2       VOLUME 3.19    25234636491.041668      2286.927002 udio/volume/volume.c:621  volume_prepare()
    0      2          SRC 3.20    25234638784.479168      2293.437500  src/audio/src/src.c:792  src_prepare()
    0      2        MIXER 1.4     25234641079.583336      2295.104248    src/audio/mixer.c:367  mixer_prepare()
    0      2       VOLUME 1.3     25234641410.625000       331.041656 udio/volume/volume.c:621  volume_prepare()
    0      2          DAI 1.9     25234641744.739586       334.114594      src/audio/dai.c:450  dai_prepare()
    0      2          IPC         25234657535.208336     15790.468750    src/ipc/handler.c:457  ipc: comp 18 -> trigger cmd 0x40000
    0      2         PIPE 3.24    25234657593.281250        58.072918 src/audio/pipeline.c:711  pipeline_trigger()
    0      2         HOST 3.18    25234657631.197918        37.916668     src/audio/host.c:299  host_trigger()
    0      2       VOLUME 3.19    25234657734.479168       103.281250 udio/volume/volume.c:563  volume_trigger()
    0      2          SRC 3.20    25234657780.104168        45.625000  src/audio/src/src.c:679  src_trigger()
    0      2  SCHEDULE_LL         25234657829.322918        49.218750 _multi_chan_domain.c:109  dma_multi_chan_domain_register()
    0      2        MIXER 1.4     25234657904.687500        75.364586    src/audio/mixer.c:232  mixer_trigger()
    0      2       VOLUME 1.3     25234657951.145836        46.458332 udio/volume/volume.c:563  volume_trigger()
    0      2          DAI 1.9     25234657996.302086        45.156250      src/audio/dai.c:536  dai_comp_trigger(), command = 1
    0      2          DAI 1.9     25234658097.187500       100.885414      src/audio/dai.c:552  dai_comp_trigger(), START
    0      2          SSP         25234658131.979168        34.791668 /intel/haswell/ssp.c:481  ssp_trigger()
    0      2          SSP         25234658168.229168        36.250000 /intel/haswell/ssp.c:417  ssp_start()
    0      2  SCHEDULE_LL         25234658254.427086        86.197914 _multi_chan_domain.c:109  dma_multi_chan_domain_register()
    0      2  SCHEDULE_LL         25234658296.875000        42.447918 _multi_chan_domain.c:70   dma_multi_chan_domain_irq_register()
```

bisect to be caused by c89d916917edf171bcfef7601fc807f185482453@https://github.com/thesofproject/sof/pull/1950

arecord is not influenced by this case.

paulstelian97, 2019-12-18:
Related to #2214 (same bisect, same manifestation). In my case, once aplay fails I cannot start it again and arecord also fails with IPC timeout.

xiulipan, 2019-12-18:
@paulstelian97  update with some debug finding:
Add trace here and find there is no such caller to dma chan in the `notify->list`
https://github.com/thesofproject/sof/blob/3d44dead6fd750e308ee160ab8da34f48f55feeb/src/lib/notifier.c#L118-L122

Then add log to https://github.com/thesofproject/sof/blob/3d44dead6fd750e308ee160ab8da34f48f55feeb/src/lib/notifier.c#L69 the chan was unregistered from https://github.com/thesofproject/sof/blob/3d44dead6fd750e308ee160ab8da34f48f55feeb/src/lib/notifier.c#L100 
Then find it was called in https://github.com/thesofproject/sof/blob/3d44dead6fd750e308ee160ab8da34f48f55feeb/src/drivers/dw/dma.c#L178 https://github.com/thesofproject/sof/blob/3d44dead6fd750e308ee160ab8da34f48f55feeb/src/drivers/dw/dma.c#L191
Not sure why the chan is freed before we are trying to use the notifier register with them.  @tlauda @akloniex @jajanusz  could you help to keep track the issue?

paulstelian97, 2019-12-18:
@xiulipan Can you actually point out to where the notifier_register would happen after a dw_dma_channel_put happened? Also, would XRUN recovery have anything to do with that situation?

lgirdwood, 2019-12-18:
@xiulipan, @fredoh9  what's ETA for BDW on CI ?

ranj063, 2019-12-18:
@xiulipan How could this happen within the last 2 days? When we merged https://github.com/thesofproject/sof/pull/2188, wasnt aplay working?

closed 2019-12-19 by lgirdwood