thesofproject/sof #2236: [BUG][APL] Sream stall on simple stream operation stress tests.

jdrzaszcz, 2019-12-20:
**Describe the bug**
When running simultaneous capture and playback on external loopback (same SSP) with pause/release on playback pipeline after some iteration capture stream or playback stream is not running, stream stall.

**Topology**
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```

**To Reproduce**
Slim Driver python test:
On APL run test:
02_12_TestPtpSimpleStreamOperations48000Hz24b32b8ch with --playback_iterations>=1000

**Reproduction Rate**
30 %

**Environment**
Sof master branch 468d8973145e66a47836e2fcab158dd1d4180edf commit

Example log from APL:
[2019_12_17_13_01_12.zip](https://github.com/thesofproject/sof/files/3987301/2019_12_17_13_01_12.zip)
[2019_12_19_10_23_40.zip](https://github.com/thesofproject/sof/files/3987302/2019_12_19_10_23_40.zip)

lgirdwood, 2019-12-20:
@jdrzaszcz I like your topology diagrams :) Do you have a tool that generates these ?
Also can you paste the relevant error messages from the logs into the report as this allows the developers to quickly check and report any observations that could help. Just 10 or so lines is enough.

jdrzaszcz, 2019-12-20:
@lgirdwood You can use this tool http://asciiflow.com/ to generate topology diagrams.
I have added log with information about drift. In one example, there are no more logs after drift, in another there is additional information.

lgirdwood, 2019-12-20:
@jdrzaszcz thanks, lot more meaningful now. Btw, I've used markdown on the log but it looks like it's has some garbage in the file ? (maybe CR/LF ?)
@tlauda could IPC processing get lost with the drift error ?

paulstelian97, 2019-12-20:
The garbage must be ANSI codes to show colors etc. Looks like that.

jdrzaszcz, 2019-12-20:
@lgirdwood Thanks for making the log info more transparent. These are some special characters (as far as I know they are useful to coloring the lines) and we received output  this log from sof_logger,  if we would like to get output without this signs, then some changes should be made in sof_logger.

paulstelian97, 2019-12-20:
In the source code, `tools/logger/logger.c:268` there is the condition to check if the output file is a tty and to disable colors if it's not. Why shouldn't that work? At least on my end it works fine.

jdrzaszcz, 2019-12-20:
@paulstelian97 Thanks for information.

paulstelian97, 2019-12-20:
Anyway this looks like the host doesn't handle IPC in time and the IPC queue fills up. But why would the host be slow to respond?

slawblauciak, 2020-01-28:
@jdrzaszcz is this issue still reproducible on master? If yes, does it happen with high frequency clock enabled in the platform settings of menuconfig?

jdrzaszcz, 2020-01-30:
This issue dosen't occure on master branch now.
I checked it with 0e83f7d094630c3cdbc93a138ccc91efcd4c7ffa commit.

closed 2020-01-30 by jdrzaszcz