thesofproject/sof #2241: [BUG] Audio device not recognized on HP Omen 17

singingtelegram, 2019-12-21:
**Describe the bug**
No audio device (except dummy device) available. 

**To Reproduce**
Upgrade to 5.4 kernel. / Attempt to make DMIC available under Linux by blacklisting snd_hda_intel and switching to SOF (kernel bug #201251).

**Reproduction Rate**
10/10

**Expected behavior**
SOF firmware is loaded and the audio device is detected.

**Impact**
Unable to output audio.

**Environment**
Platform: HP OMEN by HP Laptop 17-cb0xxx
Graphics: IGP Disabled / Nvidia Turing
Audio device: 00:1f.3 Multimedia audio controller: Intel Corporation Cannon Lake PCH cAVS (rev 10) / Realtek ALC285 / Microphone Array - Intel Smart Sound Technology (under Windows - not detected with snd_hda_intel)
Kernel: 5.4.x and 5.5rc2
Topology: probably irrelevant as firmware wasn't even loaded, but all topology files were copied to /usr/lib/firmware/intel/sof-tplg.

**Screenshots or console output**
* lspci -kv:
```
00:1f.3 Multimedia audio controller: Intel Corporation Cannon Lake PCH cAVS (rev 10)
	Subsystem: Hewlett-Packard Company Cannon Lake PCH cAVS
	Flags: bus master, fast devsel, latency 32, IRQ 16
	Memory at da210000 (64-bit, non-prefetchable) [size=16K]
	Memory at da100000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: sof-audio-pci
	Kernel modules: snd_hda_intel, snd_sof_pci
```
* dmesg entries:
```
[    2.606198] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.647708] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.647721] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.648231] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    2.648235] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.648390] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.649128] sof-audio-pci 0000:00:1f.3: error: init i915 and HDMI codec failed
[    2.649239] sof-audio-pci 0000:00:1f.3: error: failed to probe DSP -19
[    2.649267] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -19
```

**Related reports**
[https://bugs.archlinux.org/task/64720](https://bugs.archlinux.org/task/64720) - same device

lgirdwood, 2019-12-23:
@mengdonglin this looks like we have some kconfig? misconfiguration of graphics since i915 init is failing. Can someone in PRC check this. Thanks

plbossart, 2020-01-02:
@lgirdwood @kv2019i this looks like a larger problem: if the intel graphics is not enabled, then all the logic in the machine driver to use iDISP links will fail, and so will the UCM parts...That really means we need to enable the case when iDISP is not present

We don't have such a device in our development set so no wonder it wasn't seen earlier...