thesofproject/sof #2245: [BUG][tag-v1.4.1]DSP panic when arecord via DMIC

dengyangchao, 2019-12-31:
**Describe the bug**
DSP panic when change DMIC arecord between HW:0,6 and HW:0,7 quickly.
Same issue #2176  has been fixed on master through PR #2223, should back port to release v1.4.1

Can execute a script test1.sh like below:
```
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,6 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
arecord -Dhw:0,7 -fs16_le -c4 -r 48000 test.wav -vv -i -d 1
```
Arecord -l list
```
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**To Reproduce**
1.Reboot system via"sudo reboot".
2.Wait runtime PM to suspend.
3.Execute script via./test1.sh
test1.sh.txt

**Reproduction Rate**
3/3

**Expected behavior**
Change dmic arecord successfully

Impact
Cause dsp panic

**Environment**
    * Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev 7fde4754
    * SOF: https://github.com/thesofproject/sof/commits/v1.4.1  fcf6c8c
    * Topology: file: topology/sof-hda-generic-4ch.tplg same as firmware
    * Platform: ICL-U RVP with onboard codec ALC700 in HDA mode

<details>
 <summary>Dmesg</summary>

````
[   77.397486] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[   77.397490] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   77.397688] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   77.397703] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   77.397712] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   77.397720] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   77.397900] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   77.397904] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   77.397908] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   77.397911] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   77.397916] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   77.397920] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060325 SAR     0x00000000
[   77.397925] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00da03 EPC2     0xbe01cee5 EPC3     0x00000000 EPC4    0x00000000
[   77.397929] sof-audio-pci 0000:00:1f.3: EPC5     0xbe00d0f6 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   77.397933] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00060d20
[   77.397937] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   77.397940] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe04bfa0
[   77.397946] sof-audio-pci 0000:00:1f.3: 0xbe04bfa0: be00da03 00000000 00000000 00000000
[... 107 lines cut ...]
    0      2          DAI 11.74   153278632.552083         9.218750      src/audio/dai.c:440 	dai_prepare()
    0      2          IPC         153281232.760417      2600.208252    src/ipc/handler.c:459 	ipc: comp 68 -> trigger cmd 0x40000
    0      2         PIPE 11.75   153281268.906250        36.145832 src/audio/pipeline.c:664 	pipeline_trigger()
    0      2         HOST 11.68   153281276.458333         7.552083     src/audio/host.c:295 	host_trigger()
    0      2          DMA         153281285.000000         8.541667 intel/cavs/hda-dma.c:456 	hda-dmac: 4 channel 1 -> start
    0      2          DMA         153281292.864583         7.864583 intel/cavs/hda-dma.c:321 	hda-dmac: 4 channel 1 -> enable
    0      2       VOLUME 11.69   153281358.489583        65.625000 udio/volume/volume.c:548 	volume_trigger()
    0      2       EQ_IIR 11.70   153281366.770833         8.281250 udio/eq_iir/eq_iir.c:758 	eq_iir_trigger()
    0      2          DAI 11.74   153281375.156250         8.385417      src/audio/dai.c:526 	dai_comp_trigger(), command = 1
    0      2          DAI 11.74   153281382.395833         7.239583      src/audio/dai.c:537 	dai_comp_trigger(), START
    0      2         DMIC         153281389.531250         7.135417 rs/intel/cavs/dmic.c:1429	dmic_trigger()
    0      2         DMIC         153281396.927083         7.395833 rs/intel/cavs/dmic.c:1229	dmic_start()
    0      2         DMIC         153281410.781250        13.854167 rs/intel/cavs/dmic.c:1247	dmic_start(), dmic->fifo_b
    0      2         DMIC         153281418.541667         7.760417 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         153281426.979167         8.437500 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         153281434.947917         7.968750 rs/intel/cavs/dmic.c:1331	dmic_start(), dmic_active_fifos = 1
    0      2  SCHEDULE_LL         153281445.052083        10.104167 edule/timer_domain.c:47  	timer_domain_register()
    0      2         DMIC         153281455.468750        10.416667 rs/intel/cavs/dmic.c:1344	dmic_start(), done

````
</details>

[dmesg.log](https://github.com/thesofproject/sof/files/4012001/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/4012002/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/4012003/logger-error.log)

lgirdwood, 2020-01-06:
Relevent context
Kernel
```
[   76.360924] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   76.361199] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   76.361204] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[   76.361825] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 1
[   76.362369] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[   76.362373] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   76.362608] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   77.184888] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   77.184928] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   77.386166] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 0
[   77.386175] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   77.386344] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   77.386860] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[   77.386871] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   77.387026] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   77.387052] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 1
[   77.387161] sof-audio-pci 0000:00:1f.3: pcm: close stream 6 dir 1
[   77.395582] sof-audio-pci 0000:00:1f.3: pcm: open stream 7 dir 1
[   77.395586] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[   77.395588] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[   77.395590] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[   77.395959] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
[   77.395964] sof-audio-pci 0000:00:1f.3: generating page table for 00000000b249d770 size 0xf9c0 pages 16
[   77.395972] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[   77.395994] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   77.396000] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e40
[   77.396001] sof-audio-pci 0000:00:1f.3: periods:4
[   77.396030] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   77.396041] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   77.396306] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   77.396309] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790604
[   77.396945] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 1
[   77.397486] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[   77.397490] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   77.397688] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   77.397703] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   77.397712] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   77.397720] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   77.397900] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   77.397904] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   77.397908] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   77.397911] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   77.397916] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   77.397920] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000000 PS       0x00060325 SAR     0x00000000
[   77.397925] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00da03 EPC2     0xbe01cee5 EPC3     0x00000000 EPC4    0x00000000
[   77.397929] sof-audio-pci 0000:00:1f.3: EPC5     0xbe00d0f6 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   77.397933] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00060d20
[   77.397937] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   77.397940] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe04bfa0
[   77.397946] sof-audio-pci 0000:00:1f.3: 0xbe04bfa0: be00da03 00000000 00000000 00000000
[   77.397951] sof-audio-pci 0000:00:1f.3: 0xbe04bfa4: 00000000 00000000 be080c40 be080d00
[   77.397956] sof-audio-pci 0000:00:1f.3: 0xbe04bfa8: 4379d000 8b324fac 5b597028 ffffa383
[   77.397961] sof-audio-pci 0000:00:1f.3: 0xbe04bfac: 5b597028 ffffa383 00000099 00000000
[   77.397965] sof-audio-pci 0000:00:1f.3: 0xbe04bfb0: 59cb2000 ffffa383 5bb47140 ffffa383
[   77.397970] sof-audio-pci 0000:00:1f.3: 0xbe04bfb4: 5a9646c0 ffffa383 5c3b1100 ffffa383
[   77.397976] sof-audio-pci 0000:00:1f.3: 0xbe04bfb8: a3287d30 ffffffff 00000000 00000000
[   77.397981] sof-audio-pci 0000:00:1f.3: 0xbe04bfbc: 00000000 00000000 00000000 00000000
[   77.397984] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   77.905248] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[   77.905255] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   78.408931] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   78.408938] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
```
Firmware
```
[... cut ...]
    0      2         DMIC         153281426.979167         8.437500 rs/intel/cavs/dmic.c:1274	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC         153281434.947917         7.968750 rs/intel/cavs/dmic.c:1331	dmic_start(), dmic_active_fifos = 1
    0      2  SCHEDULE_LL         153281445.052083        10.104167 edule/timer_domain.c:47  	timer_domain_register()
    0      2         DMIC         153281455.468750        10.416667 rs/intel/cavs/dmic.c:1344	dmic_start(), done
```
Firmware trace is not as uptodate as error log above.

perexg, 2020-01-10:
It looks like a serious issue. When will be 1.4.2 released?

plbossart, 2020-01-10:
> It looks like a serious issue. When will be 1.4.2 released?

It's annoying but not necessarily a crippling problem IMO.

The microphone users will typically always use the full-band 48kHz input, the 16kHz path is really intended for application such as hotwording/event detection. switching between the two cases is unlikely.

To be honest, I don't even know why this path would be exposed to users in a regular HDaudio+PulseAudio case and without additional non-public libraries in the firmware? this 16khz path could confuse most Linux users, it's mostly used by ChromeOS.

lgirdwood, 2020-01-11:
@plbossart IIUC, it looks like the 48kHz path on DMIC is also causing the issue when switching between PCM6 and 7 ?

@xiulipan @fredoh9 @zrombel it seems like we have a small hole in the CI stress test coverage (i.e. alternating playback and capture devices). We need a test that stress tests alternating capture/playback to multiple PCM devices (like the test in this report).

xiulipan, 2020-01-13:
@lgirdwood We have had a test (https://github.com/thesofproject/sof-test/blob/master/test-case/check-playback.sh with -r 3 option) for this situation after we find the issue (https://github.com/thesofproject/sof/issues/2176) in master branch and got that fixed (https://github.com/thesofproject/sof/pull/2223).

lgirdwood, 2020-01-13:
@plbossart @xiulipan ok thanks, I now have #2223 in the 1.4-stable branch

slawblauciak, 2020-01-21:
Can you please confirm whether this is fixed as of #2223 ?

lgirdwood, 2020-01-27:
I guess @dengyangchao is still on vacation, #2223 has been reported by others to fix this.

closed 2020-01-27 by lgirdwood