thesofproject/sof #2248: [BUG][BYT] Build failure when CONFIG_GDB_DEBUG=y

dragosht, 2020-01-03:
**Describe the bug**
Enabling the GDB stub for BYT currently ends up in a build error.

**To Reproduce**
    make menuconfig
    Select Debug --> GDB Stub
    make

    sof/src/debug/gdb/ringbuffer.c:12:44: error: 'SRAM_DEBUG_BASE' undeclared here (not in a function); did you mean 'MAILBOX_DEBUG_BASE'?
    volatile struct ring * const rx = (void *) SRAM_DEBUG_BASE;
                                                ^~~~~~~~~~~~~~~
                                                MAILBOX_DEBUG_BASE

paulstelian97, 2020-01-03:
An easy fix would be to copy src/platform/imx8/include/lib/memory.h lines 73-100 ("Mailbox configuration") to src/platform/baytrail/include/lib/memory.h somewhere, they're missing. Maybe I'll do it on Monday.

lgirdwood, 2020-01-06:
@xiulipan looks like a gap in the CI, i.e. we are not building with some debug options.

paulstelian97, 2020-01-06:
Guess CI again only builds on defconfig. Maybe there should be an extra build for all non-conflicting options enabled, possibly two on BYT which cannot simultaneously enable everything due to lack of IRAM?

xiulipan, 2020-01-07:
@paulstelian97 @lgirdwood will try to do some random config on daily build in CI to catch this kind of issues. It would take too much time on PR test, but I think it would be OK to do a daily scan for this.

jajanusz, 2020-01-07:
@lgirdwood @xiulipan It was discussed many times over few months that these tests should be added because some1 finds out every month it fails when enabling/disabling traces/debug options.
I'd not worry about travis taking more time to complete as it runs tests (successfully) in ~2 minutes, while there are CIs that take 20-40m to complete (when build is ok and it has to go on with python tests), so you have a lot free time here.
To not waste CI time you should add condition that builds with other flags are run only when standard build was ok.

lgirdwood, 2020-01-07:
I also dont mind if this is daily and then bisected if there is a failure, I would not want CI to take longer than it already does given the number of time I have to manually restart it for many PRs.

tlauda, 2020-03-13:
@lgirdwood Even after resolving this build issue GDB code still won't fit into text. IRAM on BYT is simple too small to have it supported along with the rest of default features. Should we somehow block this config option on BYT platform?

lgirdwood, 2020-03-13:
@tlauda lets fix the compilation part - the linker error will stay to remind users to remove features.

PR #2549 (2020-03-16): gdb: use mailbox_get_debug_base() macro
Changes ring buffer to use mailbox_get_debug_base() macro.
It points to the same memory region as previously, but it's
defined for all platforms.

Fixes #2248.
closed 2020-03-16 by dbaluta