thesofproject/sof #2251: [BUG]Missing firmware/failure to boot on Lenovo M720s with Coffee Lake CPU

robhancock, 2020-01-06:
**Describe the bug**
The latest Fedora 31 kernel update with version 5.4.7 fails to load any DSP firmware for Coffee Lake on a Lenovo M720s machine:
```
Jan 02 10:13:12 kernel: sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cfl.ri failed with error -2
Jan 02 10:13:12 kernel: sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cfl.ri failed err: -2
Jan 02 10:13:12 kernel: sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
Jan 02 10:13:12 kernel: sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2
```

Attempting to symlink the signed/sof-cnl-signed-intel.ri to sof-cfl.ri inside /lib/firmware/intel/sof allows the firmware to load, however selecting the headphone output in the GNOME sound settings results in the machine immediately hard-freezing and needing a power cycle. See the kernel output and other info in the Fedora bug report here: https://bugzilla.redhat.com/show_bug.cgi?id=1787387

**Reproduction Rate**
Every time

**Expected behavior**
Audio works, no lockup

**Impact**
Severe

**Environment**
This is using Fedora kernel 5.4.7-200.fc31 and alsa-firmware 1.2.1-3.fc31.

perexg, 2020-01-06:
Just for reference: The Fedora's alsa-firmware package contains v1.4.1 firmware files.

lgirdwood, 2020-01-06:
It looks like we are loading the wrong firmware from dmesg i.e. loading v1.1 instead of v1.4.1, but I'm suspicious here as ABI matches (and v1.1 was ABI ~3.8 IIRC).

@plbossart @ranj063 does the kernel side look OK, I see we are using MSI IRQs and the we get an spurious IRQ at firmware boot.

@robhancock can you try and disable Pulseaudio and try again (renaming it will stop it being started after boot), this should give us an idea if it's more firmware or driver related. 

```
Jan 03 17:47:43 kernel: usbcore: registered new interface driver snd-usb-audio
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
Jan 03 17:47:43 kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
Jan 03 17:47:43 kernel: EXT4-fs (nvme0n1p2): mounted filesystem with ordered data mode. Opts: (null)
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC233: line_outs=1 (0x1b/0x0/0x0/0x0/0x0) type:line
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=1 (0x14/0x0/0x0/0x0/0x0)
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:    inputs:
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:      Mic=0x1a
Jan 03 17:47:43 kernel: snd_hda_codec_realtek ehdaudio0D0:      Front Mic=0x19
[... cut ...]
Jan 03 17:47:43 kernel: input: sof-hda-dsp Front Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input14
Jan 03 17:47:43 kernel: input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input15
Jan 03 17:47:43 kernel: input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input16
Jan 03 17:47:43 kernel: input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input17
Jan 03 17:47:44 kernel: EXT4-fs (dm-3): mounted filesystem with ordered data mode. Opts: (null)
```

perexg, 2020-01-06:
> It looks like we are loading the wrong firmware from dmesg i.e. loading v1.1 instead of v1.4.1, but I'm suspicious here as ABI matches (and v1.1 was ABI ~3.8 IIRC).

How you can detect the firmware version at boot? From the hash? `1:1:0-fcf6c`

plbossart, 2020-01-06:
> It looks like we are loading the wrong firmware from dmesg i.e. loading v1.1 instead of v1.4.1, but I'm suspicious here as ABI matches (and v1.1 was ABI ~3.8 IIRC).

no, fcf6c8c416f790ad3d9f76bf75b5b528c11896c7 is v1.4.1. I don't know what the format of the 
version 1:1:0-fcf6c is supposed to mean, the sha1 is correct but the 1.1. looks like something that was never updated.

> 
> @plbossart @ranj063 does the kernel side look OK, I see we are using MSI IRQs and the we get an spurious IRQ at firmware boot.

> Jan 03 17:47:43 kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!

no, that's the usual warning that we need to remove. nothing problematic.

> 
> @jajanusz are you seeing FW version being reported correctly on CFL ?
> 
> @robhancock can you try and disable Pulseaudio and try again (renaming it will stop it being started after boot), this should give us an idea if it's more firmware or driver related.

I would recommend testing with the topic/sof-dev branch to rule out any backport issues or missing patches.

robhancocksed, 2020-01-06:
Some updates I added to the RH Bugzilla entry:

I confirmed that snd-intel-dspcfg.dsp_driver=1 on the kernel command line restores the working behavior, so that is a workaround for now. Also, the update to alsa-firmware seems to allow the firmware to load properly now with the default options. But the lockup issue persists in that configuration.

It seems like the output from "aplay -L" is rather thin in the SOF mode:

```
sysdefault
    Default Audio Device
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Default ALSA Output (currently PulseAudio Sound Server)
sysdefault:CARD=sofhdadsp
    sof-hda-dsp, 
    Default Audio Device
```
With the working setup there are a lot more outputs listed:
```
sysdefault
    Default Audio Device
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Default ALSA Output (currently PulseAudio Sound Server)
sysdefault:CARD=PCH
    HDA Intel PCH, ALC233 Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    Front speakers
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
hdmi:CARD=PCH,DEV=0
    HDA Intel PCH, HDMI 0
    HDMI Audio Output
hdmi:CARD=PCH,DEV=1
    HDA Intel PCH, HDMI 1
    HDMI Audio Output
hdmi:CARD=PCH,DEV=2
    HDA Intel PCH, HDMI 2
    HDMI Audio Output
hdmi:CARD=PCH,DEV=3
    HDA Intel PCH, HDMI 3
    HDMI Audio Output
hdmi:CARD=PCH,DEV=4
    HDA Intel PCH, HDMI 4
    HDMI Audio Output
```
Also, I tried having "journalctl -f" open while selecting the headphone output in GNOME sound settings after booting in SOF mode. I saw these kernel messages (which also happened previously during boot):
```
sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
sof-audio-pci 0000:00:1f.3: firmware boot complete
```
This from gsd-media-keys:
```
Unable to get default sink
gvc_mixer_card_get_index: assertion 'GVC_IS_MIXER_CARD (card)' failed
```
gnome-settings-daemon also complained with that same assertion.

robhancocksed, 2020-01-06:
Also, given that the proper firmware symlink wasn't even in place for Coffee Lake (and some other CPUs) in the upstream firmware repositories, it seems hard to believe that this setup was ever fully tested on this hardware. How did this code get merged into the upstream kernel in such a state? At this point it's a clear regression. Maybe defaulting to dsp_driver=1 for now is the best interim solution.

lgirdwood, 2020-01-06:
@robhancock fwiw, theres still some discussion on upstreaming the firmware binaries so your link should work fine once the binaries are merged. There is currently a FW repo here (on SOF project), feel free to use but we are rebasing it frequently atm until upstream discussion is aligned.

robhancocksed, 2020-01-06:
@lgirdwood , my point was that since the firmware symlink wasn't in place to allow the driver to even load, it's obvious that the rest of the code was never tested on this hardware, at least not in the configuration that was merged upstream. And indeed, once the firmware does get loaded it doesn't work. It sure looks like Intel dropped the ball on this - as was mentioned elsewhere on Bugzilla, 
'it seems that the word "working" has different meaning for Intel and others. They claimed to have the Linux support a half of year ago.'

plbossart, 2020-01-06:
@robhancock we've contributed patches to make the life of distros simpler and while things are not perfect we've not dropped the ball. I don't help to use sarcasm or diss the work of people who are trying to fix things.

perexg, 2020-01-06:
@plbossart : I am really sorry, that were my words. I am a bit tired to get all components together (fighting with the driver/firmware incompatibilities, topologies, UCM, pulseaudio to get things working properly in last months). My point is that upstream should be first, then you can say that Linux has full support for the hardware. Thank you for your work.

jajanusz, 2020-01-07:
@lgirdwood I've loaded binary mentioned here on production platform, however you are right that version string is weird:
```
[    5.638126] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[    5.644556] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[    5.644573] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[    5.644685] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    5.644689] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:9:0
[    5.644691] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel 
```
 I'll check if everything is ok on fw side with versioning, maybe some tags messed up git tag commands used in buildsystem.

jajanusz, 2020-01-07:
@lgirdwood version string got fixed in #2255 , just version from tag was wrong, commit hash was ok , so you can assume that in mentioned log the binary with version 1.4.1 was used

lgirdwood, 2020-01-07:
Thanks @jajanusz now merged #2255

slawblauciak, 2020-02-18:
Is this still an issue after that PR's been merged?

closed 2020-02-18 by lgirdwood

kreaninw, 2022-05-15:
I know this bug is closed. But I want to update the situation regarding SOF driver on the Coffee Lake CPU, **i5-8250U** on Acer Swift SF514-52T  in my case.

I am running on Fedora 36, my packages are:
```
alsa-lib-1.2.6.1-4.fc36.x86_64
alsa-utils-1.2.6-2.fc36.x86_64
alsa-sof-firmware-2.0-3.fc36.noarch
```

When I enabled the SOF driver with `options snd-intel-dspcfg dsp_driver=3` in the `/etc/modprobe.d/snd-intel.conf` file. Then running `sudo dracut --force` and reboot.

After rebooted, my aplay -l shows:

```
$ aplay -l
aplay: device_list:274: no soundcards found...
```
Basically, nothing in my system is detected by the SOF driver at all.

My full **[alsa-info](http://[alsa-project.org/db/?f=8542e92049210a9185600e2d049aee38beb09add](http://alsa-project.org/db/?f=8542e92049210a9185600e2d049aee38beb09add))**.

If I use `options snd-intel-dspcfg dsp_driver=2` (SST driver - default setting on Fedora 36), my speakers, DMIC, and bluetooth headsets work. But **HDMI doesn't work**.

If I use `options snd-intel-dspcfg dsp_driver=1` (HDaudio legacy driver), my speakers and HDMI audio work (I didn't test the bluetooth headsets). But **DMIC doesn't work** which is the [expected behavior](https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#:~:text=This%20will%20typically%20work%20for%20speakers%20and%20headphones/headsets%2C%20but%20will%20not%20allow%20DMIC%20capture.).

Is there any hope for the sound on this platform to be fully functional on Linux?

Thanks.