thesofproject/sof #2256: [FEATURE] Stub loader for Kabby and Skylake platforms

cujomalainey, 2020-01-08:
**Is your feature request related to a problem? Please describe.**
Kabby and Skylake platforms have immutable keys for loading DSP firmware which prevents SOF from being loaded without having Intel sign the blobs.

**Describe the solution you'd like**
The alternative is have a stub loaded that is signed with the intel key and have that stub load OTC signed builds through the existing communication mechanisms with the DSP. Ideally this would be able to load SOF builds with minimal changes to the build format. There will obviously be changes needed since the bootloader will handle some init code that the firmware will not need to rerun. 

**Describe alternatives you've considered**
Alternative is to have Intel sign all builds, this prevents upgrades without having intel sign every build produced by any non-intel party. This makes development on those platforms basically impossible externally.

**Additional context**
#1899

cujomalainey, 2020-01-08:
@dgreid

jajanusz, 2020-01-08:
@lgirdwood @mmaka1 Doesn't it break security at all?

lgirdwood, 2020-01-08:
@jajanusz I don't think so, IIUC it means that the stub will only boot code signed with Google Private key, in a similar way we have today with the OEM key manifest on APL+. It also wont work on any system with secure OS boot enabled in BIOS. I would say it's only useful on Chrome and Linux/Windows booting with secure boot disabled (where there is no security anyway)

plbossart, 2020-01-08:
and Chromebooks use the community key anyways starting with GLK, so it'd be the same level of security (essentially guaranteed by the protected filesystem, not by firmware signature).

cujomalainey, 2020-01-08:
That is correct, we use the open source keys since we rely on our fs verification for firmware integrity. Once the device is in dev mode users can create their own firmware signed with the open source keys.

chrysophylax, 2020-01-21:
As a Linux end-user, I'd gladly rather take no secure boot and have working sound :+1:

lgirdwood, 2020-01-22:
@cujomalainey @dgreid @chrysophylax I'm working on some prototype code now, will test on APL.

JannF, 2020-09-22:
Is there any progress besides the closed referenced ticket?

cujomalainey, 2020-09-22:
@JannF this is also closed based no the resolution in #1899

closed 2020-09-22 by cujomalainey