thesofproject/sof #2264: [BUG] Fedora 31 (kernel 5.4.7) SOF firmware load fails on Yoga C940 (i7-1065G7 Ice Lake)

pfps, 2020-01-08:
**Describe the bug**

Fedora 31 has SOF audo enabled, but the firmware fails to load on my Yoga C940, resulting in no audio devices being available.  The laptop has a digital microphone array, several HDA speakers, and headphone jack with microphone input.

**To Reproduce**
Boot into current version of Fedora 31.

**Reproduction Rate**
Always

**Expected behavior**
The firmware should load and at least the digital mic, regular soundbar speakers, and headphone/microphone jack should be available.

**Impact**
Not having sound is a real problem.
Reverting to hda via
  echo "options snd-intel-dspcfg dsp_driver=1" > /etc/modprobe.d/alsa.conf
brings back the situation from kernel 5.3 (headphone jack fully working, only one pair of speakers working, no microphone).

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
This is from fedora, so no commit hashes, I think
    * Kernel:  5.4.7-200.fc31.x86_64+debug
   * Alsa Firmware:  1.2.1-3.fc31
2) Name of the topology file
    * Topology:  I'm not sure what to provide for this.
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo C940 laptop with Ice Lake processor
[boot.txt](https://github.com/thesofproject/sof/files/4037258/boot.txt)

**Screenshots or console output**
The relevant messages are probably 
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
Jan 08 06:08:55 idefix kernel: sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
I have attached a full log at well.

I don't have sof logs as I'm now running with the workaround in place.  I could undo the workaround and reboot if the sof logs are needed.

jajanusz, 2020-01-09:
> I don't have sof logs as I'm now running with the workaround in place. I could undo the workaround and reboot if the sof logs are needed.

FW logs won't help because FW didn't even start at that stage. It failed on phase when ROM verifies FW image. Is it vanilla FW that comes in Fedora 31 ? Can you paste here fw binary that you have/had in /lib/firmware/intel/sof* ? (if I'm correct the file should be named sof-icl.ri for your model)

lgirdwood, 2020-01-09:
@pfps are you able to try the latest kernel topic/sof-dev branch ? @plbossart @ranj063 fyi

pfps, 2020-01-09:
Probably, but I'll need some instructions.

lgirdwood, 2020-01-09:
@pfps oh, sorry I'm no Fedora kernel expert, lets see what the others say before you have to try this. The ROM error may be indicative we are missing a kernel patch in the Fedora kernel (since kernel programs HW to boot ROM code). @perexg how up to date is this kernel with sof-dev ?

pfps, 2020-01-09:
@jajanusz  Maybe.   I had been running some test Fedora kernels before, but I thought everything was upgraded to stock.  However, I ended up with /lib/firmware/intel/sof-icl.ri and /lib/firmware/intel/sof-icl-v1.4.1.ri - the latter is from the earlier test setup.
I've attached both.  (I had to attach renamed copies.)
[sof-icl-v1.4.1.ri.txt](https://github.com/thesofproject/sof/files/4041096/sof-icl-v1.4.1.ri.txt)
[sof-icl.ri.txt](https://github.com/thesofproject/sof/files/4041125/sof-icl.ri.txt)

There are also a bunch of topology files in /lib/firmware/intel/sof-tplg.

I'll see what happens when I remove the 1.4.1 files

perexg, 2020-01-09:
We are using the properly signed 1.4.1 firmware binaries (from the 1.4.1 release tag from this source - github) and the driver code from 5.5 and partially from 5.6 kernels. So basically, it should be latest and I don't think that there are some fixes in the firmware loader code. The firmware boots on other machines properly.

The used firmware files: ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.4.1.tar.bz2

perexg, 2020-01-09:
It might be that ICL firmware files have the trouble with the signing again.

pfps, 2020-01-09:
Stay tuned.  I just did a dnf update and an alsa-firmware update is being installed - version 1.2.1-4.fc31 .   Maybe the last dnf update picked up the new kernel but not the new firmware.

perexg, 2020-01-09:
You may compare files with the source:

  https://github.com/thesofproject/sof/releases
  https://github.com/thesofproject/sof/releases/download/v1.4.1/sof-icl-v1.4.1.ri

perexg, 2020-01-09:
Just checked and the sha256sum matches... ba3b8945dd4efe6fa4738b0fb66333a2e5b63d249eab0bae498faeb7fa90e8ce

lgirdwood, 2020-01-09:
@lbetlej  @mmaka1 any ideas on the ROM error in code loader on ICL ?

lgirdwood, 2020-01-09:
@tlauda @jajanusz have you guys seen this ROM code ?

plbossart, 2020-01-09:
I just checked that this can't possibly due to a timeout value being too low. The timeout is set at 3s, so it's for sure this comes from a bad configuration, not something that needs more time (as we've seen in the past).

jajanusz, 2020-01-09:
> @tlauda @jajanusz have you guys seen this ROM code ?

It means `ADSP CSE VALIDATION FAILED`.
It's not some timeout, CSE says that it validated image and image is invalid.
Why? Hard to say with current info, maybe mismatch in bios+CSME combo, FW has to be signed for CSME (engine itself and keys that are there), if you change there is possibility that differently signed FW is needed.

Is BIOS there stock? Isn't it some debug device?
Did it work before and stopped working after bios/os update?
If you have Windows there, does it work on it?

pfps, 2020-01-10:
The Lenovo C940 is a laptop.  The BIOS is whatever Lenovo put on it.  The report came from a debug kernel, if that matters, but I'm also getting the same error on the non-debug kernel.

This is the first time I've even got this far under Linux, so I don't know if there is some new problem with the kernel.  I did run some custom kernels a while ago, but as far as I can tell I'm running standard Fedora 31 now.

I have run the machine under Windows and I think the entire audio system works there.  If not there would have been lots of howls from other users of the C940, which is a flagship Lenovo product.  I still have the original Windows NVME M.2 drive, so I can run Windows again, but not until I get to where the drive is, which won't happen until Jan 18.

perexg, 2020-01-10:
It's a bit dup of: https://github.com/thesofproject/sof/issues/2134

@pfps: If I understand correctly, you was able to use the latest 1.4.1 ICL firmware - https://github.com/thesofproject/sof/issues/2134#issuecomment-563895508 . Or not?

lgirdwood, 2020-01-10:
@jajanusz @lbetlej IIUC, we may have two CSME BIOS version on production ICL that require different audio FW binaries now (same code but signed differently )?

plbossart, 2020-01-10:
> @jajanusz @lbetlej IIUC, we may now have two CSME BIOS versions on production ICL that require different audio FW binaries now (same DSP code but signed differently )?

that would be a huge monkey wrench if we can't detect which CSME version is being used...

pfps, 2020-01-10:
@perexg Aaah, I had forgotten that I was able to get the firmware loaded once. I didn't have time then to check out what happened with sound though.
But I was running an unsigned kernel so I had secure boot turned off. (I'm still amazed that Lenovo lets me turn of secure boot - very refreshing.)

So I turned off secure boot and ... the firmware loads!

Jan 10 10:00:29 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Jan 10 10:00:29 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Jan 10 10:00:29 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete

But something is still wrong (besides signing issues I guess).  No soundcards are available (aplay -l and arecord -l) and pulse doesn't show any inputs or outputs.  Should I open a new bug for that?  What information would be best to provide?

paulstelian97, 2020-01-10:
You'll need to provide what topology file you are using, kernel version, firmware version (the logs from your last reply help). I'd argue there is a mismatch between topology and firmware that could cause a lack of soundcards. Feel free to open new bug in that case. If you see a topology load failure in the logs do provide it as well.

perexg, 2020-01-10:
@pfps : Show the boot log again.

pfps, 2020-01-10:
@prexg : I did a new boot, and now things are even better.  I don't know why the last boot ended up in a state with no soundcards.  Here is some current state

idefix testing> aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
idefix testing> arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Integration with pulseaudio isn't perfect.  Pulse came up in profile Off and I had to switch to Play HiFi quality Music.  Pulse switches back to Off too often.  Other defaults are not great either.   Where would this be fixed?

But there is a digital mic available.   I can record with arecord and play with aplay (and other programs).

So, this can be counted as a partial success, as long as secure boot can be turned off.  This doesn't address the C940 and C930 problem that several speakers don't work because their amps are not turned on (but I didn't expect it to).

Anyway, I've attached the boot log just in case there is something to be glreaned from it.
[boot.txt](https://github.com/thesofproject/sof/files/4047548/boot.txt)

pfps, 2020-01-10:
So I rebooted this morning and most everything was working!  I could record using the digital mic and play on the main speakers.  Integration with PulseAudio wasn't perfect but usable.  I've attached the boot log.
[boot.txt](https://github.com/thesofproject/sof/files/4048032/boot.txt)
A new Fedora kernel came out so I upgraded and rebooted.  The firmware loaded but something later failed.  It looks as if the first report of problems is 
Jan 10 16:16:43 idefix kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to init link iDisp1: -517
I've attached the boot log for the bad situation.
[bood-bad.txt](https://github.com/thesofproject/sof/files/4048052/bood-bad.txt)
I'm not sure what is causing the difference - the updated kernel or some timing issue.  I'll try a couple of experiments.

pfps, 2020-01-10:
I rebooted five or six times into the kernel that was working this morning (5.4.7).  Each time I got the error about failing to init iDisp1.  I'm not sure whether this is due to some other change in the upgrade (but I didn't see anything that looked relevant) or whether there is some timing or other environmental issue that changed behaviour.

lgirdwood, 2020-01-10:
@kv2019i any inputs on the HDMI error ?
```
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Jan 10 16:16:42 idefix kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
...
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Jan 10 16:16:43 idefix kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to init link iDisp1: -517
```

perexg, 2020-01-10:
Latest pulseaudio have UCM fixes from me. I need to manage the pulseaudio update for Fedora (I am not the maintainer for this package).

Speakers: https://bugzilla.kernel.org/show_bug.cgi?id=205755

lgirdwood, 2020-01-11:
@perexg I'm also assuming your next signed kernel will have the updates in the unsigned kernel that's working for @pfps  ? The OS boot code signing should have no impact on DSP capability or DSP code signing (two different code signing mechanisms).

pfps, 2020-01-11:
@lgirdwood  Hmm, that leaves me even more confused. 

But just now I rebooted (after making some minor changes) and things are working again.   This is with a signed kernel - Fedora 5.4.7-200.fc31.x86_64+debug - but with secure boot off.
 I'll try to figure out what makes the difference.  I'm going to keep kernel logs for my attempts.

pfps, 2020-01-11:
I tried rebooting quite a few times this afternoon.  I didn't find any method that guaranteed a success.  The last seven times I just tried doing the same thing - restart from the XFCE logout dialog box without touching the sound system beyond checking whether it was working using aplay -l.  Five restarts were successes and two were failures.  Each time I recorded the boot journal using journalctl -b.

The only differences between the restarts that I can think of being possibly relevant is the time it took me to select the kernel from the list of kernels.  I kept track of this time, but I couldn't see any relationship between it and successfully starting the sound system.

Each failure appears to be signalled by the message
skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to init link iDisp1: -517
right after
sof-audio-pci 0000:00:1f.3: firmware boot complete

I haven't bothered to upload all the journal logs, but I can do so if requested.

lgirdwood, 2020-01-13:
@pfps thanks for this testing. Sorry if your confused a little on the code signing, the OS and DSP FW are two separate signing systems, they _should_ have no impact on each other.  
@kv2019i fyi, are we missing one of your HDMI updates here or any comments here? It looks like we may have a race between display and audio init ?

kv2019i, 2020-01-13:
@lgirdwood wrote:
> @kv2019i any inputs on the HDMI error ?
>
> ```
> Jan 10 16:16:43 idefix kernel: sof-audio-pci 0000:00:1f.3: error: codec #2 probe error, ret: -5

This looks like a known bug I fixed in i915 driver and merged to 5.5-rc kernel. If  the tested Fedore kernel (5.4.8-200.fc31.x86_64+debug), does not have these patches, this is probably the reason.

87c1694533c9 drm/i915: save AUD_FREQ_CNTRL state at audio domain suspend
1580d3cdddbb drm/i915: Fix audio power up sequence for gen10+ display

If not, then this is potentially a new issue with gen10 display (Icelake and Geminilake). FYI @pfps and @perexg

perexg, 2020-01-13:
@kv2019i : Thanks for the comment. The video driver updates has not been backported yet. I will do this ASAP.

pfps, 2020-01-13:
@perexg Given that SOF appears to be out of Fedora 31 for now, is Rawhide suitable for testing SOF (and the fixes for the other speakers)?
@lgirdwood No problem.  I'm interesting in getting the microphone array on my machine to work and have some time now to put into getting to do so.

lgirdwood, 2020-01-14:
@pfps @perexg we are now validation testing v1.4.2 should be able to sign and release it next week.

lgirdwood, 2020-01-27:
@pfps did the v1.4.2 release work for you ?
https://github.com/thesofproject/sof/releases/tag/v1.4.2
Sorry, we don't yet have binaries packaged yet, still under discussion.

benoitg, 2020-01-27:
ftp://ftp.alsa-project.org/pub/misc/sof/sof-firmware-1.4.2.tar.bz2 did solve the problem for me on Ubuntu (kernel 5.5.0-050500rc7-generic)

pfps, 2020-01-27:
@lgirdwood  I'm still trying to figure out an easy way to test.

pfps, 2020-02-01:
@lgirdwood I think I'm now running with 1.4.2.   How can I check?

What I am running is Fedora 31 kernel 5.4.14-200.fc31.x86_64 and alsa-firmware 1.2.1.5.fc31, which according to https://bugzilla.redhat.com/show_bug.cgi?id=1772498#c100 has the 1.4.2 firmware.

Some boots are successful, but sometimes I see
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Feb 01 15:02:07 idefix kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Feb 01 15:02:07 idefix kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to init link iDisp1: -517
and no sound cards are available.

I'm not sure what is causing this problem.

lgirdwood, 2020-02-03:
@pfps sorry it still looks like 1.4.1 since it had a bug in getting tags from git
```: version 1:1:0-fcf6c```. Can you copy https://github.com/thesofproject/sof/releases/download/v1.4.2/sof-icl-v1.4.2.ri to /lib/firmware/intel/sof (and overwrite the current sof-icl.ri)

slawblauciak, 2020-02-18:
Can you make sure your BIOS is up to date?

lgirdwood, 2020-02-18:
@slawblauciak not following your BIOS comment ? v1.4.2 is booting for @pfps now, issue seems to be related to kernel ?

slawblauciak, 2020-02-18:
Yes, you're right, the comments were a bit hard to follow.

pfps, 2020-02-18:
@slawblauciak I'll try updating my BIOS soon.  It's been a while and I should do this anyway, but I have to switch disks to update the BIOS so I don't do it very often.

lgirdwood, 2020-02-18:
> @slawblauciak I'll try updating my BIOS soon. It's been a while and I should do this anyway, but I have to switch disks to update the BIOS so I don't do it very often.

@pfps please hold off the BIOS update for the moment, can you try v1.4.2 and latest kernel.

perexg, 2020-02-18:
I think that this bug may be closed. From the @pfps comments in Fedora's bug https://bugzilla.redhat.com/show_bug.cgi?id=1772498 the 1.4.2 firmware loads fine now with the i915 patch https://lore.kernel.org/r/20200206200223.7715-3-kai.vehmanen@linux.intel.com .

The other major issue we're facing right now with the 5.5 kernel: https://github.com/thesofproject/linux/issues/1747

closed 2020-02-18 by lgirdwood

danielcherubini, 2020-02-24:
@pfps  how are you updating bios? windows dualboot?

pfps, 2020-02-24:
@danielcherubini I have two SSDs, one with windows and one with linux.  To update the bios I just swap them.  It does look, however, that Lenovo now has a way to update the BIOS from a USB stick, but I haven't used that method.  I just ran across some working that looked like this was possible when I grabbed the last BIOS update.

danielcherubini, 2020-02-24:
@pfps I'm looking now and i don't see the USB method you're talking about.. can you enlighten?

pfps, 2020-02-24:
@danielcherubini In https://support.lenovo.com/us/en/downloads/ds541233 "UEFI shell environment" second looks like it is providing a way to update the BIOS without using Windows.

danielcherubini, 2020-02-24:
@pfps last message .. sorry to derail... i only see the .exe and a .txt file.. i think the instructions at the bottom might be generated somewhere