thesofproject/sof #2265: [BUG][CNL 1.4.1] sof-hda-generic-4ch.tplg

hsanjuan, 2020-01-09:
I get lots of errors from sof-audio-pci:

```
[   12.390543] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   12.390577] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   12.390688] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   12.398963] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   12.416209] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   12.416211] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   12.522117] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   12.522119] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   12.608703] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[   12.608705] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
[   12.608706] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   12.608943] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   12.612688] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:10:0
[   12.612694] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   12.613475] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   12.613477] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   12.620361] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   12.620364] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   12.623128] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   12.623133] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   12.624102] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   12.624104] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   12.624106] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   12.624108] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   12.624110] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   12.624111] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   12.624723] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   12.624725] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   12.625689] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   12.625691] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   12.627872] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   12.641020] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   12.714906] input: sof-skl_hda_card Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[   12.714952] input: sof-skl_hda_card Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[   12.715096] input: sof-skl_hda_card HDMI/DP, pcm=11 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[   12.715135] input: sof-skl_hda_card HDMI/DP, pcm=12 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[   12.715172] input: sof-skl_hda_card HDMI/DP, pcm=13 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[   13.184635] uvcvideo 1-8:1.2: Entity type for entity Microsoft Extended Controls Uni was not initialized!
[   21.326450] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   21.424195] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   32.567157] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   32.666353] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   39.932183] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   40.031771] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   74.437227] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[   74.437233] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[   74.438191] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[   74.438195] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[   80.144326] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   83.224187] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   83.224233] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   83.224240] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   83.224242] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -110
[   83.735589] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   83.735605] sof-audio-pci 0000:00:1f.3: error: status = 0x001c003c panic = 0x00000003
[   83.735615] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0x00000000 rirb 00
[   83.735618] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000006
[   83.735625] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040000 mask 0x00000000
[   83.735628] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp3
[   83.735631] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -110
[   84.247592] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   84.247602] sof-audio-pci 0000:00:1f.3: error: status = 0x001c003c panic = 0x00000003
[   84.247610] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0x00000000 rirb 00
[   84.247612] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000006
[   84.247619] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040000 mask 0x00000000
[   84.247621] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp3
[   84.247624] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -110
[... 48 lines cut ...]
[   87.831603] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -110
[   87.831658] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -16
[   87.856831] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -16
[   87.856893] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -16
[   87.856932] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -16
[   87.856966] sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -16
[line repeats...]
```

Pulseaudio seems to hang and fails to restart of retry:

```
I: [pulseaudio] (alsa-lib)pcm_hw.c: SNDRV_PCM_IOCTL_HW_PARAMS failed (-16)
I: [pulseaudio] (alsa-lib)pcm_hw.c: SNDRV_PCM_IOCTL_HW_PARAMS failed (-16)
I: [pulseaudio] (alsa-lib)pcm_hw.c: SNDRV_PCM_IOCTL_HW_PARAMS failed (-16)
I: [pulseaudio] (alsa-lib)pcm_hw.c: SNDRV_PCM_IOCTL_HW_PARAMS failed (-16)
I: [pulseaudio] (alsa-lib)pcm_hw.c: SNDRV_PCM_IOCTL_HW_PARAMS failed (-16)
I: [pulseaudio] alsa-util.c: snd_pcm_hw_params failed: Device or resource busy
I: [pulseaudio] (alsa-lib)conf.c: Unknown parameter 1
I: [pulseaudio] (alsa-lib)conf.c: Parse arguments error: No such file or directory
I: [pulseaudio] (alsa-lib)pcm.c: Unknown PCM plug:hw:sofsklhdacard,5
I: [pulseaudio] alsa-util.c: Error opening PCM device plug:hw:sofsklhdacard,5: No such file or directory
```

Can anyone help? Is this normal? I built the topology files from the 1.4.1 release tag.

lgirdwood, 2020-01-09:
@hsanjuan this is not normal. The only thing I can see that could be an issues is the different ABI versions for some ingredients.
```
[   12.608703] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[   12.608705] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:10:0
[   12.608706] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   12.608943] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   12.612688] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:10:0
```
Are you able to try the latest linux topic/sof-dev branch HEAD alongside topology from SOF v1.4.1 tag ?

hsanjuan, 2020-01-09:
Thanks, some more information from my side.

I'm now on Kernel 5.4.8. Also had problem on 5.4.7, but things worked better on 5.3.x (and maybe early 5.4.x). The Kernel 5.3 with sof v1.3 worked (sound system started fine at least), but I was also using older alsa (1.1.5 instead of 1.2.1.2) and pulseaudio (11.1 instead of 13.0).

Unfortunately it is likely not possible to me to test with special kernel branches (I am very time constrained right now). Given it's a bit older, eventually the Kernel ABI will catch up with the Firmware one right? Do you think there have been changes that could cause this?

The error messages seem to be  triggered when I restart pulseaudio. Alsa tools (alsamixer, aplay -l, arecord -l) seem to detect everything alright though. I have a Thinkpad X1 Carbon 7th Gen. but I have not seen other people with very similar configurations complain about this, so I'm wondering if I am using very recent versions of everything.

I use the sof-hda-generic-4ch topology and the rest of the setup in https://gist.githubusercontent.com/hamidzr/dd81e429dc86f4327ded7a2030e7d7d9/raw/36fa50d847dd3d914146347a1660c3a93747791c/sof-audio-setup-carbonx1.sh

lgirdwood, 2020-01-09:
@hsanjuan kernel ABI will catch up with FW (upstreaming means official kernel releases are usually behind though from time to time), although the intention is that MINOR ABI updates wont break anything but rather limit some functionality.

It's difficult to test all ABI combinations, since we have kernel, topology and FW - but usually CI is validating everything from HEAD (so ABI is usually at most equal or one MINOR difference).

I'm suspicious we are missing a fix in the kernel, I assume you are using vanilla upstream or distro versions ? @plbossart @ranj063 fyi

hsanjuan, 2020-01-09:
> I assume you are using vanilla upstream or distro versions

Yes, the Opensuse stable kernels (maybe I can run a nightly kernel build too, I'll report back).

hsanjuan, 2020-01-09:
Ok, I am now testing with `5.5.0-rc5-1.g288dac0-default #1 SMP Mon Jan 6 05:28:20 UTC 2020 (288dac0)` which is more recent.

* The first pulseaudio autostart still seems to hang: need to kill it and restart it (I need to investigate more what is happening).
* After that, it starts fine but the Cannon Point-LP High Definition Audio Controller profile is set to `Off`. The `Play Hifi quality music` says "Unavailable"
* However it can be selected and it plays through the speakers just fine!
* The normal microphone appears as `unplugged`, but it works nevertheless
* I don't see unplugged HDMI outputs as I used to.

* dmesg shows no sof errors after multiple pulseaudio restarts (win!):

```
[    3.082887] software IO TLB: mapped [mem 0x62edf000-0x66edf000] (64MB)
[   13.457368] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[   13.460710] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   13.461055] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   13.461059] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   13.461179] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   13.498481] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   13.508691] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   13.508693] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   13.581181] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   13.581183] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   13.666466] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[   13.666468] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[   13.666674] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   13.734210] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:11:0
[   13.734214] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   13.735012] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   13.735014] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   13.735816] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   13.735817] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   13.737204] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   13.737205] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   13.738091] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   13.738092] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   13.738093] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   13.738095] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   13.738096] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   13.738097] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   13.742561] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   13.742562] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   13.743453] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   13.743455] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   13.744212] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   13.751554] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   13.861398] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[   13.861480] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[   13.863279] input: sof-hda-dsp HDMI/DP, pcm=18 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[   13.865959] input: sof-hda-dsp HDMI/DP, pcm=19 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[   13.868137] input: sof-hda-dsp HDMI/DP, pcm=20 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
```

I still get some pulseaudio errors (seem related to the missing HDMI outputs):

```
I: [pulseaudio] (alsa-lib)conf.c: Unknown parameter 1
I: [pulseaudio] (alsa-lib)conf.c: Parse arguments error: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hw:sofhdadsp,5
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hw:sofhdadsp,5: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] alsa-ucm.c: UCM jack HDMI/DP,pcm=5 has_control=0
I: [pulseaudio] (alsa-lib)conf.c: Unknown parameter 1
I: [pulseaudio] (alsa-lib)conf.c: Parse arguments error: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hw:sofhdadsp,4
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hw:sofhdadsp,4: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] alsa-ucm.c: UCM jack HDMI/DP,pcm=4 has_control=0
I: [pulseaudio] (alsa-lib)conf.c: Unknown parameter 1
[... cut ...]
I: [pulseaudio] (alsa-lib)conf.c: Unknown parameter 1
I: [pulseaudio] (alsa-lib)conf.c: Parse arguments error: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hw:sofhdadsp,0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hw:sofhdadsp,0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] alsa-ucm.c: UCM jack Mic has_control=1
```

But from the looks of it, it seems my problems now fall in alsa/pulseaudio land?

hsanjuan, 2020-01-09:
>     * The first pulseaudio autostart still seems to hang: need to kill it and restart it (I need to investigate more what is happening).

This was me starting pulseaudio and systemd trying to do the same upon request from RealtimeKit. Now pulseaudio starts fine first time.

lgirdwood, 2020-01-10:
@kv2019i @juimonen any comments on the pulseaudio / UCM errors ? Are our updates all merged ?
@hsanjuan this looks better but not perfect, the other thing I would do it check out SOF git and rebuild the topologies to the latest version.
```
git clone https://github.com/thesofproject/sof.git
cd sof.git
./scripts/build-tools.sh
```
You should then be able to copy all the new *.tplg files to ```/lib/firmware/intel/sof-tplg/```

hsanjuan, 2020-01-10:
>  the other thing I would do it check out SOF git and rebuild the topologies to the latest version.

Actually, forgot to mention that I was testing with topology built from master.

> copy all the new *.tplg

Are ALL needed? I just copied the  sof-hda-generic-4ch.tplg and set the sof-hda-generic.tplg symlink to it.

lgirdwood, 2020-01-10:
@hsanjuan ok, good - you have updated the correct file.
@kv2019i @juimonen any clues on the pulseaudio and UCM stuff ?
@kv2019i does the HDMI unplugged event need any of your latest HDA/HDMI kernel patches ?

kv2019i, 2020-01-10:
@lgirdwood wrote:
> @kv2019i @juimonen any clues on the pulseaudio and UCM stuff ?
> @kv2019i does the HDMI unplugged event need any of your latest HDA/HDMI kernel patches ?

No extra patches needed for HDMI, 5.5-rc5 has also the common HDMI driver support in.

We'd need a bit more data to try to see what is happening. @hsanjuan could you post the UCM file you have installed for sof-hda-dsp, the kernel config (to see which HDA codec driver you have built with) and a bit more of the pulseaudio log (some of the errors are harmless, Pulseaudio tries to discover what mixer entities there are, but it would help to see the complete logs to verify PA can open the critical nodes)?

hsanjuan, 2020-01-11:
Thank you so much for helping me through! I very much appreciate it.

> could you post the UCM file you have installed for sof-hda-dsp

I have been using the alsa provided version https://github.com/alsa-project/alsa-ucm-conf/tree/v1.2.1.2/ucm2/sof-hda-dsp (in the 5.4.x I had to rename it back to `sof-skl_hda_card/` but now the devices are correctly named `sof-hda-dsp`). It seemed to me that this was very much based on the original lenovo-provided UCM file (which is at https://gist.githubusercontent.com/hamidzr/dd81e429dc86f4327ded7a2030e7d7d9/raw/36fa50d847dd3d914146347a1660c3a93747791c/sof-audio-setup-carbonx1.sh), with improvements.

> the kernel config (to see which HDA codec driver you have built with)

Regarding kernel configuration, OpenSUSE HEAD builds seems to use this one:

[default.txt](https://github.com/thesofproject/sof/files/4049160/default.txt)

I picked that file from https://build.opensuse.org/package/show/Kernel%3AHEAD/kernel-default. The sources seem to come from https://github.com/openSUSE/kernel

Regarding pulseaudio logs, this is the full startup log:

<details>
  <summary>pulseaudio log (click to expand) </summary>
<pre>
[10:39:08] ~ $ pulseaudio -vvvvv
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: RealtimeKit worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 13.0-rebootstrapped
D: [pulseaudio] main.c: Compilation host: x86_64-suse-linux-gnu
D: [pulseaudio] main.c: Compilation CFLAGS: -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type  -g -fPIE -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
D: [pulseaudio] main.c: Running on host: Linux x86_64 5.5.0-rc5-1.g288dac0-default #1 SMP Mon Jan 6 05:28:20 UTC 2020 (288dac0)
D: [pulseaudio] main.c: Found 4 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Running from build tree: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is 795aa2e781f24353a3fc7915960d760e.
I: [pulseaudio] main.c: Session ID is 1.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/hector/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib64/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
D: [pulseaudio] memblock.c: Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
I: [pulseaudio] cpu-x86.c: CPU flags: CMOV MMX SSE SSE2 SSE3 SSSE3 SSE4_1 SSE4_2 
I: [pulseaudio] svolume_mmx.c: Initialising MMX optimized volume functions.
I: [pulseaudio] remap_mmx.c: Initialising MMX optimized remappers.
I: [pulseaudio] svolume_sse.c: Initialising SSE2 optimized volume functions.
I: [pulseaudio] remap_sse.c: Initialising SSE2 optimized remappers.
I: [pulseaudio] sconv_sse.c: Initialising SSE2 optimized conversions.
I: [pulseaudio] svolume_orc.c: Initialising ORC optimized volume functions.
D: [pulseaudio] database-tdb.c: Opened TDB database '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-device-volumes.tdb'
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-stream-volumes.tdb'
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry0
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry2
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry3
[... cut ...]
I: [pulseaudio] module-suspend-on-idle.c: Sink auto_null idle for too long, suspending ...
D: [pulseaudio] sink.c: auto_null: suspend_cause: (none) -> IDLE
D: [pulseaudio] sink.c: auto_null: state: IDLE -> SUSPENDED
D: [pulseaudio] source.c: auto_null.monitor: suspend_cause: (none) -> IDLE
D: [pulseaudio] source.c: auto_null.monitor: state: IDLE -> SUSPENDED
D: [pulseaudio] core.c: Hmm, no streams around, trying to vacuum.
</pre>
</details>

(My pulseaudio daemon config is vanilla and should not have anything manually introduced).

kv2019i, 2020-01-13:
@hsanjuan wrote:
> I have been using the alsa-provided version https://github.com/alsa-project/alsa-ucm-conf/tree/v1.2.1.2/ucm2/sof-hda-dsp (in the 5.4.x I had to rename it back to `sof-skl_hda_card/` but now the [...]

Ok. That seems to be the (last?) remaining issue. In 5.4/5.5 we switched to a common HDMI codec driver. This unfortunately impacts the user-space interface seen by UCM. The old Lenovo UCM file you provide a link above, is written for the old driver (it has rules like "cset "name='hif5-0 Jack Switch' on"). The upstream 1.2.1.2 version https://github.com/alsa-project/alsa-ucm-conf/tree/v1.2.1.2/ucm2/sof-hda-dsp only supports to new driver. 

The kernel you built has only old driver enabled:
> [default.txt](https://github.com/thesofproject/sof/files/4049160/default.txt)

```
# CONFIG_SND_SOC_SOF_HDA_COMMON_HDMI_CODEC is not set
```

So the kernel you've built and the UCM file are not matching. You can fix this by updating the UCM file to use old-style rules for HDMI (copy from Lenovo file), or modify the kernel config and rebuild.

Partly triggered by this issue (and how to support this in distros), support to have conditional statements in UCM was recently added by Jaroslav to alsa-lib upstream. We did have UCM patches to support both HDMI codec drivers in the same UCM file, but apparently they have not ended up in the upstream UCM repository. @juimonen @libinyang any thoughts on this? 

> Regarding pulseaudio logs, this is the full startup log:

Ack. These seem mostly ok, but the HDMI setup is only partial due to mismatching UCM&kernel. As a result, Pulseaudio cannot detect connection status of the monitors and the HDMI outputs are not shown in the GUIs (typically only connected monitors are shown).

hsanjuan, 2020-01-13:
OK, in the meantime I asked the openSUSE folks to enable this:

https://bugzilla.opensuse.org/show_bug.cgi?id=1160819

hsanjuan, 2020-01-27:
Ok, I know have a `CONFIG_SND_SOC_SOF_HDA_COMMON_HDMI_CODEC=y` kernel. I see the HDMI outputs in pulseaudio, however the ` Cannon Point-LP High Definition Audio Controller profile` still says `Unavailable` (but can be selected anyways and works):

[image]

Would this be an UCM problem to report in alsa? I'm happy if this issue is closed, and thanks so much for the help!

lgirdwood, 2020-01-27:
@hsanjuan I think @juimonen and @kv2019i have done some recent UCM updates and PA integration fixes , can you guys comment.

kv2019i, 2020-01-29:
@hsanjuan @lgirdwood Yes, I think there is still some issue with UCM and/or the Pulseaudio version. Can you provide a copy of your UCM file (a provide a git pointer where you got it) and the pulseaudio log?

hsanjuan, 2020-01-29:
@kv2019i I use the standard alsa files from: https://github.com/alsa-project/alsa-ucm-conf/tree/v1.2.1.2/ucm2/sof-hda-dsp. Here's a fresh pulseaudio log:

<details>
  <summary>pulseaudio log (click to expand) </summary>
<pre>
[09:34:04] ~ $ pulseaudio -k; pulseaudio -vvvvv
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: RealtimeKit worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 13.0-rebootstrapped
D: [pulseaudio] main.c: Compilation host: x86_64-suse-linux-gnu
D: [pulseaudio] main.c: Compilation CFLAGS: -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type  -g -fPIE -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
D: [pulseaudio] main.c: Running on host: Linux x86_64 5.5.0-rc7-2.g7193c66-default #1 SMP Fri Jan 24 08:18:17 UTC 2020 (7193c66)
D: [pulseaudio] main.c: Found 4 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Running from build tree: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is 795aa2e781f24353a3fc7915960d760e.
I: [pulseaudio] main.c: Session ID is 1.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/hector/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib64/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
D: [pulseaudio] memblock.c: Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
I: [pulseaudio] cpu-x86.c: CPU flags: CMOV MMX SSE SSE2 SSE3 SSSE3 SSE4_1 SSE4_2 
I: [pulseaudio] svolume_mmx.c: Initialising MMX optimized volume functions.
I: [pulseaudio] remap_mmx.c: Initialising MMX optimized remappers.
I: [pulseaudio] svolume_sse.c: Initialising SSE2 optimized volume functions.
I: [pulseaudio] remap_sse.c: Initialising SSE2 optimized remappers.
I: [pulseaudio] sconv_sse.c: Initialising SSE2 optimized conversions.
I: [pulseaudio] svolume_orc.c: Initialising ORC optimized volume functions.
D: [pulseaudio] database-tdb.c: Opened TDB database '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-device-volumes.tdb'
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-stream-volumes.tdb'
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/hector/.config/pulse/795aa2e781f24353a3fc7915960d760e-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry0
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry2
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry3
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry4
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry5
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry6
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry7
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry8
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry9
[... cut ...]
D: [pulseaudio] alsa-mixer.c: Probe of element 'Rear Mic' succeeded (volume=0, switch=0, enumeration=0).
D: [pulseaudio] alsa-mixer.c: Probe of element 'Dock Mic' succeeded (volume=0, switch=0, enumeration=0).
D: [pulseaudio] alsa-mixer.c: Probe of element 'Dock Mic Boost' succeeded (volume=0, switch=0, enumeration=0).
D: [pulseaudio] alsa-mixer.c: Probe of element 'Internal Mic Boost' succeeded (volume=0, switch=0, enumeration=0).
D: [pulseaudio] alsa-mixer.c: Probe of element 'Front Mic Boost' succeeded (volume=0, switch=0, enumerati

kv2019i, 2020-01-29:
Thanks @hsanjuan ! This looks pretty good, all the devices are found and no errors are seen. The only odd thing is this, which I think makes the profile marked as not available in UI:

This is good:
I: [pulseaudio] module-alsa-card.c: Found UCM profiles
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,5: sink dev HDMI3
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,4: sink dev HDMI2
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,3: sink dev HDMI1
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,0: source dev Headphones2
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,6: source dev Mic
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,0: sink dev Speaker
D: [pulseaudio] alsa-ucm.c: UCM mapping: Mapping HiFi: hw:sofhdadsp,0: sink dev Headphones1
D: [pulseaudio] alsa-mixer.c: Profile HiFi (Play HiFi quality Music), input=(null), output=(null) priority=8000, supported=yes n_input_mappings=2, n_output_mappings=4

But then this is the odd thing:
D: [pulseaudio] module-alsa-card.c: Jack 'Headphone Jack' is now unplugged
D: [pulseaudio] card.c: Setting card alsa_card.pci-0000_00_1f.3-platform-skl_hda_dsp_generic profile HiFi to availability status no

So unplugging the headphone jack makes the whole profile to be unavailable. @juimonen any idea about this?

lgirdwood, 2020-02-12:
This is also showing up on APL CI for multiple PRs now.
```
kernel: [  559.964590] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  559.964596] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  560.469206] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x80010000 size 120
```

lgirdwood, 2020-02-12:
@slawblauciak please re-assign as needed. It seems this is now being seen on multiple CAVS versions.

lgirdwood, 2020-02-12:
Related to #2347 ?

slawblauciak, 2020-02-18:
Can you please test it with 1.4.2? Ideally it would also have to be with the latest kernel, to make sure there are no ABI/functionality incompatibilities.

tlauda, 2020-03-03:
@hsanjuan @lgirdwood @slawblauciak Can we close this bug? DAI config timeouts seem to be fixed now.

hsanjuan, 2020-03-03:
Sorry, I did not manage to test yet, but I'll re-open if it still happens when I do.

closed 2020-03-03 by hsanjuan

hsanjuan, 2020-03-04:
With Linux 5.5.7 and topologies from master:

```
[   22.413018] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog
[   22.413110] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, 
[   22.413243] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver 
[   22.413389] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog
[   22.413488] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   22.420762] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_compon
[   22.429016] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   22.429017] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_gene
[   22.599025] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   22.599026] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   22.703230] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[   22.703232] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[   22.703439] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   22.746986] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:11:0
[   22.746990] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not
[   22.747809] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not
[   22.747810] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not
[   22.748533] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not
[   22.748534] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not
[   22.754140] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out no
[   22.754142] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Pl
[   22.757485] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out no
[   22.757487] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU P
[   22.757488] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not
[   22.757489] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CP
[   22.757489] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out no
[   22.757491] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Ca
[   22.760029] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out no
[   22.760031] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU C
[   22.762198] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out no
[   22.762200] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CP
[   22.762965] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out no
[   22.774363] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widg
[   22.906744] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_ds
[   22.906772] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_
[   22.906798] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/
[   22.906821] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/
[   22.906844] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/
[   30.102501] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   30.221949] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

There is no IPC timeout. Sound works. And the speakers are no longer appearing as "unavailable" in pulseaudio.