thesofproject/sof #2269: [FEATURE] Make dai.c generic and move DAI specific code into DAI drivers.

lgirdwood, 2020-01-09:
**Is your feature request related to a problem? Please describe.**
dai.c contains programming code for ssp, hda etc, this IP specific code should be in the dai driver and not the DAI core.

**Describe the solution you'd like**
Move all DAI specific logic to dai drivers.

lgirdwood, 2020-08-18:
@RanderWang any progress here ?

mengdonglin, 2020-09-01:
@lgirdwood Now the SSP specific code is in https://github.com/thesofproject/sof/blob/master/src/drivers/intel/ssp/ssp.c
and HDA spefic code is in https://github.com/thesofproject/sof/blob/master/src/drivers/intel/hda/hda.c 

In dai.c, only function dai_config() has some type-specific logic https://github.com/thesofproject/sof/blob/master/src/audio/dai.c#L759

Do you think we need to modify this part? There is some Freescale code involved.

lgirdwood, 2020-09-01:
@mengdonglin Ok, lets push into v.1.7 @dbaluta fyi some imx parts also need moved.

paulstelian97, 2020-09-01:
@lgirdwood Should I also take care of moving the i.MX parts from there in my existing PR #3368 or that is worth a new separate one?

lgirdwood, 2020-09-01:
@paulstelian97 It's probably best if this was done atomically in one PR - we can do for v1.7

paulstelian97, 2020-09-01:
> @paulstelian97 It's probably best if this was done atomically in one PR - we can do for v1.7

Fair enough.

lgirdwood, 2021-01-12:
@paulstelian97 any update ?

paulstelian97, 2021-01-12:
In the backlog, might try to sprint it these days if I get a bit of extra time (right now trying to update Xtensa compiler for i.MX platforms).

lgirdwood, 2021-01-12:
Ok, lets move it for later.

lgirdwood, 2021-05-10:
Update: half implemented in v1.8 as part of IPC refactoring, to be completed in v1.9 with IPC abstraction

kv2019i, 2023-02-15:
Moving to v2.6 to check this. We now lifted the DAI abstraction into Zephyr, so we should have a clean split now.

mengdonglin, 2023-04-07:
@kv2019i  Can we close this? Already done in Zephyr or are there any remaining items?

kv2019i, 2023-04-12:
Ack @mengdonglin , I think we can close this. @lgirdwood Ok for you? There is still a few place that relate to converting from IPC DAI type specific structures into the generai dai.h interface of Zephyr. No actual IP programming is anymore present.

mengdonglin, 2023-04-13:
Thank you @kv2019i  Closing.

closed 2023-04-13 by mengdonglin