thesofproject/sof #2270: [BUG][APL][KD] Detection signal data a the beginning of drained data

zrombel, 2020-01-09:
**Describe the bug**
BUG found on APL platform in Keyword Detection test incorporating DMIC interface. Test plays 6 second wave file on dmic injector where 3 s is silent and 3 s contain sine signal. When Dummy component detects sine signal on DMIC interface it triggers KPB component to drain history buffer and continue playing realtime data. Recorded wave file should contain ~2.1 s of silence and 3 s of sine signal but at the begging of file there is a short sine signal of exactly 20 ms length (320 samples).

[image]

[image]

**Topology**
```
        +----------+   +------------------+
        |    OS    |-->|   AUDIO DEVICE   |-----------------------+
        +----------+   +------------------+                       |
                                                                  |
                   +-------------------------+                    |
        +--------+ | +---+   +-----+   +---+ |                    |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                  |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap        |
                   +-------------------------+ |                  |
                                               |                  |
                             +-----------------+                  |
        +--------------------|----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-| DMIC |---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ |
        +-------------------------------------------------------+
```
**To Reproduce**
On APL platform run python test:
07_04_TestKdDmic16000Hz16b16b2ch with --playback-iterations > 100

**Reproduction Rate**
~1%

**Environment**
1) SOF master branch hash 79d6ea654cd8403eaf21e7d69e466d3fb73e4d87 and earlier
2) Python tests master branch hash 5b92681e5d967d13f89b67cc11730b64fc465816

**Logs**
[07_04_TestKdDmic16000Hz16b16b2ch_APL_KD_BUG.zip](https://github.com/thesofproject/sof/files/4040659/07_04_TestKdDmic16000Hz16b16b2ch_APL_KD_BUG.zip)

lgirdwood, 2020-01-09:
@singalsu any inputs ?

singalsu, 2020-01-09:
@lgirdwood DMIC starts with filters muted but there's possibility the FIFOs could contain old samples when started. I'll do some checks with my APL board.

singalsu, 2020-01-09:
Also to be safe the KPB should reject any positive detections for the first 500 ms or so after start. As seen in HDA capture many codecs output some noise in the beginning.

zrombel, 2020-01-10:
@singalsu I was suspecting the same about DMIC containing some old data but in my opinion is not the issue. First of all detection is triggered in right moment - after 3s of playback, if there was some data in DMIC FIFO detection would happen at the very beginning of playback. Second I've modified test so the sine signal fade over time, but data at the beginning still have full amplitude just like data from detection moment. So i my opinion this is a short mismatch between realtime and drain data somewhere in KPB.

[image]

lgirdwood, 2020-01-10:
@singalsu @zrombel are we sure we have cleared all buffers prior to DMIC trigger start ?
@singalsu how big is the DMIC FIFO ? does it need cleared prior to start ?

zrombel, 2020-01-10:
@lgirdwood is there a way to explicitly clear buffers? This issue can be reproduced on first iteration, right after ADSP power on and in that case i thing that buffers are empty.

lgirdwood, 2020-01-10:
@zrombel pipeline should clear buffer before trigger start. If in doubt check with @tlauda

mrajwa, 2020-01-27:
@zrombel, can you please retest with this patch https://github.com/thesofproject/sof/pull/2329

Thanks

zrombel, 2020-01-29:
PR #2329 does not fix this bug, Further more I found that this bug is also reproducible with SSP interface, so it's not only DMIC bug.

mrajwa, 2020-01-29:
@zrombel, ok we know what happens. Can you in test change history_depth (the amount of data host requested from history buffer) from 2,1 seconds to 2 seconds? It looks like we have a delay on FW side and new data overwrite oldest data in history buffer.

zrombel, 2020-02-03:
After validating with PR https://github.com/thesofproject/sof/pull/2343 on APL platform bug no longer can be found. PR https://github.com/thesofproject/sof/pull/2343 is fixing it.

lgirdwood, 2020-02-20:
@mrajwa any update on the Prs ? Is this blocking on @lbetlej returning ?

mrajwa, 2020-02-21:
@lgirdwood yes it's my responsibility, I will submit PR for soon.

slawblauciak, 2020-03-31:
@mrajwa any updates on this issue?

mrajwa, 2020-04-01:
@slawblauciak @zrombel the fix should be covered by this PR https://github.com/thesofproject/sof/pull/2688

mrajwa, 2020-04-07:
Problem no longer reproducible after PR #2688, therefore I am closing this issue.

closed 2020-04-07 by mrajwa