thesofproject/sof #2271: [BUG][APL-nocodec]Unexpected sof error trace print after bootup

Liviali155, 2020-01-10:
**Describe the bug**
Unexpected sof error trace print after bootup
just an error message output

**To Reproduce**
1.sudo reboot
2.sudo sof-logger -l /etc/sof/sof-*.ldc

**Reproduction Rate**
10/10

**Expected behavior**
There is no sof error trace after boot up

**Actual result**
There is sof error trace output after bootup

```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1       VOLUME 2.11     71656512.916667  71656512.000000 udio/volume/volume.c:298    volume_set_chan: Limite                                   d request 0 to VOL_MIN.
    0      1       VOLUME 2.11     71656518.802083         5.885417 udio/volume/volume.c:298    volume_set_chan: Limite                                   d request 0 to VOL_MIN.
```
There is no dmesg error log
[dmesg0110.log](https://github.com/thesofproject/sof/files/4044313/dmesg0110.log)
[sof_errortrace.txt](https://github.com/thesofproject/sof/files/4044314/sof_errortrace.txt)
[sof_trace0110.log](https://github.com/thesofproject/sof/files/4044315/sof_trace0110.log)

lgirdwood, 2020-01-10:
@singalsu any ideas ?

singalsu, 2020-01-10:
Yes, this is known and harmless. But it can be fixed easily. I actually tried once but static code analysis nagged unnecessarily about it so I kept it as is. I can do it now another way.

lgirdwood, 2020-01-10:
@singalsu ok, thanks. Tagged for v1.5

Liviali155, 2020-01-13:
Issue can also be reproduced on CML Mantis with  ALC3204 in HDA mode

Tried with PR #2272 ,error not seen on both APL NOCODEC and CML Mantis with  ALC3204 in HDA mode

Liviali155, 2020-01-14:
PR #2272  merged,closed

closed 2020-01-14 by Liviali155