thesofproject/sof #2275: [BUG] Microphone array not working on Lenovo Yoga C740 (Comet Lake, i7-10510U)

p-try, 2020-01-12:
**Describe the bug**
On my Lenovo Yoga C740 with a Comet Lake CPU (i7-10510U), the dual-microphone array does not work. Speakers are working. In GNOME settings, "Input device" shows "Multichannel input (sof-hda-dsp)", but the input monitor bar (below the input selection combo box) does not register any input noise and Audacity records only silence.
I have "dmic0" and "dmic1" in alsamixer, though.

**To Reproduce**
I'm on Ubuntu 19.10. In order to have proper SOF support, I installed [mainline kernel 5.5-rc5](https://kernel.ubuntu.com/~kernel-ppa/mainline/). I stole the firmware files from the [latest Fedora package](https://koji.fedoraproject.org/koji/buildinfo?buildID=1426494).

**Reproduction Rate**
100%

**Expected behavior**
Microphone input to be recorded in Audacity.

**Impact**
showstopper if microphone is needed (not for me, though)

**Environment**
Ubuntu 19.10 with mainline kernel 5.5-rc5, alsa-firmware-1.2.1-4 borrowed from Fedora 31

**Screenshots or console output**
This is my `dmesg | grep audio`:
```
[   21.174697] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   21.174946] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   21.174960] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   21.175292] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   21.175301] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   21.175469] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   21.175726] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   21.185616] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   21.185619] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   21.542017] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   21.542020] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   21.642253] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[   21.642256] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[   21.642490] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   21.880019] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[   21.880027] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   21.881436] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   21.881439] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   21.882280] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   21.882282] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   21.883293] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   21.883296] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   21.884120] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   21.884123] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   21.884125] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   21.884127] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   21.884129] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   21.884131] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   21.884854] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   21.884856] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   21.885585] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   21.885588] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   21.886324] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   21.893543] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   21.912348] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[   21.912350] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   21.912352] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   21.912353] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[   21.912354] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[   21.912356] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[   21.959138] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[   21.959145] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[   46.131032] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   46.224672] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   62.963068] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   63.062240] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

I'd like to help to make this work. Let me know if I can provide any further info.

juimonen, 2020-01-13:
the boot log doesn't look that bad... probably your pulseaudio is using sw volume, so could you check with amixer if the dmic mute switch is on (this is not visible in alsamixer), so something like: "amixer contents | grep "Dmic0 Capture Switch" -A2"... also you could check with "dmesg -w" during recording runtime do you get some other errors? Then you could try just bypassing pulseaudio and trying to record from dmic with arecord.

lgirdwood, 2020-01-13:
@p-try thanks, would be good to see your arecord dmesg and cmd line with pulseaudio disabled as suggested by @juimonen

p-try, 2020-01-14:
Thank you very much for your answers and suggestions.

I stopped pulseaudio using
```
systemctl --user stop pulseaudio
systemctl --user stop pulseaudio.socket
```

`dmesg -w` during `arecord test.wav`
```
[  631.371430] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  631.464809] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

`amixer contents | grep "Dmic0 Capture Switch" -A2`
```
numid=43,iface=MIXER,name='Dmic0 Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on

```

lgirdwood, 2020-01-14:
@p-try can you try arecord when pulseaudio is disabled and send the dmesg. Fwiw, there will be some dmic fixes in 1.4.2 due out next week.

p-try, 2020-01-19:
I haven't had any luck with disabling pulseaudio using [these](https://kodi.wiki/view/PulseAudio/HOW-TO:_Disable_PulseAudio_and_use_ALSA_(without_removing_PulseAudio)_for_Ubuntu#Removing_PulseAudio_from_Ubuntu_-_why_it_may_not_be_such_a_good_idea) instructions (the pulseaudio process was still running) so I removed it completely using `apt purge`.

I still get the same dmesg when running arecord:
```
[  228.521187] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  228.620918] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

The recording contains only silence.

juimonen, 2020-01-20:
@p-try the dmesg messages you have should not be fatal... could you still try explicit hw definitions with arecord, so omething like: "arecord -Dhw:0,6 -d 10 -f S16_LE -c 2 -r 48000 test.wav " ?

slawblauciak, 2020-01-21:
Can you please post the full dmesg log? Thanks

p-try, 2020-01-22:
@juimonen Your command worked, sound was recorded properly this time. Does this mean pulseaudio is the problem?

The full dmesg is here: [https://drive.google.com/file/d/1UZZbn0VNiJHv-Npxbo3Bl0ZutHlPD6kv/view?usp=sharing](https://drive.google.com/file/d/1UZZbn0VNiJHv-Npxbo3Bl0ZutHlPD6kv/view?usp=sharing)

juimonen, 2020-01-22:
@p-try yes I think it is. Your pulseaudio probably doesn't create a source for dmic at all... you now have several input sources from alsa, and pulseaudio just doesn't handle the naming (it is a quite new thing, so pulseaudio paths configuration just doesn't support it) You could check for example with "pactl list sources". 
We are working on this in upstream by trying to switch pulseaudio to use ucm configuration (with dmic defined). There is no very easy way for you to "fix" it, if you have time/patience you could try installing latest/greatest pulseaudio and alsa-ucm-conf from git...

p-try, 2020-01-26:
Thank you so much for your help. The ucm file for sof-hda-dsp looks promising, the microphone is listed there as `hw:${CardId},6` and according to `arecord -l` this is exactly my hardware layout.

I have compiled and installed the latest commit of pulseaudio and copied the alsa-ucm-conf files to /usr/share/alsa/ucm. I have now run into the issue that the ucm configuration for sof-hda-dsp is provided only in ucm2 format, and I still have alsa 1.1.9 on Ubuntu Eoan which does not seem to support ucm2. A build for alsa 1.2.1 is in the process of being made for Focal so I might switch to the Focal beta in order to try it out once the alsa build is ready.

lgirdwood, 2020-01-27:
@p-try ok, please keep us updated, fwiw new stable FW is out v1.4.2
https://github.com/thesofproject/sof/releases/tag/v1.4.2

p-try, 2020-01-29:
Again, I stole some packages from Fedora to get alsa-utils version 1.2.1 which supports ucm2. Aaaand..... it is working now! GNOME settings is now showing "Digital Microphone - sof-hda-dsp" as the input device and the volume bar is registering noise. Also, recording in Audacity is working now. I expect that everything will be working out-of-the-box as soon as the updated alsa and ucm packages are pushed to the Ubuntu repositories. I will continue my efforts there.

Again, thank you so much for your most efficient help.

lgirdwood, 2020-01-29:
@p-try your welcome.

closed 2020-01-29 by lgirdwood

aravind816, 2020-11-11:
@lgirdwood 

 Recently I have installed Deepin 20 in my Lenovo C740 i5 10th gen processor.

 Both the microphone and fingerprint is not working.

 Could you please advise how can I resolve this issue?

lgirdwood, 2020-11-11:
> @lgirdwood
> 
> Recently I have installed Deepin 20 in my Lenovo C740 i5 10th gen processor.
> 
> Both the microphone and fingerprint is not working.
> 
> Could you please advise how can I resolve this issue?
> 

Hi @aravind816 it looks like you have an older kernel and FW. Have you tried a more recent kernel and FW ? 
Also please see https://thesofproject.github.io/latest/contribute/process/bug-tracking.html#how-to-report-a-bug for the information required for reporting a bug. It's best to paste dmesg directly rather than as a screenshot as it makes it easier for developers to read.