thesofproject/sof #2280: [CML][BUG]Multipipe display "input/output error" when aplay/arecord is more than 6 pipleline

sinahuang, 2020-01-13:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

sinahuang, 2020-01-14:
This issue is caused by commit https://github.com/thesofproject/sof/commit/91fb594bd9455fcf8415a588cf45e6294ac46e0e

ClarexZhou, 2020-01-14:
@mmaka1 could you please check first?

paulstelian97, 2020-01-14:
Is it possible to dynamically use a different oscillator dependent on load? As in using the higher performance one when there is need.

lgirdwood, 2020-01-14:
@plbossart any ideas, nothing obvious in any of the logs. Not sure yet if it's SDW only.

lgirdwood, 2020-01-14:
@sinahuang can you retry this with 2,3,4,5 streams - it may still be an issue with less streams.

sinahuang, 2020-01-15:
Test with 2,3,4,5 streams, issue can't be reproduced. Issue only can be reproduced when more than 6 piplines.

plbossart, 2020-01-15:
on this hardware there are 3 playback devices and 2 capture ones.
What you are doing is enabling all of them, and then add HDMI.

I suspect you must run out of resources at some point and underflow. That would explain why there's no error in dmesg, but you consistently cannot play

Rather than using a scripts, please try manually adding a stream for each of the devices in a different ssh terminal, and send the logs when the error happen (terminal output, dmesg logs, DSP logs).

sinahuang, 2020-01-15:
On CML-H Laptop there are  5 playback devices and 2 capture ones.
```ubuntu@cd-CML-H-rt711-rt1308-rt715:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 0: Headphone (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 2: SDW1-speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 3: SDW2-speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 5: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 6: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 7: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cd-CML-H-rt711-rt1308-rt715:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 1: Headset mic (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 4: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cd-CML-H-rt711-rt1308-rt715:~$
```

If I manually adding a stream for each of the devices in a different ssh terminal, the error happened after added the 7th devices.

 dmesg logs:
[dmesg.log](https://github.com/thesofproject/sof/files/4063229/dmesg.log)
[kern.log](https://github.com/thesofproject/sof/files/4063230/kern.log)
[sof_error_trace.log](https://github.com/thesofproject/sof/files/4063232/sof_error_trace.log)

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA         439946318.854167 439946304.000000      src/lib/agent.c:54  	validate(), ll drift detected, delta = 25223
    0      1  SCHEDULE_LL         439950019.375000      3700.520752 edule/timer_domain.c:34  	timer_report_delay(): timer 3 delayed by 35 uS 844 ticks
```

Terminal output:

[image]

lgirdwood, 2020-01-15:
@tlauda @mmaka1 any comments here ? Are we running out of FW resources with 7 streams ? I'm assuming cycles ? Would scheduler spit out debug if scheduling requirements are not met ?

paulstelian97, 2020-01-15:
If you revert 91fb594 you can play more than 6 just fine? From what I see that commit simply selects a lower power clock for the DSP. I understand that you did identify it as the culprit in the bisect but it is not clear to me if reverting it would fix it (with the side effect of higher power consumption).

lgirdwood, 2020-01-15:
@paulstelian97 @sinahuang good find, I did not see that in the log.
@mmaka1 looks like we have hit our CPU limit with the low power clock, It looks like we need a more friendly message here to help the users understand this. We should also continue with the streams 0 - 5 and kill stream 6 if the scheduler can never meet it's deadline (as the newest work)

sinahuang, 2020-01-16:
If you revert https://github.com/thesofproject/sof/commit/91fb594bd9455fcf8415a588cf45e6294ac46e0e , issue can't be reproduced.

plbossart, 2020-01-16:
@mmaka1 If we use the 120MHz ring oscilator, can we use 4 cores? Wondering if we could move some of the HDMI stuff to different cores.
Or when we use HDMI we switch to 400MHz, it's not a low-power usage anyways.

lgirdwood, 2020-01-16:
@sinahuang this is a kconfig option, between LP and HP clocks, so it's now a configuration limitation rather than a bug. 
To me this issue is now more about why stream 0 - 5 are being killed when stream 6 puts CPU utilisation > 100%

plbossart, 2020-01-16:
> @sinahuang this is a kconfig option, between LP and HP clocks, so it's now a configuration limitation rather than a bug.
> To me this issue is now more about why stream 0 - 5 are being killed when stream 6 puts CPU utilisation > 100%

well, you can't really guarantee a quota for previous allocated streams, and only fail for new streams.  when you end-up exceeding the available resources, I don't think we can gracefully exit, it's just completely random which stream underflows and gets killed.

paulstelian97, 2020-01-16:
A priority system maybe? Or at least only one stream should Xrun rather than all, even if it's not that deterministic which of them?

plbossart, 2020-01-16:
IIRC with EDF scheduling, once you exceed the total resources there's no way you can guarantee what happens in a predictable way. EDF is optimal until it breaks.

lgirdwood, 2020-01-16:
> well, you can't really guarantee a quota for previous allocated streams, and only fail for new streams. when you end-up exceeding the available resources, I don't think we can gracefully exit, it's [...]

We can timestamp a stream/task at startup and if EDF is consistently above a threshold for X iterations we can kill the newest task until EDF stabilises. It's not perfect but it's a lot better than all streams stopping.
It's probably also feasible to determine whether we should start a stream given current EDF utilisation (per core) and pipeline budget, again not perfect but better than killing all streams that we have today.

plbossart, 2020-01-16:
Right, I think some resource management is inevitable and expected in fact with EDF.

lgirdwood, 2020-01-17:
@tlauda @mmaka1 @lbetlej do you guys have EDF resource management in mind atm ?

YvonneYang2, 2020-01-19:
Do 6 multipiplines aplay/arecord , and pause/release on aplay0,2(RT1308) can also cause IO error. 
May be related to this issue? 

e.g 
Terminal 1: aplay -Dhw:0,0 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
Terminal 2: aplay -Dhw:0,2 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
Terminal 3: aplay -Dhw:0,6 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
Terminal 4: aplay -Dhw:0,7 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
Terminal 5: arecord -Dhw:0,1 -r48000 -c2 -fs16_le test1.wav -vv -i
Terminal 6: arecord -Dhw:0,4 -r48000 -c2 -fs16_le test4.wav -vv -i

Pause/release in terminal 2 will cause IO error.

[dmesg.log](https://github.com/thesofproject/sof/files/4082424/dmesg.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/4082428/sof_trace.log)

Test recipe:
Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev  (commit: 674afed)
FW: https://github.com/thesofproject/sof/commits/master commit: 415273f

lgirdwood, 2020-01-20:
@YvonneYang2 yep, any EDF overload will currently kill all streams and this is currently showing in low power clock mode. We are looking at EDF resource management.

lgirdwood, 2020-01-27:
EDF resource management target v1.6

tlauda, 2020-03-10:
@lgirdwood @lbetlej What is the expectation here? Pipelines are scheduled with low latency scheduler (timer) and not with EDF, so EDF resource management won't help here. I imagine that the first step would be to review what pipelines are run here and measure their cycles. We need to assess whether some of the components is not optimized. Assuming everything will look correct, I think the resource management should be manually set based on worst case scenario and taking into consideration topology and clock settings.

sinahuang, 2020-03-11:
Test with sof-dev (commit:d05959) +master (commit:8e5ab9c), issue also can be reproduced on TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode when playback/capture more than 5 piplines at the same time.

lgirdwood, 2020-03-12:
@tlauda we do need resource management, i.e. each component has a measured worst case cost per number of samples per platform. e.g volume 24bit cost on platform X is Y cycles. This can then bo worked out at runtime and rejected if new load + current load > budget.

lbetlej, 2020-03-12:
@lgirdwood @lauda resource management is an interesting topic itself:

1. I'm with you on component level worst case compute requirements for each audio format
2. The measurement process has to account: DSP configuration, SRAM memory access time (waitstates) and cache worst case latency for each audio format (SRC as an example)
3. The worst case utilization has to be visible at 2 levels:

-  topology so that during topology creation the system integrator can verify if the workload fit's in DSP
- at FW level to verify if new pipeline is exceeding compute budget

mengdonglin, 2020-03-24:
Caused by DSP overload, please see comments of @tlauda  https://github.com/thesofproject/sof/issues/2531#issuecomment-601175843

slawblauciak, 2020-04-07:
@sinahuang should no longer occur on master, high performance clock is enabled by default

sinahuang, 2020-04-07:
Issue can't be reproduced with sof-dev commit:052e45d and master commit: b4e8b1e on on CML-H and CML-U, so closed it.

closed 2020-04-07 by sinahuang