thesofproject/sof #2293: [FEATURE] Add flush timeout to trace

cujomalainey, 2020-01-16:
**Is your feature request related to a problem? Please describe.**
When tracing, if no additional traces come after your trace and there was a flush in the middle of the traces then it appears SOF hung in the middle of your code, this is not ideal.

**Describe the solution you'd like**
A simple timer to track the last trace, and flush the buffer if no additional traces arrive in that time.

**Describe alternatives you've considered**
Reducing watermark (more interrupts, not as useful)
Adding flush (doesn't work globally)

**Additional context**
@benzh fyi

lgirdwood, 2020-01-16:
@cujomalainey ACK, my understanding is that we should be flushing trace when buffer reaches watermark or after a short timeout (whichever is sooner). Fwiw, the trace_error() data is synced into the smaller error buffer, but it's probably too small for general debug.

cujomalainey, 2020-01-16:
This very well may be mute, we may have been looking at old code given https://github.com/thesofproject/sof/commit/b37efd554f39ae00af1e24350101360604d08164#diff-b855e4ebd5e9885452d01c6822f59b35

cujomalainey, 2020-01-17:
Reading top of tree this looks like it should be working, I will follow up with Ben offline and see if we can trace why some of the traces are falling being. In the mean time I will convert this to a bug report.

@bzhg can you add the snippets (traces and logs) just so we have context?

lgirdwood, 2020-01-17:
@cujomalainey ok, it's marked as required fix for v1.5

lgirdwood, 2020-01-27:
@cujomalainey @bzhg any updates ?

cujomalainey, 2020-01-27:
@bzhg is OOO till tomorrow, 1.5 is fine, it is also fine if it gets punted, this appears to be a stray case limited to us.

lgirdwood, 2020-02-20:
@cujomalainey @bzhg any update - will push into v1.6

bzhg, 2020-02-20:
Hi Liam, let me test with master again. I was on an old branch.

bzhg, 2020-02-20:
Just rebased my code on master @ 135cace3 and the tracer is working perfectly. Will bisect and backport the appropriate patches when working on old branches. I think we can close this now. Sorry about the spam.

closed 2020-02-20 by lgirdwood