thesofproject/sof #2295: [FEATURE] host XRUNs should not stop DSP pipelines

lgirdwood, 2020-01-16:
**Is your feature request related to a problem? Please describe.**
An XRUN on the host side will stop the DSP pipeline.
```
    0      1          DMA         488769847.343750      1348.645874 intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1          DAI 2.14    488769851.822917         4.479167      src/audio/dai.c:618 	dai_report_xrun() error: underrun due to no data available
    0      1         COMP 2.14    488769854.739583         2.916667      src/audio/dai.c:678 	comp_underrun() error: dev->comp.id = 14, source->avail = 64, copy_bytes = 0
    0      2         PIPE 2.15    488769858.385417         3.645833 src/audio/pipeline.c:715 	pipeline_trigger()
    0      2         HOST 2.10    488769861.250000         2.864583     src/audio/host.c:299 	host_trigger()
    0      2          DMA         488769871.979167        10.729167 intel/cavs/hda-dma.c:538 	hda-dmac: 5 channel 1 -> stop
    0      2       VOLUME 2.11    488769875.052083         3.072917 udio/volume/volume.c:564 	volume_trigger()
    0      2          DAI 2.14    488769877.916667         2.864583      src/audio/dai.c:536 	dai_comp_trigger(), command = 8
    0      2          DAI 2.14    488769887.135417         9.218750      src/audio/dai.c:594 	dai_comp_trigger(), XRUN
    0      2          DAI 2.14    488769889.479167         2.343750      src/audio/dai.c:600 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         488769891.979167         2.500000 intel/cavs/hda-dma.c:538 	hda-dmac: 7 channel 1 -> stop
    0      2  SCHEDULE_LL         488769895.677083         3.697917 hedule/ll_schedule.c:374 	task cancel 0xbe035340
    0      2  SCHEDULE_LL         488769899.166667         3.489583 hedule/ll_schedule.c:236 	num_tasks 7 total_num_tasks 7
    0      1         PIPE         488769905.468750         6.302083 src/audio/pipeline.c:866 	pipeline_copy() error: ret = -61, start->comp.id = 14, dir = 1
    0      1         PIPE 2.15    488769909.531250         4.062500 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

**Describe the solution you'd like**
Pipeline to ramp last host period to silence and play silence (with ramp up) until new host data is available.

@singalsu does the ramping here sound reasonable for host playback XRUNs

paulstelian97, 2020-01-16:
For ramping to silence there needs to be enough of a warning beforehand in order to make a ramp that isn't that abrupt, @singalsu Is the 1ms (default pipeline half-buffer) enough to make a ramp-down that isn't stupidly abrupt when it happens?

There were some stubs related to firmware XRUN handling, and it's a somewhat recent change that the default config has become to simply not handle them and stop everything.

plbossart, 2020-01-16:
On 1/16/20 5:05 AM, Liam Girdwood wrote:
> *Is your feature request related to a problem? Please describe.*
> An XRUN on the host side will stop the DSP pipeline.

when would you see an XRUN on the host side first (where the bandwidth 
is typically orders of magnitude larger than the audio link and we use 
larger buffers and burst transfers). XRUNs are usually due to issues on 
the link side due to shallower FIFOs/smaller buffers and isochronous 
transfers.

> |0 1 DMA 488769847.343750 1348.645874 intel/cavs/hda-dma.c:789 
> hda_dma_link_check_xrun() error: xrun detected 0 1 DAI 2.14 
> 488769851.822917 4.479167 src/audio/dai.c:618 dai_report_xrun() error: 
> underrun due to no data available 0 1 COMP 2.14 488769854.739583 

And here is looks like a DAI issue first?

lgirdwood, 2020-01-16:
> On 1/16/20 5:05 AM, Liam Girdwood wrote: *Is your feature request related to a problem? Please describe.* An XRUN on the host side will stop the DSP pipeline.
> when would you see an XRUN on the host [...]

Host usually has high bandwidth, but the latency is not guaranteed especially when data is coming from a media with occasional high latency like MMC. Fwiw, aplay buffer defaults are especially bad with MMC.

> |0 1 DMA 488769847.343750 1348.645874 intel/cavs/hda-dma.c:789 hda_dma_link_check_xrun() error: xrun detected 0 1 DAI 2.14 488769851.822917 4.479167 src/audio/dai.c:618 dai_report_xrun() error: [...]

Yes, you are right, I did not see the "_link_" part in the HDA trace, that being said we should not stop the stream on XRUNs.

plbossart, 2020-01-16:
@lgirdwood I don't really believe in concealment for things that are fundamentally broken. You will just delay the inevitable and muck with timestamps, which could e.g. ripple into A/V sync, not to mention pops due to discontinuities if your last sample was full-scale. 

Why not play the last buffer backwards while we are at it, that would keep the same spectral content and bpm :-)

If the data producer is not capable of sustaining an average rate of 1.5 Mbits/s, nothing at the audio level is really going to help. Additional buffers are required, be it in the ALSA ring buffers or in the MMC layers.

lgirdwood, 2020-01-17:
@plbossart XRUNs in userspace or disc IO, whilst uncommon do happen so I don't think we can just stop the streams. We should at least try and continue and let userspace then manage any recovery or stream state change.

Pops will still happen if last sample is full scale and we stop the stream.

Fwiw, aplay defaults are not well suited to MMC (or any source with occasional high latency), they can sustain a high rate but can fall over due in aplay PCM R/W mode since the main thread blocks on PCM IO until the next period + 1 is needed.

plbossart, 2020-01-17:
It's not really our job to compensate for aplay using bad defaults...
I'd rate this feature as a very low priority, and I don't see which products would actually be enhanced by this.

paulstelian97, 2020-01-17:
I'd argue that on Xrun the pipeline ought to either play silence until userspace resolves it or trigger a pause, not outright stop. But if userspace can handle the stop (I see about 10 Xruns per minute on my system with some non-default buffer sizes doing arecord|aplay and although the Xruns cause issues with the sound at times, just wait a few seconds for the next Xrun which fixes it, and I only get a stop after quite a while)

plbossart, 2020-01-17:
Nothing is going to help if your system underflows 10 times per minute.

paulstelian97, 2020-01-17:
Yeah in my case it's caused by the much smaller buffer sizes than the aplay/arecord defaults (768 bytes instead of ~64 kB) but just wanted to point out how they are handled. Of course I wouldn't expect that many if I left it with the default. What I mentioned was that play-silence can be useful in case of the odd-xrun, and maybe it's less visible to the user as well as recovery can be faster in this case?

singalsu, 2020-01-20:
It's not simple to conceal gaps. Doing something like described (attenuate and repeat a block) would make just funny/annoying banjo sounds. The Rx speech error concealment in early GSM phones was like that :^)

lgirdwood, 2020-08-18:
@plbossart @paulstelian97 @singalsu  I think we have merged this now. Closing now. Any objections can re-open.

closed 2020-08-18 by lgirdwood