thesofproject/sof #2296: [BUG] APL: HDA playback host DMA is starved and stream is stopped.

lgirdwood, 2020-01-16:
**Describe the bug**
Seeing this on several Jenkins CI results for multiple PRs.
```
    0      1          DMA         488769847.343750      1348.645874 intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1          DAI 2.14    488769851.822917         4.479167      src/audio/dai.c:618 	dai_report_xrun() error: underrun due to no data available
    0      1         COMP 2.14    488769854.739583         2.916667      src/audio/dai.c:678 	comp_underrun() error: dev->comp.id = 14, source->avail = 64, copy_bytes = 0
    0      2         PIPE 2.15    488769858.385417         3.645833 src/audio/pipeline.c:715 	pipeline_trigger()
    0      2         HOST 2.10    488769861.250000         2.864583     src/audio/host.c:299 	host_trigger()
    0      2          DMA         488769871.979167        10.729167 intel/cavs/hda-dma.c:538 	hda-dmac: 5 channel 1 -> stop
    0      2       VOLUME 2.11    488769875.052083         3.072917 udio/volume/volume.c:564 	volume_trigger()
    0      2          DAI 2.14    488769877.916667         2.864583      src/audio/dai.c:536 	dai_comp_trigger(), command = 8
    0      2          DAI 2.14    488769887.135417         9.218750      src/audio/dai.c:594 	dai_comp_trigger(), XRUN
    0      2          DAI 2.14    488769889.479167         2.343750      src/audio/dai.c:600 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         488769891.979167         2.500000 intel/cavs/hda-dma.c:538 	hda-dmac: 7 channel 1 -> stop
    0      2  SCHEDULE_LL         488769895.677083         3.697917 hedule/ll_schedule.c:374 	task cancel 0xbe035340
    0      2  SCHEDULE_LL         488769899.166667         3.489583 hedule/ll_schedule.c:236 	num_tasks 7 total_num_tasks 7
    0      1         PIPE         488769905.468750         6.302083 src/audio/pipeline.c:866 	pipeline_copy() error: ret = -61, start->comp.id = 14, dir = 1
    0      1         PIPE 2.15    488769909.531250         4.062500 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2  SCHEDULE_LL         488769913.333333         3.802083 hedule/ll_schedule.c:104 	task complete 0xbe035340
    0      2  SCHEDULE_LL         488769915.781250         2.447917 hedule/ll_schedule.c:107 	num_tasks 6 total_num_tasks 6
    0      2  SCHEDULE_LL         488959913.906250    189998.125000 hedule/ll_schedule.c:104 	task complete 0xbe038e00
    0      2  SCHEDULE_LL         488959916.822917         2.916667 hedule/ll_schedule.c:107 	num_tasks 5 total_num_tasks 5
    0      2  SCHEDULE_LL         488959920.468750         3.645833 hedule/ll_schedule.c:104 	task complete 0xbe038c00
    0      2  SCHEDULE_LL         488959922.864583         2.395833 hedule/ll_schedule.c:107 	num_tasks 4 total_num_tasks 4
    0      2          IPC         490484082.968750   1524160.125000    src/ipc/handler.c:450 	ipc: comp 10 -> trigger cmd 0x50000
    0      2         PIPE 2.15    490484087.708333         4.739583 src/audio/pipeline.c:715 	pipeline_trigger()
    0      2         PIPE 2.15    490484090.677083         2.968750 src/audio/pipeline.c:698 	already stopped in xrun
```

**To Reproduce**
Run the multiple-pipeline test. i.e.
https://sof-ci.01.org/sofpr/PR2
285/build4541/devicetest/APL_UP2_PCM512X/multiple-pipeline-playback.sh/slogger.txt

Other configuration is default CI.

lgirdwood, 2020-01-27:
@slawblauciak @bkokoszx moving this into v1.6 since it involves XRUN handling changes and is linked to our EDF resource management work.

slawblauciak, 2020-02-18:
@zrombel can you please check if this is reproducible on our platforms?

zrombel, 2020-02-18:
I will try to reproduce it.

dengyangchao, 2020-02-19:
I tried this Scenario on sof-dev + master , do multiple-pipeline-playback stress test and **fail at 487th**.

[image]

Platform: APL UP2 with PCM512X in I2S mode
Firmware: 53d35e49 https://github.com/thesofproject/sof/commits/master
Kernel: 861c3087 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-apl-pcm512x.tplg same as firmware

no dmesg error

**logger-error**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         5436224502.447917              nan intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.14    5436224507.500000         5.052083      src/audio/dai.c:632 	dai_report_xrun() error: underrun due to no data available
    0      1         COMP 2.14    5436224510.729167         3.229167      src/audio/dai.c:707 	comp_underrun() error: dev->comp.id = 14, source->avail = 64, copy_bytes = 0
    0      1         PIPE         5436224554.322917        43.593750 src/audio/pipeline.c:767 	pipeline_copy() error: ret = -61, start->comp.id = 14, dir = 1
    0      1         PIPE 2.15    5436224557.916667         3.593750 src/audio/pipeline.c:934 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

[dmesg.log](https://github.com/thesofproject/sof/files/4223501/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/4223502/logger.log)
[sof-error.log](https://github.com/thesofproject/sof/files/4223503/sof-error.log)

add test scripts:
[multiple-pipeline-playback.txt](https://github.com/thesofproject/sof/files/4223511/multiple-pipeline-playback.txt)

jdrzaszcz, 2020-02-19:
I run test 10_01_TestPtpParallelInternal8000Hz16b16b1ch with playback_iterations=1000 on APL RVP and couldn't reproduse this bug.
Above test runs simultaneous playback and capture on three pipelines connected do SSP interfaces.
I will try with parallel exteralnal.
Parallel exteralna test runs simultaneous playback and capture on multiple pipelines connected do various interfaces (SSP loopback, SSP master/slave, DMIC)

tlauda, 2020-02-19:
@jdrzaszcz It's playback and capture on HDA Link, not SSP.

jdrzaszcz, 2020-02-19:
@tlauda As far as I know  APL UP2 don't have HDA.

tlauda, 2020-02-19:
@jdrzaszcz `hda_dma_link_check_xrun() error: xrun detected` it says here that XRUN happened on HDA Link.

jdrzaszcz, 2020-02-19:
@tlauda Thank for the explanation.

jdrzaszcz, 2020-02-19:
In our tests this bug is not reproducible and we can't do it with DAI HDA, because we don't have the proper setup.

lgirdwood, 2020-02-19:
We need to role this into EDF scheduler work i.e. how to correctly reschedule and start after xrun. Lets close now.

closed 2020-02-19 by lgirdwood