thesofproject/sof #2298: [BUG][cml-008-patch-backport] switch recording format from s16_le to s24_le on wov pipeline while speaker playback causes dsp panic

aiChaoSONG, 2020-01-17:
**Describe the bug**
while speaker playback, if we run arecord with s16_le via wov pipeline for several times, then use s24_le run wov pipeline, there will be dsp panic.

**To Reproduce**
1.	Reboot
2.	Play sound via speaker: aplay -Dhw:0,5 -f dat -c2 -i -vvv ./somewave.wav
3.	Run arecord -Dhw:0,8 -M -N -f s16_le -r 16000 -c 2 -i -vvv -d 4 –buffer-size=68000 ./wov.wav, and trigger recording
4.	Run step 3 for 6 times
5.	Run arecord -Dhw:0,8 -M -N -f s24_le -r 16000 -c 2 -i -vvv -d 4 –buffer-size=68000 ./wov.wav, and trigger recording, you should see the issue.

**Reproduction Rate**
almost every time try 6 times with s16_le, then switch to s24_le, we see this issue.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: chrome-4.19-backport {4840bc29}
    * SOF: cml-008-patch-bakcport {7a84205}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {cml chromebook with rt1011 & rt5682 codecs}

[trace_logger_switch_format.log](https://github.com/thesofproject/sof/files/4075595/trace_logger_switch_format.log)
[error_logger_switch_format.log](https://github.com/thesofproject/sof/files/4075596/error_logger_switch_format.log)
[dmesg_switch_format.log](https://github.com/thesofproject/sof/files/4075597/dmesg_switch_format.log)

lgirdwood, 2020-01-17:
@mrajwa @keyonjie not sure if it's related, but we do get a KPB allocation error
``
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM          81683566.718750  81683568.000000      src/lib/alloc.c:273 	error: 155 blocks needed for allocation but only 154 blocks are remaining
    0      2      KEYWORD         147849475.104167  66165908.000000 /audio/detect_test.c:128 	notify_host()
    0      2      KEYWORD         147849519.635417        44.531250 /audio/detect_test.c:105 	notify_kpb(), preamble: 33600
    0      2          KPB         147849526.354167         6.718750      src/audio/kpb.c:792 	kpb_event_handler(): received event with ID: 2 
    0      2          KPB         147849531.562500         5.208333      src/audio/kpb.c:891 	kpb_init_draining(): requested draining of 2100 [ms] from history buffer
    0      2          KPB         147849536.614583         5.052083      src/audio/kpb.c:984 	kpb_init_draining: unlimited draining speed selected.
    0      2          KPB         147849540.937500         4.322917      src/audio/kpb.c:987 	kpb_init_draining(), schedule draining task
    0      2          KPB         147849556.197917        15.260417      src/audio/kpb.c:1044	kpb_draining_task(), start.
    0      2        POWER         147849561.770833         5.572917 avs/lib/pm_runtime.c:81  	pm_runtime_enable_dsp dsp_d0_sref 2
```

mrajwa, 2020-01-17:
@lgirdwood, allocation error log is just a misleading warning which really says "there is not enough memory for one of the requested memory pool" and since we don't see concecutive kpb_prepare(0 error message saying sth like "allocation of histor buffer failed" it means that the next memory pool was sufficient to fullfill the request.... I have to go back to https://github.com/thesofproject/sof/pull/2058 and pick some approach to hide this error message which is in fact a warning.

lgirdwood, 2020-01-17:
@mrajwa fwiw, @mmaka1 is thinking over changes to future trace APIs

mrajwa, 2020-01-27:
@aiChaoSONG, I have tested cml-008-patch-bakcport as well as master branches and I admit this bug is on both of them. This is very rare case but after debugging it a come up with a fix. Therefore could you please test below patch and check if it fixes the isssue:

https://github.com/thesofproject/sof/pull/2329

aiChaoSONG, 2020-02-12:
@mrajwa Are you sure #2329 fix this issue？ since #2329 cannot directly applied to 008-backport, I port it myself, but only get ipc error when run wov pipeline. if I am porting it wrong, can you  provide the fw you compiled?

mengdonglin, 2020-03-17:
@aiChaoSONG @mrajwa Since #2329 is merged by master, customer can get the fix from master later. Closed this issue on cml-008-patch-backport as won't fix.

closed 2020-03-17 by mengdonglin