thesofproject/sof #2300: [BUG][ICL][TGL][WHL] - Stream stall in mixer topology

zrombel, 2020-01-17:
**Describe the bug**
Test is creating topology containing 3x Host-Buffer piplines connected to mixer pipeline. After sending trigger_start to host pipelines data is not copied through playback pipelines. 
Bug is audio format independent.

**Topology**
```
+---------------------    ------------------------------+
| +------+   +-----+ |    | +-----+   +-----+   +-----+ |   +----+
| | Host |-->| Buf |----->| | Mix |-->| Buf |-->| Dai |---->|SSPx|
| +------+   +-----+ |    | +-----+   +-----+   +-----+ |   +----+
+---------------------    ----^^------------------------+    |  |
                              ||                             |  |
+--------------------+        ||                             |  |
| +------+   +-----+ |        ||                             |  |
| | Host |-->| Buf |----------+|                             |  |
| +------+   +-----+ |         |                             |  |
+--------------------+         |                             |  |
                               |                             |  |
+--------------------+         |                             |  |
| +------+   +-----+ |         |                             |  |
| | Host |-->| Buf |-----------+                             |  |
| +------+   +-----+ |                                       |  |
+--------------------+                                       |  |
                                                             |  |
+----------------------------------------+                   |  |
| +------+   +-----+             +-----+ |                  +----+
| | Host |<--| Buf |<------------| Dai |<-------------------|SSPx|
| +------+   +-----+             +-----+ |                  +----+
+----------------------------------------+
```

**To Reproduce**
Run Python test incorporating slim driver:
tests/sof/fw_12_mixer/test_00_mixer_playback.py::TestMixerPlayback48000Hz32b32b2ch with --playback_iterations=100

**Reproduction Rate**
25%

**Environment**
* Python tests: branch Master hash f869b5d63
* SOF FW: branch Master hash 8e9df078 and earlier

**Logs**
[MixerStreamStallBugTGL.zip](https://github.com/thesofproject/sof/files/4077144/MixerStreamStallBugTGL.zip)
[MixerStreamStallBugICL.zip](https://github.com/thesofproject/sof/files/4077147/MixerStreamStallBugICL.zip)

lgirdwood, 2020-01-27:
@zrombel moving to v1.6 since we have no users for this topology yet.

jdrzaszcz, 2020-02-24:
I checked it with 1d273c3a9f155624144199e36a14ffafeec345ca commit and don't see this issue after updating piplines prioritys.

closed 2020-02-24 by jdrzaszcz