thesofproject/sof #2311: [FEATURE] Refactor include/sof/drivers

jajanusz, 2020-01-22:
**Is your feature request related to a problem? Please describe.**
Headers in include/sof/drivers should be platform-agnostic, however there are symbols specific for platform/vendor (f.e. in ssp/mn/mu). These files are included by all platforms so only generic symbols should be there.

**Describe the solution you'd like**
Non-platform-agnostic symbols should be moved to `platform/*` folders and just be included by top-level headers.

**Additional context**
Mentioned in #2308

lgirdwood, 2020-08-18:
@jajanusz any update on the progress here ? Are we good now or is there more todo ?

paulstelian97, 2020-08-31:
I have identified a few todos in i.MX specific driver headers so keeping it open. Fixes should be relatively trivial so I'll try to do them today.

paulstelian97, 2020-08-31:
#3368 should fix it for i.MX headers once it's in a good shape.

lgirdwood, 2020-09-07:
@jajanusz I think we are good on Intel side too now ? Or are there still some items still TODO ?

lgirdwood, 2020-09-17:
moved to v1.7 since it depends on https://github.com/thesofproject/sof/pull/3368

lgirdwood, 2021-01-12:
Move to v1.9 as there is impact with Zephyr work.

kv2019i, 2023-02-15:
@paulstelian97 @lgirdwood I'd propose to close this as we are now moving drivers to Zephyr side. I'll park this to TBD milestone until then.

closed 2024-04-09 by kv2019i