thesofproject/sof #2320: [BUG] Mixed channnels on sdw_alh_loopack_volume

jdrzaszcz, 2020-01-23:
**Description**
When running simultaneous playback and capture on SdW interface on internal loopback with volume components on both pipelines after few iteration there are mixed channels on recorded files. It heppend on 8 channels audio formats.
When running the same test without volume components this problem didn't occur.

**Topology**
```
                pipe_plb
                +----------------------------------+
+----------+    | +------+   +-----+   +---------+ |     +-------+      +------+
| Host plb |----->|Buffer|-->| VOL |-->| Dai ALH | |---->| PDI A | ---> | DP A |
+----------+    | +------+   +-----+   +---------+ |     +-------+      +------+
                +----------------------------------+                       ||
                                                                           ||
                pipe_cap                                                   ||
                +----------------------------------+                       ||
+----------+    | +------+   +-----+   +---------+ |     +-------+      +------+
| Host cap |<-----|Buffer|<--| VOL |<--| Dai ALH | |<----| PDI B |<---- | DP B |
+----------+    | +------+   +-----+   +---------+ |     +-------+      +------+
                +----------------------------------+
```
**To Reproduce**
Slim Driver python test:
On WHL run test:
e.g. 16_01_TestAlhLoopbackVolume48000Hz16b16b8ch with --playback_iterations>=10

**Reproduction Rate**
100 %

**Expected behavior**
Channales shouldn't be mixed on recorded files.

**Environment**
Commit: b00869e204e5ad6b4c1a0e6cddea6b03e992e2b5 on SOF master branch

**Example log for topology with volume component and without this component in attachment**
[16_00_TestSdwAlhLoopback48000Hz16b16b8ch.zip](https://github.com/thesofproject/sof/files/4102102/16_00_TestSdwAlhLoopback48000Hz16b16b8ch.zip)
[16_01_TestAlhLoopbackVolume48000Hz16b16b8ch.zip](https://github.com/thesofproject/sof/files/4102103/16_01_TestAlhLoopbackVolume48000Hz16b16b8ch.zip)

lgirdwood, 2020-01-23:
@singalsu any comment here ?
@jdrzaszcz I guess this is working when chans < 8 ?

jdrzaszcz, 2020-01-23:
Yes, when less than 8 channels, recorded files looks fine.

tlauda, 2020-02-26:
@lgirdwood @jdrzaszcz Issue happens also without volume with simple host - dai pipeline. It's rather related to SDW configuration on host side and has nothing to do with FW. Someone with better knowledge on SDW should look into configuration done by test framework.

lgirdwood, 2020-02-26:
@plbossart @ranj063 any comments here ?

plbossart, 2020-02-26:
Can you clarify what 'internal loopback' refers to? Is this a loopback inside the Cadence IP (or ALH), or do you push data on the bus with a data port and capture it with another?

I would also recommend that you use test modes such as Logic1 and PRBS to look at data integrity issues. That's precisely for this sort of integration that we added them in the standard :-)

jdrzaszcz, 2020-02-27:
This is test with ALH loopback.

zrombel, 2020-05-25:
I had some time to look at the problem and will summarize it a little.
This bug only shows on audio formats with 8 channels, namely:
8000Hz 16b/16b 8ch
8000Hz 24b/32b 8ch
8000Hz 32b/32b 8ch
16000Hz 16b/16b 8ch
16000Hz 24b/32b 8ch
16000Hz 32b/32b 8ch
24000Hz 16b/16b 8ch
24000Hz 24b/32b 8ch
24000Hz 32b/32b 8ch
48000Hz 16b/16b 8ch
48000Hz 24b/32b 8ch
48000Hz 32b/32b 8ch
For other formats with channel count less than 8 bug doesn't occurs.
Channel swapping cannot be seen on first iteration. Most commonly the pattern is:
1 iteration - no bug
2 iteration - channel swapped
3 iteration - no bug 
4 iteration - channel swapped
... and so on
By iteration we understand situation where fw stream is created, started, stopped and freed, and then again created, started, stopped and freed. I have looked through registers that we are configuring and don't see any missing or incorrectly set values. 
Only case when I'm able to force it to work is to switch off and on SdW by writing to SNDWLCTL shim register.

And a little more info about loopback used in python tests - it is done by defining soundwire frame able to contain audio sample and configure two data ports - playback a capture - to point at the same place in frame.

@plbossart any ideas where too look for any errors in SdW registers configurations? I can prepare register dump if it would be helpful. Could you please provide some more info about Logic1 and PRBS modes that you mentioned?

plbossart, 2020-05-26:
@zrombel I still don't have a good understanding of what loopback you are enabling. What registers are you playing with?

For the PRBS stuff, the code is enabled w/ https://github.com/thesofproject/linux/pull/2117

zrombel, 2020-05-27:
@plbossart  Thank for the PR, will look into it.

Ok, so I'm pasting all register values that we are setting. 

Audio format: 48kHz 32b/32b 8ch 
Playback port:  SdW Instance: 0, PDI 2, DP 1
Capture port: SdW Instance: 0, PDI 3, DP 2
SdW Frame: Columns: 4 (index 1), Rows: 100 (index 9)
Platform: ICL

```
=> SHIM REGISTERS
	 SNDWLCTL                                : 0x00000F0F
	 SNDWSYNC                                : 0x0000176F
	 SNDW0IOCTL                              : 0x00000041
	 SNDW0ACTMCTL                            : 0x0000000B
	 SNDW1IOCTL                              : 0x00000041
	 SNDW1ACTMCTL                            : 0x0000000B
	 SNDW2IOCTL                              : 0x00000041
	 SNDW2ACTMCTL                            : 0x0000000B
	 SNDW3IOCTL                              : 0x00000041
	 SNDW3ACTMCTL                            : 0x0000000B
	 SNDW0PCM2CM                             : 0x00000770
	 SNDW0PCM3CM                             : 0x00008870
=> LINK HUB REGISTERS
	 STRM7CFG                                : 0x00070003
	 STRM8CFG                                : 0x00070003
=> MASTER REGISTERS
	 MCP_Config Instance 0                   : 0x0F1F0000
	 MCP_Control Instance 0                  : 0x00000702
	 MCP_SSPCtrl Instance 0 Bank0            : 0x00000018
	 MCP_SSPCtrl Instance 0 Bank1            : 0x00000018
	 MCP_ClockCtrl Instance 0 Bank0          : 0x00000001
	 MCP_ClockCtrl Instance 0 Bank1          : 0x00000001
	 MCP_Control Instance 0                  : 0x00000702
	 MCP_SlaveIntMask0 Instance 0            : 0xFFFFFFFF
	 MCP_SlaveIntMask1 Instance 0            : 0x0000FFFF
	 MCP_FIFOLevel Instance 0                : 0x00000001
	 MCP_FrameShapeInit Instance 0           : 0x00000049
=> PDI DP Playback
	 DP_Config Instance 0 Bank 1 DP 1        : 0x00001F00
	 DP_ChannelEn Instance 0 Bank 1 DP 1     : 0x000000FF
	 DP_OffsetCtrl Instance 0 Bank 1 DP 1    : 0x00000000
	 DP_SampleCtrl Instance 0 Bank 1 DP 1    : 0x0000018F
	 DP_HCtrl Instance 0 Bank 1 DP 1         : 0x00000013
	 DP_Port_Ctrl Instance 0 Bank 1 DP 1     : 0x00000000
	 PDI_Config Instance 0 PDI 2             : 0x0000FF01
=> PDI DP Capture
	 DP_Config Instance 0 Bank 1 DP 2        : 0x00001F00
	 DP_ChannelEn Instance 0 Bank 1 DP 2     : 0x000000FF
	 DP_OffsetCtrl Instance 0 Bank 1 DP 2    : 0x00000000
	 DP_SampleCtrl Instance 0 Bank 1 DP 2    : 0x0000018F
	 DP_HCtrl Instance 0 Bank 1 DP 2         : 0x00000013
	 DP_Port_Ctrl Instance 0 Bank 1 DP 2     : 0x00000001
	 PDI_Config Instance 0 PDI 3             : 0x0000FF02
```

jdrzaszcz, 2020-08-25:
@plbossart @zrombel This issue occurred with dsm (smart_amp) component where topology with 8 channels is used.

Also with to  SDW ALH internal loopbacks, channels are swapped for audio formats with 8 channels. This is seen in the atomic dai test where two ALH loopback are used to verify dai synchronisation,.

plbossart, 2021-06-10:
Where do we stand on this, the last update was in August 2020? @jdrzaszcz @zrombel @slawblauciak @lgirdwood

lgirdwood, 2021-06-11:
@mwasko fyi - we seem to have a lot of old issues that can be closed once confirmed fixed. @slawblauciak can we go over and close all old issue at next scrub ?