thesofproject/sof #2330: [FEATURE] Add a driver for SDMA controller found on i.MX8MP

dbaluta, 2020-01-28:
**Is your feature request related to a problem? Please describe.**
SDMA controller can be found on some i.MX NXP boards. We now target a SOF driver for i.MX8MP.

There is already a driver in Linux for SDMA that might be used as an example:
https://elixir.bootlin.com/linux/v5.5/source/drivers/dma/imx-sdma.c

**Describe the solution you'd like**
The driver should implement DMA API from SOF.

**Describe alternatives you've considered**
We thought about using EDMA which theoretically could work, but nevertheless we should SDMA because of the hardware configuration.

lgirdwood, 2020-01-28:
@dbaluta given the differences between imx8 and imx8mp I think it may justify it's own row in the platforms table in sof-docs. That way it should be easier to differentiate if we add new columns for other features etc.

dbaluta, 2020-01-28:
@lgirdwood I was waiting for toolchain support for imx8m (imx8mp is just one instance of imx8m) to be merged and then will update the document. So, far there is no mention of imx8m in the sof-docs (as far as i remember).

dbaluta, 2020-01-28:
@lgirdwood I remember it wrong. Actually, documentation for i.MX8M was already merged. Anyhow, let my try to look for platforms table in sof-docs, I don't know where it can be found :D

dbaluta, 2020-01-28:
@lgirdwood found it will send a patch, asap.

dbaluta, 2020-01-28:
@lgirdwood done. https://github.com/thesofproject/sof-docs/pull/168

closed 2020-08-04 by dbaluta