thesofproject/sof #2342: [FEATURE] Developer guide document for testbench development

lgirdwood, 2020-01-31:
**Is your feature request related to a problem? Please describe.**
A simple document describing what the test bench is, why it is useful and how to use it. i.e all the steps needed with some example where users can see it working..

lgirdwood, 2020-01-31:
I would also add context here about how this also allow GDB and valgrind usage too, i.e. since SOF memory resource calls are mapped to malloc and friends then valgrind can detect leaks, out of bounds access etc

singalsu, 2020-02-03:
Yep, that's very useful.

lgirdwood, 2020-08-18:
@singalsu are we good, can we close now ?

singalsu, 2020-09-01:
@lgirdwood Sorry this is not yet done, this is open and should be kept until I've done this.

lgirdwood, 2020-09-17:
@singalsu ping

singalsu, 2020-09-21:
The PR https://github.com/thesofproject/sof-docs/pull/286 is not yet merged. It has the content that was discussed here so I expect only grammar improvements needed.

lgirdwood, 2020-09-21:
@singalsu it should be merged soon by @deb-intel
@juimonen can you approve too. Thanks.

deb-intel, 2020-09-21:
@lgirdwood @singalsu Yes, I will merge / edit PR 286 (and 248) today.

closed 2020-09-23 by lgirdwood