thesofproject/sof #2347: [BUG][WHL][CML][DMIC] IPC timeout in tests incorporating DMIC interface

zrombel, 2020-02-03:
**Describe the bug**
Bug occurs on WHL platform, in tests with DMIC interface.
IPC timeout happens when sending IPC **DaiConfig** or **PcmParams** to FW.
No Trace Logs are received, memory window debug might be needed.
After bisect commit bce7f241d293d65d0fd38d280fb577b162621daa is found to introduce the bug.

**To Reproduce**
Any Python scripts DMIC test with option --iterations=100

**Reproduction Rate**
25%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Python script tests: 99e665ecf5d62a9710875fd4163e09b85b705009
    * SOF: bce7f241d293d65d0fd38d280fb577b162621daa 
2) Example test: 05_06_TestDmicBothFifoSimpleCapture16000hz16b16b2ch --iterations=100

**Logs**
[05_06_TestDmicBothFifoSimpleCapture48000hz32b32b2ch.zip](https://github.com/thesofproject/sof/files/4147614/05_06_TestDmicBothFifoSimpleCapture48000hz32b32b2ch.zip)
[05_06_TestDmicBothFifoSimpleCapture16000hz16b16b2ch.zip](https://github.com/thesofproject/sof/files/4147615/05_06_TestDmicBothFifoSimpleCapture16000hz16b16b2ch.zip)

lgirdwood, 2020-02-12:
I think this is a duplicate of #2265 
This also appears to happen with SSP DAI config too (taken from recent CI failures on APL).
```
src/ipc/handler.c:556 	ipc: dai 1.0 -> config 
    0      2          DAI 1.8        353178.385417         4.270833      src/audio/dai.c:680 	dai_config() dai 1.0
    0      2          DMA            353182.135417         3.750000 src/drivers/dw/dma.c:182 	dw_dma_channel_get(): dma 0 request channel 0
    0      2          SSP            353190.729167         8.593750 ers/intel/cavs/ssp.c:140 	ssp_set_config(), config->format = 0x4001
    0      2          SSP            353201.666667        10.937500 ers/intel/cavs/ssp.c:619 	ssp_set_config(), sscr0 = 0xc1c00f7f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2100000
    0      2          SSP            353204.843750         3.177083 ers/intel/cavs/ssp.c:622 	ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x03030000, ssioc = 0x00000020
    0      2          SSP            353207.395833         2.552083 ers/intel/cavs/ssp.c:624 	ssp_set_config(), ssrsa = 0x00000003, sstsa = 0x00000003
    0      2          DAI 2.14       353210.833333         3.437500      src/audio/dai.c:680 	dai_config() dai 1.0
    0      2          DMA            353213.854167         3.020833 src/drivers/dw/dma.c:182 	dw_dma_channel_get(): dma 1 request channel 0
    0      2          SSP            353220.833333         6.979167 ers/intel/cavs/ssp.c:140 	ssp_set_config(), config->format = 0x4001
    0      2          SSP            353228.802083         7.968750 ers/intel/cavs/ssp.c:619 	ssp_set_config(), sscr0 = 0xc1c00f7f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2100000
    0      2          SSP            353231.822917         3.020833 ers/intel/cavs/ssp.c:622 	ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x03030000, ssioc = 0x00000020
    0      2          SSP            353234.375000         2.552083 ers/intel/cavs/ssp.c:624 	ssp_set_config(), ssrsa = 0x00000003, sstsa = 0x00000003
    0      2          IPC            353281.250000        46.875000    src/ipc/handler.c:556 	ipc: dai 1.0 -> config 
    0      2          DAI 1.8        353284.895833         3.645833      src/audio/dai.c:680 	dai_config() dai 1.0
    0      2          SSP            353295.416667        10.520833 ers/intel/cavs/ssp.c:140 	ssp_set_config(), config->format = 0x4001
    0      2          SSP            353303.854167         8.437500 ers/intel/cavs/ssp.c:619 	ssp_set_config(), sscr0 = 0xc1c00f7f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2100000
    0      2          SSP            353322.447917        18.593750 ers/intel/cavs/ssp.c:140 	ssp_set_config(), config->format = 0x4001
    0      2          SSP            353330.468750         8.020833 ers/intel/cavs/ssp.c:619 	ssp_set_config(), sscr0 = 0xc1c00f7f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2100000
    0      2          SSP            353333.333333         2.864583 ers/intel/cavs/ssp.c:622 	ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x03030000, ssioc = 0x00000020
    0      2          SSP            353335.989583         2.656250 ers/intel/cavs/ssp.c:624 	ssp_set_config(), ssrsa = 0x00000003, sstsa = 0x00000003
```
The funny thing is that we iterate over the same component twice here ? @tlauda list corruption ?
```c
int ipc_comp_dai_config(struct ipc *ipc, struct sof_ipc_dai_config *config)
{
	bool comp_on_core[PLATFORM_CORE_COUNT] = { false };
	struct sof_ipc_comp_dai *dai;
	struct sof_ipc_reply reply;
	struct ipc_comp_dev *icd;
	struct list_item *clist;
	int ret = -ENODEV;
	int i;

	/* for each component */
	list_for_item(clist, &ipc->comp_list) {
		icd = container_of(clist, struct ipc_comp_dev, list);
		/* make sure we only config DAI comps */
		if (icd->type != COMP_TYPE_COMPONENT) {
			platform_shared_commit(icd, sizeof(*icd));
			continue;
		}

		if (!cpu_is_me(icd->core)) {
			comp_on_core[icd->core] = true;
			ret = 0;
			platform_shared_commit(icd, sizeof(*icd));
			continue;
		}

		if (icd->cd->comp.type == SOF_COMP_DAI ||
		    icd->cd->comp.type == SOF_COMP_SG_DAI) {
			dai = (struct sof_ipc_comp_dai *)&icd->cd->comp;
			platform_shared_commit(icd, sizeof(*icd));
			/*
			 * set config if comp dai_index matches
			 * config dai_index.
			 */
			if (dai->dai_index == config->dai_index &&
			    dai->type == config->type) {
				ret = comp_dai_config(icd->cd, config);
				platform_shared_commit(icd, sizeof(*icd));
				if (ret < 0)
					break;
			}
		}
	}

	if (ret < 0) {
		trace_ipc_error("ipc_comp_dai_config() error: "
				"comp_dai_config() failed");
		return ret;
	}

	/* message forwarded only by master core */
	if (!cpu_is_slave(cpu_get_id())) {
		for (i = 0; i < PLATFORM_CORE_COUNT; ++i) {
			if (!comp_on_core[i])
				continue;

			ret = ipc_process_on_core(i);
			if (ret < 0)
				return ret;

			/* check whether IPC failed on slave core */
			mailbox_hostbox_read(&reply, sizeof(reply), 0,
					     sizeof(reply));
			if (reply.error < 0)
				/* error reply already written */
				return 1;
		}
	}

	return ret;
}
```

zrombel, 2020-02-12:
Seems like bug no longer occurs.
Fix found in commit 9985b014d24af2781420de183ab648b9327b6724 - no error in 400 iterations.
Confirming on top master 7e88cc01f4c5feabf148aefeb032d82d94264e50 - no error in 400 and still running up to 1000 iterations. I will give update tomorrow when test are over.

lgirdwood, 2020-02-13:
@zrombel ok, thanks for checking. I will keep checking CI too.

zrombel, 2020-02-13:
Confimed fix on top master 7e88cc0 - no error in 1000 iterations. Funny thing is that commit 9985b01 only touches traces, is it possible that bug still exists but is well hidden ? @lgirdwood what do you thing about closing the bug?

lgirdwood, 2020-02-13:
@xiulipan @mengdonglin any objections ?

xiulipan, 2020-02-14:
@lgirdwood @lgirdwood 
This looks like the race condition with APL NOCODEC, same error with HW PARAMS

> IPC timeout happens when sending IPC DaiConfig or PcmParams to FW.

This bug will also influence WHL 

@zrombel are you testing with XCC FW or GCC FW?

zrombel, 2020-02-14:
@xiulipan I'm testing only XCC builds

tlauda, 2020-02-14:
@lgirdwood @xiulipan I've rootcaused this issue to too small default stack size for EDF tasks. Here is the fix: https://github.com/thesofproject/sof/pull/2388. Let's see if it will also help with APL failures.

xiulipan, 2020-02-16:
@tlauda Thanks for the fix, it seems the issue on APL is gone.

lgirdwood, 2020-02-17:
@zrombel can we close ?

zrombel, 2020-02-18:
I will run tests once again on top of master and if the bug doesn't reproduce I will close the bug.

zrombel, 2020-02-18:
No reproduction on master top 53d35. Closing Issue.

closed 2020-02-18 by zrombel