thesofproject/sof #2353: [BUG] WHL SSP Stream stall in KD tests

kfialkox, 2020-02-06:
**Describe the bug**
In KD tests - Keyword detection notification timeout

**To Reproduce**
07_02_TestKdSsp16000Hz24b32b2ch 
topology_iterations=10
playback_iterations=10

**Reproduction Rate**
1 on 80

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: 0469a018ba856b0fd96b042c4011301e988d0f8f
    * python tests: eba84c33043558f16dad27a118b1e8cb8792134c 

**Logs**

[logs.zip](https://github.com/thesofproject/sof/files/4159870/logs.zip)

lgirdwood, 2020-02-06:
@kfialkox could you paste the appropriate parts of the log here. i.e. just a the trace section that shows the fault. This helps with discussion as new reviewers dont have to keep opening and searching the zip file.

zrombel, 2020-02-12:
Bug cannot be reproduced neither on hash 0469a01 nor on current master. It probably was a platform issue. Closing issue.

closed 2020-02-12 by zrombel