thesofproject/sof #2360: [BUG] Keyword detection timeout in D0iX

wwittbrx, 2020-02-07:
**Describe the bug**
Keyword detection timeout in slim driver tests with D0iX flow.

It is caused by lost notification from detector component. Detector itself works ok, and triggers the draining (see wave screenshot, it looks as expected), but the notification does not arrive to driver, which causes the test to time out.

[image]

The reproduction rate is 3/500 on WHL RVP platform and 23/500 on CML RVP from my tests.
There are no obvious errors in the log. I have included successful log and timeout log below.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: d6fef

[dma_trace_timeout.log](https://github.com/thesofproject/sof/files/4171225/dma_trace_timeout.log)
[dma_trace_success.log](https://github.com/thesofproject/sof/files/4171226/dma_trace_success.log)

lgirdwood, 2020-02-07:
@wwittbrx there are some pending PRs to be merged for keyword. Could you retest with @mrajwa latest unmerged fixes.

keyonjie, 2020-02-10:
Hi @wwittbrx I believe you keep the x86 system in S0 state when performing the test, right?
Can the similar issue being reproduced with Linux driver? if so, we can debug it in Shanghai at the same time. @aiChaoSONG

slawblauciak, 2020-02-10:
The reproduction rate is low and requires specific conditions, which are most likely can't occur under full stack. The fix would be nice-to-have as it impacts CI.

amerysong, 2020-02-12:
@wwittbrx could you please detailed list the test procedures？and is there any script i can use？ because the reproduction rate is low， manually trigger is really time consuming @keyonjie

aiChaoSONG, 2020-02-12:
@wwittbrx could you please detailed list the test procedures？and is there any script i can use？ because the reproduction rate is low， manually trigger is really time consuming @keyonjie

wwittbrx, 2020-03-03:
Issue not reproduced on master with notification fixes and improvements in test flow, closing.

closed 2020-03-03 by wwittbrx