thesofproject/sof #2361: [BUG] left/right speaker test backwards for apl-up2-pcm512x

stripes416, 2020-02-08:
**Describe the bug**
When running ./test-speaker.sh on apl-up2-pcm512x I noticed that the 'front left' & 'front right' audio is coming out of the wrong side of the headset. The audio is clear, just the sides are reversed.

**To Reproduce**
With headset inserted, run the sof-test/test-speaker.sh via ./test-speaker.sh.

**Reproduction Rate**
10/10

**Expected behavior**
Audio should come out of the expected side of the headset.

**Impact**
minor issue.

**Environment**
*confirmed issue today wasn't a problem with my particular headset, but has been around for a while, so not a new issue
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 4cef169b
    * SOF: d6fef4e4
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL-UP2-pcm512x

[cmd.txt](https://github.com/thesofproject/sof/files/4173491/cmd.txt)
[testing_log.txt](https://github.com/thesofproject/sof/files/4173487/test-speaker-log_apl-up2-pcm512x_5.5.0-rc7-daily-default-20200207_speaker-test.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/4173484/dmesg_apl-up2-pcm512x_5.5.0-rc7-daily-default-20200207_speaker-test.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4173492/slogger.txt)

lgirdwood, 2020-02-10:
@stripes416 I'm guessing this is good with all other APL boards ? Can you check if the SSP topology config is that same as a working board. If not, it could be the SSP mode config, if same it could be a codec setting that's swapping channels. @bardliao @plbossart @tlauda fyi

bardliao, 2020-02-10:
@stripes416 I just checked it on my side and I didn't see the issue with my Up2 board. Your headset is actually connected to a Y cable and the Y cable is connected to the PCM512 board, right? Could you double check is the Y cable connected to the PCM512 board in the same color? ie. red to red and white to white.

stripes416, 2020-02-10:
@bardliao -- @fredoh9 confirmed with his apl-up2-pcm512x that this is an issue with his as well. We connect our headset directly to the codec itself.

[image]

bardliao, 2020-02-11:
Looks our PCM512 board are not identical. Mine doesn't have jack connector. 
[image]
I confirmed that our kernel, SOF, and topology are the same. 
The only different I can imagine is mixer settings. Here is my result of "amixer contents"
[up2_amixer.txt](https://github.com/thesofproject/sof/files/4183906/up2_amixer.txt)
Could do the same on your side? So we can compare the mixer settings.

lgirdwood, 2020-02-11:
@stripes416 this could even be a PCB layout bug if it's working with @bardliao codec board. If codec settings look sane then check codec pin directly i.e. play left only audio and use scope on pin.

stripes416, 2020-02-18:
@bardliao here is amixer contents from my apl-up2-pcm512x.
[kledford_up2_amixer_contents.txt](https://github.com/thesofproject/sof/files/4221745/kledford_up2_amixer_contents.txt)

(sorry for the time delay, was out of town for a few days)

bardliao, 2020-02-19:
The mixer settings are the same except HDMI settings. So, it is not an amixer setting issue. As @lgirdwood mentioned, we should think about the possibility of layout issue. But I think we can close the issue since there is nothing we can do.

lgirdwood, 2020-02-19:
@bardliao ack - now closing. It can be re-opened if this happens on other APL boards

closed 2020-02-19 by lgirdwood

stripes416, 2020-02-19:
sounds good. Thanks for looking into this guys!