thesofproject/sof #2371: [FEATURE] trace: make last trace item persistent

lgirdwood, 2020-02-12:
**Is your feature request related to a problem? Please describe.**
Currently only trace_error() messages are entered into the persistent buffer. This means we are unable to find the fault if FW crashes and DMA trace buffer is not flushed.

**Describe the solution you'd like**
Reserve the first item in the persistent trace buffer (error buffer) for the last trace message. This way on any FW crash the last trace item is always available regardless of DMA trace buffer copy status. i.e. we always write the current trace message to the top of the persistent buffer for every trace call.

paulstelian97, 2020-02-12:
I typically enable TRACEM so that the mailbox also receives the non-error traces, is that what you mean? (that one only gets the errors by default)

lgirdwood, 2020-02-12:
I mean here if TRACE_M is disabled, i.e. you need to retain errors and last trace message.

paulstelian97, 2020-02-12:
I see, so a form of limited TRACE_M even when TRACE_M itself is disabled.

lgirdwood, 2020-02-12:
yeah, ideally to fast track CI reports and any user bug reports.

lgirdwood, 2020-08-18:
@slawblauciak @ktrzcinx @bkokoszx any update here, I do remeber merging a similar feature ??? or shall we move to v1.7 ?

slawblauciak, 2020-08-18:
Not familiar with this feature I'm afraid.

ktrzcinx, 2020-08-18:
I don't remember any PR related to this feature.

lgirdwood, 2020-08-20:
ok, will move - I think we do see panic() in the error log now so maybe that's what I'm seeing.

lgirdwood, 2021-01-12:
@akloniex @ktrzcinx I think this can be closed now ?

ktrzcinx, 2021-01-25:
@lgirdwood I don't think so, without CONFIG_TRACEM, traces other than critical are stored only in dma buffer.

kv2019i, 2023-02-15:
Closing now, with Zephyr mtrace, the trace messages are stored in SRAM window that can (and is) queried.

closed 2023-02-15 by kv2019i