thesofproject/sof #2379: [BUG][BYT][RT5682]Ipc error 0X60010000 size 20 when try speaker test on media playback

dengyangchao, 2020-02-13:
**Describe the bug**
Ipc error 0X60010000 size 20 when try speaker test on media playback.
When issue occur, try aplay will occur I/O error, and only reboot can back it to normal.

**To Reproduce**
1.Reboot via "sudo reboot"
2.Speaker test via "speaker-test -Dhw:0,1 -r48000 -f1000 -c2 -t wav -l 10"

**Reproduction Rate**
5/5

**Expected behavior**
No error occur, and speaker test output normal

**Impact**
Occur ipc error

**Environment**
Platform: BYT with codec RT5682 in I2S mode
Firmware: 7e88cc01 https://github.com/thesofproject/sof/commits/master
Kernel: f19d23b4 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-byt-rt5682.tplg same as firmware

**Dmesg**

```
[   34.692088] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   35.162449] rfkill: input handler disabled
[ 3266.777104] sof-audio-acpi 80860F28:01: pcm: open stream 1 dir 0
[ 3266.777121] sof-audio-acpi 80860F28:01: period min 192 max 262144 bytes
[ 3266.777131] sof-audio-acpi 80860F28:01: period count 2 max 32
[ 3266.777140] sof-audio-acpi 80860F28:01: buffer max 8388608 bytes
[ 3266.785320] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[ 3266.785336] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[ 3266.785376] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[ 3266.785386] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[ 3266.785405] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[ 3266.785414] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[ 3266.785432] sof-audio-acpi 80860F28:01: pcm: hw params stream 1 dir 0
[ 3266.785448] sof-audio-acpi 80860F28:01: generating page table for 00000000f5e0b3a5 size 0x7ff800 pages 2048
[ 3266.785511] sof-audio-acpi 80860F28:01: stream_tag 0
[ 3266.785532] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 3266.785847] sof-audio-acpi 80860F28:01: error: ipc error for 0x60010000 size 20
[ 3266.785868] sof-audio-acpi 80860F28:01: error: hw params ipc failed for stream 0
[ 3266.785883] sof-audio-acpi 80860F28:01: ASoC: 80860F28:01 hw params failed: -12
[ 3266.785897]  Media Playback 1: ASoC: hw_params FE failed -12
[ 3266.785919] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[ 3266.787236] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
[ 3267.000850] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3267.000877] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
**logger-error-trace**
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          IPC         3265971510.208333              nan pc/ipc-host-ptable.c:163 	ipc: parse descriptors failed -12
```

[amixer.txt](https://github.com/thesofproject/sof/files/4197487/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/4197488/dmesg.log)
[sof-error-trace.log](https://github.com/thesofproject/sof/files/4197489/sof-error-trace.log)
[sof-logger.log](https://github.com/thesofproject/sof/files/4197490/sof-logger.log)

dengyangchao, 2020-02-13:
Old issue,  can run speaker test on HW:0,0 and output normal
when run on HW:0,1(media playback) cause ipc error

dengyangchao, 2020-02-17:
Issue also can be reproduced on BYT Nocodec

slawblauciak, 2020-02-18:
Is this a regression?

lgirdwood, 2020-02-18:
Duplicate of #2400 ?

dengyangchao, 2020-02-19:
> Duplicate of #2400 ?

@lgirdwood This issue is different with #2400 , found it when do speaker-test front left and front right on pipeline media playback

lgirdwood, 2020-02-19:
@dengyangchao there is no memory to allocate page descriptors ?
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          IPC         3265971510.208333              nan pc/ipc-host-ptable.c:163 	ipc: parse descriptors failed -12
```
caused here
```c
static int ipc_parse_page_descriptors(uint8_t *page_table,
				      struct sof_ipc_host_buffer *ring,
				      struct dma_sg_elem_array *elem_array,
				      uint32_t direction)
{
	int i;
	uint32_t idx;
	uint32_t phy_addr;
	struct dma_sg_elem *e;

	/* the ring size may be not multiple of the page size, the last
	 * page may be not full used. The used size should be in range
	 * of (ring->pages - 1, ring->pages] * PAGES.
	 */
	if ((ring->size <= HOST_PAGE_SIZE * (ring->pages - 1)) ||
	    (ring->size > HOST_PAGE_SIZE * ring->pages)) {
		/* error buffer size */
		trace_ipc_error("ipc_parse_page_descriptors() error: "
				"error buffer size");
		return -EINVAL;
	}

	elem_array->elems = rzalloc(RZONE_RUNTIME, SOF_MEM_CAPS_RAM,
				    sizeof(struct dma_sg_elem) * ring->pages);
	if (!elem_array->elems)
		return -ENOMEM; <<<<<<<<<<<<<<<<<<<<<<<<<<<<
	elem_array->count = ring->pages;
```
@mengdonglin can you assign - we need to also add an error trace message here too. The is no heap free with this block size.

ClarexZhou, 2020-03-05:
@xiulipan  will add an error trace message

xiulipan, 2020-03-05:
@lgirdwood PR send https://github.com/thesofproject/sof/pull/2487
Log dump on one BYT with speaker test
```
$ sudo sof-logger -l sof-byt.ldc
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          IPC         5434129892.916667              nan pc/ipc-host-ptable.c:49    ipc_parse_page_descriptors() error: Threre is no heap free with this block size: 24576
    0      1          IPC         5434129901.197917         8.281250 pc/ipc-host-ptable.c:168   ipc: parse descriptors failed -12
```

```
$ speaker-test -Dhw:0,1 -r48000 -f1000 -c2 -t wav -l 10

speaker-test 1.1.3

Playback device is hw:0,1
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 2096640
Period size range from 48 to 65520
Using max buffer size 2096640
Periods = 4
Unable to set hw params for playback: Cannot allocate memory
Setting of hwparams failed: Cannot allocate memory
````

The issue seems to be the  **max buffer size 2096640** too big. Can also reproduce by following aplay.

@singalsu @tlauda Is it OK for us to have such big `BUFFER_SIZE BUFFER_TIME BUFFER_BYTES` with SRC comp?
```
aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=2096640 --dump-hw-params
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
HW Params of device "hw:0,1":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 96000]
PERIOD_TIME: [250 8190000]
PERIOD_SIZE: [24 65520]
PERIOD_BYTES: [192 262080]
PERIODS: [2 32]
BUFFER_TIME: [500 262080000]
BUFFER_SIZE: [48 2096640]
BUFFER_BYTES: [192 8388480]
TICK_TIME: ALL
--------------------
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 1365000
PERIOD_SIZE: 65520
PERIOD_BYTES: 262080
PERIODS: 32
BUFFER_TIME: 43680000
BUFFER_SIZE: 2096640
BUFFER_BYTES: 8386560
TICK_TIME: 0
```

Normal volume pipeline hw params
```
$ aplay -Dhw:0,0 -f dat /dev/zero -vvv -i  --dump-hw-params
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
HW Params of device "hw:0,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: 48000
PERIOD_TIME: [500 85000]
PERIOD_SIZE: [24 4080]
PERIOD_BYTES: [192 16320]
PERIODS: [2 16]
BUFFER_TIME: [1000 341000]
BUFFER_SIZE: [48 16368]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL

```

lgirdwood, 2020-03-05:
@xiulipan we may need to factor this into the BYT topologies as max buffer size.  I dont think we have use case for deep buffering on BYT yet as we could work out the elems on the fly when needed (i.e. no pre-alloc and calc of elems).

mengdonglin, 2020-03-06:
@singalsu Please help to remove SRC pipeline from BYT topology. Thanks

lgirdwood, 2020-03-06:
@mengdonglin @singalsu @xiulipan lets fix the max buffer size issue first in topology before removing SRC.

mengdonglin, 2020-03-16:
Close this issue for max buffer size restriction of Baytrail. Will reopen when BUFFER_SIZE BUFFER_TIME BUFFER_BYTES for SRC compoment can be adjusted for Baytrail. @lrgirdwo @xiulipan @singalsu @dengyangchao

singalsu, 2020-03-16:
@lgirdwood @mengdonglin Sorry, I didn't understand from above if SRC should be changed or is it OK with other changes. 

Removing 44.1 kHz family from conversions and keeping only 8/16/32/48 kHz would allow SRC to work with same 2 periods buffers as other components. But the "mp3 44.1 kHz" is the most common use case for SRC so I don't know if there's any sense to reduce the BYT configuration even more.

lgirdwood, 2020-03-16:
@singalsu yes, please send a PR that reduces SRC to just 8/16/32/48 on BYT, CHT. It would also be good to put something in the Kconfig SRC help to state assumption on memory sizes and as an example 44.1kHz would not easily fit in BYT unless other baisc features were removed. Thanks.
@mengdonglin This can be closed after @singalsu has submitted his PR.

singalsu, 2020-03-16:
A fix proposal is in PR #2555. The 44.1 kHz playback still works but everything above 48 kHz is removed. Playback works on my T100 now. With git master it didn't work. Can you @dengyangchao please test with this PR if it helps!

dengyangchao, 2020-03-17:
@singalsu Since #2555 merged, tried with master df6c637.
No error observed on HW:0,0(low latency), found ipc error on HW:0,1(media playback) when try speaker-test

48K in media playback(HW:0,1): 
1. Playback can work well, but speaker-test still occur ipc error 0x60010000 
[image]
[48000-speaker-test.log](https://github.com/thesofproject/sof/files/4341348/48000-speaker-test.log)

44.1K in media playback(HW:0,1):
1. Playback can work well, but speaker-test still occur ipc error 0x60010000 then "kernel NULL pointer dereference"
[image]
[44100-speaker-test.log](https://github.com/thesofproject/sof/files/4341351/44100-speaker-test.log)

singalsu, 2020-03-17:
I get the same error with speaker-test. It doesn't look like firmware issue, nothing abnormal in the sof-logger output. There's even a kernel crash happening sometime after the IPC error.

lgirdwood, 2020-03-17:
@singalsu can you log the kernel crash.

dengyangchao, 2020-03-18:
Set buffer size to 2096640, then aplay via comand"aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=2096640 --dump-hw-params"
occur same ipc error
Also tried to set buffer size to 64K/32K/16K through below commands,  it work well, no error occur.
aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=64000 --dump-hw-params 
aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=32000 --dump-hw-params 
aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=16000 --dump-hw-params

[dmesg-16k.log](https://github.com/thesofproject/sof/files/4346643/dmesg-16k.log)
[dmesg-32k.log](https://github.com/thesofproject/sof/files/4346645/dmesg-32k.log)
[dmesg-64k.log](https://github.com/thesofproject/sof/files/4346675/dmesg-64k.log)
[dmesg-2096640.log](https://github.com/thesofproject/sof/files/4346677/dmesg-2096640.log)

```
ubuntu@sh-byt-mb-rt5682-1111:~$ aplay -Dhw:0,1 -f dat /dev/zero -vvv -i --buffer-size=2096640 --dump-hw-params
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
HW Params of device "hw:0,1":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 48000]
PERIOD_TIME: [500 8190000]
PERIOD_SIZE: [24 65520]
PERIOD_BYTES: [192 262080]
PERIODS: [2 32]
BUFFER_TIME: [1000 262080000]
BUFFER_SIZE: [48 2096640]
BUFFER_BYTES: [192 8388480]
TICK_TIME: ALL
--------------------
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 1365000
PERIOD_SIZE: 65520
PERIOD_BYTES: 262080
PERIODS: 32
BUFFER_TIME: 43680000
BUFFER_SIZE: 2096640
BUFFER_BYTES: 8386560
TICK_TIME: 0
```

singalsu, 2020-03-18:
> @singalsu can you log the kernel crash.

This kernel is not new (maybe two weeks old) but here's the dmesg:

```
[59545.402077] sof-audio-acpi 80860F28:00: pcm: trigger stream 0 dir 0 cmd 0
[59545.402092] sof-audio-acpi 80860F28:00: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[59545.402820] sof-audio-acpi 80860F28:00: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[59545.402851] sof-audio-acpi 80860F28:00: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[59545.402965] sof-audio-acpi 80860F28:00: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[59545.403112] sof-audio-acpi 80860F28:00: pcm: free stream 0 dir 0
[59545.403252] sof-audio-acpi 80860F28:00: pcm: close stream 0 dir 0
[59545.828594] sof-audio-acpi 80860F28:00: ipc rx: 0x90020000: GLB_TRACE_MSG
[59545.828616] sof-audio-acpi 80860F28:00: ipc rx done: 0x90020000: GLB_TRACE_MSG
[60166.909728] sof-audio-acpi 80860F28:00: pcm: open stream 1 dir 0
[60166.909734] sof-audio-acpi 80860F28:00: period min 192 max 262144 bytes
[60166.909736] sof-audio-acpi 80860F28:00: period count 2 max 32
[60166.909738] sof-audio-acpi 80860F28:00: buffer max 8388608 bytes
[60166.909991] sof-audio-acpi 80860F28:00: rate_min: 48000 rate_max: 48000
[60166.909993] sof-audio-acpi 80860F28:00: channels_min: 2 channels_max: 2
[60166.921755] sof-audio-acpi 80860F28:00: rate_min: 48000 rate_max: 48000
[60166.921761] sof-audio-acpi 80860F28:00: channels_min: 2 channels_max: 2
[60166.921774] sof-audio-acpi 80860F28:00: rate_min: 48000 rate_max: 48000
[60166.921778] sof-audio-acpi 80860F28:00: channels_min: 2 channels_max: 2
[60166.921789] sof-audio-acpi 80860F28:00: pcm: hw params stream 1 dir 0
[60166.921797] sof-audio-acpi 80860F28:00: generating page table for 00000000e7569615 size 0x7ff800 pages 2048
[60166.921824] sof-audio-acpi 80860F28:00: stream_tag 0
[60166.921835] sof-audio-acpi 80860F28:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[60166.922214] sof-audio-acpi 80860F28:00: error: ipc error for 0x60010000 size 20
[60166.922222] sof-audio-acpi 80860F28:00: error: hw params ipc failed for stream 0
[60166.922229] sof-audio-acpi 80860F28:00: ASoC: 80860F28:00 hw params failed: -12
[60166.922234]  Media Playback 1: ASoC: hw_params FE failed -12
[60166.922246] sof-audio-acpi 80860F28:00: pcm: free stream 1 dir 0
[60166.924339] sof-audio-acpi 80860F28:00: pcm: close stream 1 dir 0
[60166.983877] sof-audio-acpi 80860F28:00: ipc rx: 0x90020000: GLB_TRACE_MSG
[60166.983904] sof-audio-acpi 80860F28:00: ipc rx done: 0x90020000: GLB_TRACE_MSG
[60175.358390] general protection fault: 0000 [#1] SMP PTI
[60175.358414] CPU: 0 PID: 832 Comm: irqbalance Not tainted 5.5.0-rc5-test+ #34
[60175.358424] Hardware name: ASUSTeK COMPUTER INC. T100TA/T100TA, BIOS T100TA.314 08/13/2015
[60175.358445] RIP: 0010:string_nocheck+0x12/0x70
[60175.358460] Code: 00 00 4c 89 e2 be 20 00 00 00 48 89 ef e8 a6 8e 00 00 4c 01 e3 eb 81 90 49 89 f2 48 89 ce 48 89 f8 48 c1 fe 30 66 85 f6 74 4f <44> 0f b6 0a 45 84 c9 74 46 83 ee 01 41 b8 01 00 00 00 48 8d 7c 37
[60175.358471] RSP: 0018:ffffa8d00093fd30 EFLAGS: 00010086
[60175.358483] RAX: ffff99d9732d62ed RBX: ffff99d9732d7000 RCX: ffff0a00ffffff04
[60175.358493] RDX: 287c7287b7287a72 RSI: ffffffffffffffff RDI: ffff99d9732d62ed
[60175.358502] RBP: 287c7287b7287a72 R08: 0000000000000d15 R09: 0000000000000000
[60175.358512] R10: ffff99d9732d7000 R11: ffff99d9732d62eb R12: ffff0a00ffffff04
[60175.358521] R13: ffffffff985d397b R14: 0000000000000d15 R15: ffffffff985d397b
[60175.358533] FS:  00007f742ece5740(0000) GS:ffff99d973e00000(0000) knlGS:0000000000000000
[60175.358543] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[60175.358553] CR2: 00007f63a4020c50 CR3: 0000000071ee6000 CR4: 00000000001006f0
[60175.358560] Call Trace:
[60175.358578]  string+0x40/0x50
[60175.358596]  vsnprintf+0x33c/0x520
[60175.358614]  seq_vprintf+0x2b/0x40
[60175.358627]  seq_printf+0x4e/0x70
[60175.358645]  ? _copy_to_user+0x26/0x30
[60175.358661]  show_interrupts+0x232/0x320
[60175.358676]  seq_read+0x2e3/0x410
[60175.358696]  proc_reg_read+0x34/0x60
[60175.358714]  vfs_read+0x8c/0x140
[60175.358729]  ksys_read+0x54/0xd0
[60175.358745]  do_syscall_64+0x43/0x140
[60175.358762]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[60175.358776] RIP: 0033:0x7f742dde2260
[60175.358789] Code: 0b 31 c0 48 83 c4 08 e9 ae fe ff ff 48 8d 3d 27 b4 09 00 e8 b2 1e 02 00 66 90 83 3d e9 24 2d 00 00 75 10 b8 00 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 31 c3 48 83 ec 08 e8 5e de 01 00 48 89 04 24
[60175.358799] RSP: 002b:00007ffd445ee2d8 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[60175.358811] RAX: ffffffffffffffda RBX: 000000000103aa70 RCX: 00007f742dde2260
[60175.358820] RDX: 0000000000000400 RSI: 000000000103ad20 RDI: 0000000000000003
[60175.358830] RBP: 000000000000003a R08: 4950412d4f492020 R09: 2020202020202020
[60175.358839] R10: 2020202020202030 R11: 0000000000000246 R12: 00007ffd445ee368
[60175.358848] R13: 0000000000000000 R14: 000000000000003a R15: 0000000000000000
[60175.358858] Modules linked in: asix usbnet snd_soc_sst_bytcr_rt5640 intel_soc_dts_thermal intel_powerclamp snd_sof_acpi snd_sof_intel_byt snd_soc_acpi_intel_match snd_sof_intel_bdw snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_soc_rt5670 snd_soc_acpi snd_soc_rt5651 snd_soc_rt5645 snd_seq_midi snd_seq_midi_event hid_multitouch snd_soc_rt5640 snd_soc_rl6231 snd_soc_core snd_rawmidi snd_pcm snd_seq mei_txe mei wmi snd_seq_device snd_timer int3400_thermal processor_thermal_device int3403_thermal intel_soc_dts_ios [... cut ...]
[60175.359041] RDX: 287c7287b7287a72 RSI: ffffffffffffffff RDI: ffff99d9732d62ed
[60175.359050] RBP: 287c7287b7287a72 R08: 0000000000000d15 R09: 0000000000000000
[60175.359059] R10: ffff99d9732d7000 R11: ffff99d9732d62eb R12: ffff0a00ffffff04
[60175.359068] R13: ffffffff985d397b R14: 0000000000000d15 R15: ffffffff985d397b
[60175.359080] FS:  00007f742ece5740(0000) GS:ffff99d973e00000(0000) knlGS:0000000000000000
[60175.359090] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[60175.359099] CR2: 00007f63a4020c50 CR3: 0000000071ee6000 CR4: 00000000001006f0

```

singalsu, 2020-03-18:
@lgirdwood No more kernel crash with today's topic/sof-dev kernel.

singalsu, 2020-03-18:
I'll do a PR, speaker-test with 48k works, need to adjust still for 44.1k to work too...

singalsu, 2020-03-18:
The PR is #2574 . I just noticed that speaker-test can't be used with "-t wav" with different "-r" than the supplied "speaker left, speaker right" wav clip 48 kHz. Option "-f 997 -t sine" works for any "-r".

```
singalsu@taskurotta:~$ speaker-test -Dhw:0,1 -r 48000 -f 997 -c 2 -t sine -l 5

speaker-test 1.1.0

Playback device is hw:0,1
Stream parameters are 48000Hz, S16_LE, 2 channels
Sine wave rate is 997,0000Hz
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5,656434
 0 - Front Left
 1 - Front Right
Time per period = 5,947406
 0 - Front Left
 1 - Front Right
Time per period = 5,875320
 0 - Front Left
 1 - Front Right
Time per period = 5,947418
 0 - Front Left
 1 - Front Right
Time per period = 5,946918
singalsu@taskurotta:~$ speaker-test -Dhw:0,1 -r 44100 -f 997 -c 2 -t sine -l 5

speaker-test 1.1.0

Playback device is hw:0,1
Stream parameters are 44100Hz, S16_LE, 2 channels
Sine wave rate is 997,0000Hz
Rate set to 44100Hz (requested 44100Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5,501895
 0 - Front Left
 1 - Front Right
Time per period = 5,891664
 0 - Front Left
 1 - Front Right
Time per period = 5,814888
 0 - Front Left
 1 - Front Right
Time per period = 5,891432
 0 - Front Left
 1 - Front Right
Time per period = 5,891287

```

dengyangchao, 2020-03-19:
After applied #2574 , below commands can work well, issue can't be reproduced.
```
speaker-test -Dhw:0,1 -r 48000 -f 997 -c 2 -t sine -l 5
speaker-test -Dhw:0,1 -r48000 -f1000 -c2 -t wav -l 10
speaker-test -Dhw:0,1 -r 44100 -f 997 -c 2 -t sine -l 5
```
If use command "speaker-test -Dhw:0,1 -r44100 -f1000 -c2 -t wav -l 10"  it shows sample rate doesn't match 
```
ubuntu@sh-byt-mb-rt5682-1111:~$ speaker-test -Dhw:0,1 -r44100 -f1000 -c2 -t wav -l 10

speaker-test 1.1.3

Playback device is hw:0,1
Stream parameters are 44100Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 44100Hz (requested 44100Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
Sample rate doesn't match (48000) for /usr/share/sounds/alsa/Front_Left.wav
```

singalsu, 2020-03-19:
Yep, that looks like a bug in speaker-test. It should resample the wav clip or issue an error if "-r" is given with -"t wav".

It's a good idea to replace 1 kHz sine wave tests with 997 Hz. 997 is prime number so it won't repeat the same PCM code values as 1 kHz would and possibly hide buffering errors.

singalsu, 2020-03-20:
PR #2574 is merged so we can close this issue.

closed 2020-03-20 by singalsu