thesofproject/sof #2384: [FEATURE] Add support for sparse in SOF

dbaluta, 2020-02-13:
**Is your feature request related to a problem? Please describe.**
Better support for static analysis.

https://en.wikipedia.org/wiki/Sparse
https://elinux.org/Sparse

**Describe the solution you'd like**

Similar with Linux kernel, we need an option to build that runs sparse. In Linux kernel to run sparse
you just need to do 

make C=1   [targets] Check all c source with checker tool

**Describe alternatives you've considered**
We already have checkpatch.pl integrated, would be a nice addition to have sparse

**Additional context**
Add any other context or screenshots about the feature request here.

dbaluta, 2020-02-13:
Cc: @IulianOlaru249 @jajanusz @paulstelian97

paulstelian97, 2020-02-13:
I have looked a bit into this and a simple recommendation is to build sparse for the system you're building the firmware on, then use the `CC` and `REAL_CC` variables to do the actual build: `REAL_CC` would contain the path to the xtensa-gcc or xcc compiler (what `CC` was before) and `CC` would be updated to point to the `cgcc` wrapper script within Sparse itself. You still need to build Sparse for the platform you're building on (most likely Intel) but then the integration with the building process shouldn't be very difficult. (check the elinux page, lower side)

plbossart, 2020-02-13:
+1 
Sparse and cppcheck are my favorite tools these days.
Coverity is more complicated to enable.

lgirdwood, 2020-08-18:
@dbaluta @paulstelian97 any update here. Should we move to v1.7 ?

dbaluta, 2020-08-18:
@lgirdwood lets move to v1.7, no progress here. @IulianOlaru249 could you investigate this?

lgirdwood, 2021-01-12:
@dbaluta any update here - move to later ?

dbaluta, 2021-01-13:
@lgirdwood i don't remember what happened to this. Lets move it for later.

marc-hb, 2022-11-01:
This is done, see follow up issue #6317 for more details.

closed 2022-11-01 by marc-hb