thesofproject/sof #2385: [BUG] topology: rt711-rt1308-rt715: don't change DMIC device number depending on number of speakers

plbossart, 2020-02-14:
Depending on the number of playback channels, the 'DMIC" device attached to rt715 will be either hw:0,3 or hw:0,4. I suggest we keep it constant, even if it means a hole in the device list.

There is no requirement that the list of devices be incremental, it's ok to have holes.

While I am at it, we should not use "DMIC" but 'Internal mics". The RT715 can deal with analog microphones IIRC.

@RanderWang can you look into this?

plbossart, 2020-02-14:
update: the HDMI device numbers are also changed and this causes the UCM file to no longer work.

We need to keep a match between topology and UCM!

RanderWang, 2020-02-14:
yes, I agree with you. Please check my PR. thanks!

closed 2020-02-14 by lgirdwood