thesofproject/sof #2386: [BUG][CML Mantis]I/O error when running multipipeline using more than 6 pipelines

Liviali155, 2020-02-14:
**Describe the bug**
I/O error when running multipipeline using more than 6 pipelines

**To Reproduce**
1.sudo reboot
2. ./multiplex_pipeline-test.sh pcm0p pcm3p pcm4p pcm0c pcm6c pcm7c

**Reproduction Rate**
10/10

**Expected behavior**
There is no I/O error,all the pipelines can work normally

**Actual result**
I/O error
[image]

<details>
 <summary>dmesgerror.log</summary>

````
[  171.494974] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  171.497459] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  171.990195] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[  171.990211] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  171.997862] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 0
[  172.006788] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  172.493472] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  172.493476] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  172.493482] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  172.493559] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  172.493566] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000038f rirb 00
[  172.493567] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  172.493572] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  172.493573] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  172.493589] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20340000 successful
[  172.493593]  DMIC16kHz: ASoC: trigger FE cmd: 0 failed: -110
[  172.493594] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  172.493610] sof-audio-pci 0000:00:1f.3: pcm: free stream 7 dir 1
[  172.493622] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  172.493639] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  172.997790] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  172.997799] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  172.997806] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  172.997901] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
````
</details>

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         DMIC 2.1     379668653.593750 379668640.000000 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379668666.614583        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379668699.739583        33.125000 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379668712.760417        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379668745.000000        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379668758.020833        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379668791.250000        33.229168 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379668804.270833        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379668836.510417        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379668849.531250        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379668882.708333        33.177082 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379668895.729167        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379668927.968750        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379668940.989583        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379668974.166667        33.177082 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379668987.187500        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379669019.427083        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
[... 99 lines cut ...]
    0      1         DMIC 2.1     379671305.885417        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379671318.906250        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379671352.135417        33.229168 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379671365.156250        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379671397.395833        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379671410.416667        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379671443.593750        33.177082 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379671456.614583        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379671488.854167        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379671501.822917        12.968750 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379671535.000000        33.177082 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.0     379671548.020833        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.1     379671580.260417        32.239582 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC 2.1     379671593.281250        13.020833 rs/intel/cavs/dmic.c:1500	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC 2.0     379671626.458333        33.177082 rs/intel/cavs/dmic.c:1498	dmic_irq_handler() error: full fifo A or PDM overrrun
````
</details>

Platform: CML Mantis with onboard codec  ALC3204 in HDA mode
Firmware: 4d17d0de https://github.com/thesofproject/sof/commits/master
Kernel: daf71dce https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware

[dmesg0214.log](https://github.com/thesofproject/sof/files/4203323/dmesg0214.log)
[logger.log](https://github.com/thesofproject/sof/files/4203324/logger.log)

paulstelian97, 2020-02-14:
Can you disable the system agent? (HAVE_AGENT=n). If that's guilty then either it will work or it will fail above 6 pipelines but recover once you go below that I think.

Liviali155, 2020-02-14:
Bi-sect result:
caused by Firmware commit 91fb594b[](https://github.com/thesofproject/sof/commit/91fb594bd9455fcf8415a588cf45e6294ac46e0e)

Liviali155, 2020-02-14:
Related #2280

paulstelian97, 2020-02-14:
I suppose there needs to be a way for firmware to recover and continue the streams it could carry if it can't dynamically switch clocks.

We should continue the discussion on #2280, it's a manifestation of the same actual issue (not being able to manage being overloaded in terms of DSP usage)

lgirdwood, 2020-02-14:
@tlauda @lbetlej @plbossart @paulstelian97 @dbaluta this is down as a opne for EDF scheduling, lets say closure for v1.6. i.e. we reject new work that wont fit in EDF. TSC discussion subject too.

slawblauciak, 2020-02-18:
Can you please verify whether it's reproducible with high freq clock enabled in the menuconfig?
Just to confirm the rootcause.

Liviali155, 2020-03-04:
Tested on jf-cfl-rvp-hda-1 also can reproduce this issue,step as below:
1.cd sof-test(sof-test as attachment)
2.cd test-case
3. export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
4. ./multiple-pipeline-capture.sh  -l 50

I/O error and dmesg show "error: ipc timed out for 0x60050000 size 12"

Platform: CFL-S   RVP with onboard codec  ALC700 in HDA mode

<details>
 <summary>dmesg_error.log</summary>

````
[  794.651393] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  794.658420] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  794.658424] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[  795.155116] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  795.155125] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  795.155135] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  795.155318] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  795.155346] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000008f rirb 00
[  795.155350] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  795.155369] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  795.155373] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  795.155428] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  795.155452] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  795.155953] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  795.155971]  DMIC: ASoC: trigger FE cmd: 0 failed: -110
[  795.156001] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 1
[  795.658108] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  795.658111] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  795.658115] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  795.658246] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  795.658257] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000008d rirb 00
[  795.658258] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  795.658265] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  795.658266] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  795.658334] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  795.659091] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20340000 successful
[  795.659099]  HDA Analog: ASoC: trigger FE cmd: 0 failed: -110
[  795.875420] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[  796.163109] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  796.163118] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  796.163130] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  796.163305] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
````
</details>

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 2.1     1577987604.479167              nan rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     1577987637.604167        33.125000 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     1577987650.572917        12.968750 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     1577987682.656250        32.083332 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     1577987695.833333        13.177083 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     1577987728.958333        33.125000 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     1577987742.031250        13.072917 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     1577987774.062500        32.031250 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     1577987787.343750        13.281250 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     1577987820.468750        33.125000 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     1577987833.437500        12.968750 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     1577987865.416667        31.979166 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     1577987878.541667        13.125000 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     1577987911.666667        33.125000 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
[... cut ...]
    0      1          DAI 2.0     1577988824.635417        33.229168 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     1577988837.604167        12.968750 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
````
</details>

[dmesg0304.log](https://github.com/thesofproject/sof/files/4285282/dmesg0304.log)
[sof_errortrace0304.log](https://github.com/thesofproject/sof/files/4285285/sof_errortrace0304.log)
[sof_trace0304.log](https://github.com/thesofproject/sof/files/4285286/sof_trace0304.log)

Liviali155, 2020-03-06:
On TGL U HDA also can reproduce this issue
1.sudo reboot with HDMI&DP connected
2. ./multiplex_pipeline-test.sh pcm0p pcm3p pcm4p pcm0c pcm6c pcm7c

I/O error,dmesg show "ipc timed out for 0x60050000 size 12"
<details>
 <summary>dmesg_error.log</summary>

````
[  164.688692] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 0
[  164.688694] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[  164.688703] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  164.696586] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  165.192561] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  165.192566] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  165.192571] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  165.192679] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  165.192687] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000038f rirb 00
[  165.192689] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  165.192699] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  165.192701] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  165.192725] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  165.192733] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  165.192740]  DMIC: ASoC: trigger FE cmd: 0 failed: -110
[  165.192764] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 1
[  165.192769] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  165.192770] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  165.192774] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  165.696565] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  165.696571] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  165.696576] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  165.696681] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  165.696689] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000038d rirb 00
[  165.696691] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  165.696696] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  165.696698] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  165.696767] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  165.696813] sof-audio-pci 0000:00:1f.3: pcm: close stream 6 dir 1
[  165.856731] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 4 dir 0 cmd 0
````
</details>

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 2.1     509070539.843750 509070528.000000 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     509070565.416667        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     509070629.843750        64.427086 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     509070655.416667        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     509070717.760417        62.343750 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     509070743.333333        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     509070807.760417        64.427086 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     509070833.333333        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     509070895.677083        62.343750 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     509070921.197917        25.520834 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     509070985.625000        64.427086 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     509071011.197917        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     509071073.437500        62.239582 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     509071098.958333        25.520834 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.0     509071163.385417        64.427086 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.0     509071188.958333        25.572916 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1          DAI 2.1     509071251.302083        62.343750 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
[... cut ...]
    0      1          DAI 2.1     509072851.822917        62.343750 rs/intel/cavs/dmic.c:1518	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1          DAI 2.1     509072877.343750        25.520834 rs/intel/cavs/dmic.c:1520	dmic_irq_handler() error: full fifo B or PDM overrrun
````
</details>

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 9ebb2881 https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

lgirdwood, 2020-03-06:
@lbetlej it seem the Low Power clock canary is now singing :-)

Liviali155, 2020-03-09:
On TGL U HDA also can reproduce this issue when paplay then suspend/resume
1.paplay  -vv wav/48California_Dreaming.wav
2.sudo rtcwake -m mem -s 10

 Internel error ,dmesg show "error: ipc timed out for 0x60050000 size 12"

[image]

<details>
 <summary>dmesg_error.log</summary>

````
[   93.893280] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   93.893581] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   93.893635] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   93.893637] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   94.396880] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   94.396886] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   94.396891] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   94.396989] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   94.396997] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[   94.396999] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   94.397009] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   94.397011] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   94.397529] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[   94.397539]  HDA Analog: ASoC: trigger FE cmd: 0 failed: -110
[   94.397666] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[   94.398194] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x14001e successful
[   94.398201] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   94.900707] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[   94.900713] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context    
````
</details>

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 6cd8acc1 https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

[dmesg0309.log](https://github.com/thesofproject/sof/files/4305834/dmesg0309.log)
[sof_trace0309.log](https://github.com/thesofproject/sof/files/4305835/sof_trace0309.log)

Liviali155, 2020-03-10:
Test with latest sof/dev+master on TGL U HDA and CML laptop with onboard codec ALC3204 in HDA mode ,error not seen when paplay then suspend/resume,error only seen when do multipipeline

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 0996f794 https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

Platform: CML laptop with onboard codec ALC3204 in HDA mode
Firmware: 0996f794https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-2ch.tplg same as firmware

Liviali155, 2020-03-10:
Tried with PR #2512
There is I/O error when multipipeline using more than 6 pipelines,but there is no error message in dmesg,and  all the pipeline can work well after error occured

Platform: CML laptop with onboard codec ALC3204 in HDA mode
Platform: TGL-U RVP with onboard codec ALC711 in HDA mode

lgirdwood, 2020-03-11:
@Liviali155 can we close ?

Liviali155, 2020-03-11:
> @Liviali155 can we close ?

Still has I/O error

Liviali155, 2020-03-12:
Tested on latest sof/dev(e9fe7928)+master(0d0e5b73)
on TGL-U RVP with onboard codec ALC711 in HDA mode:
I/O error when multipipeline using more than 6 pipelines,dmesg show "error: ipc timed out for 0x60050000 size 12",after error occured,all the pipeline can not work

on CML laptop with onboard codec ALC3204 in HDA mode
I/O error when multipipeline using more than 6 pipelines,but no error message in dmesg,all the pipeline can work after error occured

mengdonglin, 2020-03-24:
Caused by DSP overload, commented by @tlauda  https://github.com/thesofproject/sof/issues/2531#issuecomment-601175843

Liviali155, 2020-03-25:
After  disabled low power clock: Platform -> [ ] Enable low power ring oscillator as the default clock,error not seen on TGL-U RVP with onboard codec ALC711 in HDA mode

sinahuang, 2020-03-30:
Issue still can be reproduced on CML-H with sof-dev  (commit:a272777 + PR1943 + PR1954+PR1897 )and cml-011-drop-stable  (commit:c48aeff).

[dmesg30.log](https://github.com/thesofproject/sof/files/4402579/dmesg30.log)
[sof_1_trace30.log](https://github.com/thesofproject/sof/files/4402581/sof_1_trace30.log)
[sof_error_trace30.log](https://github.com/thesofproject/sof/files/4402599/sof_error_trace30.log)

**Test recipe**
Kernel:Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit:a272777 + PR1943 + PR1954+PR1897
FW: https://github.com/thesofproject/sof/commits/cml-011-drop-stable  commit:c48aeff 
Tplg: Same with FW branch,  sof-cml-rt711-rt1308-rt715.tplg
Platform:CML-H Laptop with codec ALC711, 2 x RT1308, RT715 in SDW mode.

slawblauciak, 2020-04-07:
@sinahuang should no longer occur on master, high perf clock has been set as default

sinahuang, 2020-04-07:
Issue can't be reproduced with sof-dev  commit:052e45d and master commit:  b4e8b1e4 on on CML-H and CML-U, so closed it

closed 2020-04-07 by sinahuang