thesofproject/sof #2390: [BUG][glk][sof_v1.3]"Input/output error" when  aplay via (0,5)

Liviali155, 2020-02-17:
**Describe the bug**
"Input/output error" when  aplay via (0,5)
After issue reproduced,All the pipeline can work normally

**To Reproduce**
1.cd sof-test(sof-test as attachment)
2.cd test-cases
3.export TPLG=/lib/firmware/intel/sof-tplg/sof-glk-da7219.tplg
4. ./check-alsabat.sh

**Reproduction Rate**
10/10

**Expected behavior**
There is no I/O error when aplay via (0,5)

**Actual result**
There is I/O error when aplay via(0,5)
```
2020-02-16 21:55:37 UTC [REMOTE_COMMAND] aplay -D hw:0,5 -r 48000 -c 2 -f S16_LE /dev/zero -q
aplay: pcm_write:2011: write error: Input/output error
/home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh: line 77: kill: (7572) - No such process
2020-02-16 21:55:39 UTC [REMOTE_ERR] aplay process for pcm HDMI1 is not alive
2020-02-16 21:55:41 UTC [REMOTE_INFO] Test Result: FAIL!
```

[image]

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         749424956.093750 749424960.000000 intel/cavs/hda-dma.c:266 	hda-dmac: 5 wait for buffer full timeout rp 0x0 wp 0x0
    0      1         HOST         749424963.854167         7.760417     src/audio/host.c:770 	host_copy() error: dma_copy() failed, ret = 4294967234
    0      1         PIPE         749424969.895833         6.041667 src/audio/pipeline.c:784 	pipeline_copy() error: ret = -62, start->comp.id = 31, dir = 1
    0      1         PIPE 5.32    749424976.614583         6.718750 src/audio/pipeline.c:896 	pipeline_xrun_recover()
    0      1         COMP         749424988.593750        11.979167 rc/audio/component.c:173 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_PREPARE
    0      1         PIPE         749424994.427083         5.833333 src/audio/pipeline.c:387 	pipeline_prepare() error: ret = -22,dev->comp.id = 27
    0      1         PIPE 5.32    749425000.885417         6.458333 src/audio/pipeline.c:903 	pipeline_xrun_recover() error: pipeline_prepare() failed, ret = -22
    0      1         PIPE 5.32    749425006.354167         5.468750 src/audio/pipeline.c:974 	pipeline_task(): xrun recover failed! pipeline will be stopped!
````
</details>

There is no error dmesg

Platform: glk with onboard codec DA7219 in I2S mode
Firmware: 5dd9aac https://github.com/thesofproject/sof/commits/1.3
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware
[dmesg0217.log](https://github.com/thesofproject/sof/files/4211806/dmesg0217.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/4211808/sof_trace.log)
[sof-test.zip](https://github.com/thesofproject/sof/files/4211809/sof-test.zip)

lgirdwood, 2020-02-17:
@Liviali155 why do we care about v1.3 ? Does this happen on master or v1.4.2 ?

Liviali155, 2020-02-17:
> @Liviali155 why do we care about v1.3 ? Does this happen on master or v1.4.2 ?

Error only seen on v1.3,not seen on master and v1.4.2

lgirdwood, 2020-02-17:
@Liviali155 any users of this PCM/feature on v1.3 ?

Liviali155, 2020-02-18:
Error not seen on v1.4.2, closed it

closed 2020-02-18 by Liviali155