thesofproject/sof #2391: [BUG][APL][PCM512X][tag-v1.4.2] Tplg load fail "tplg component load failed -22"

dengyangchao, 2020-02-17:
**Describe the bug**
Tag v1.3 and tag v1.4.2 tplg load fail "tplg component load failed -22"
And sof-dev +master can't reproduce the issue.

**To Reproduce**
1.Reboot system via "sudo reboot".
2.Check via "dmesg".

**Reproduction Rate**
100%

**Expected behavior**
Tplg load normal, and no error
**Impact**
Tplg load fail

**Environment**
platform: APL UP2 with PCM512X in I2S mode
Firmware: 66a5ae9 https://github.com/thesofproject/sof/releases
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: sof-apl-pcm512x.tplg 

**Dmesg**
```
[    4.523899] sof-audio-pci 0000:00:0e.0: ASoC: physical link iDisp1 (id 1) not exist
[    4.523987] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[    4.523991] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[    4.523995] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    4.524000] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524004] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524006] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    4.524011] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524015] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524018] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    4.524024] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524027] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524029] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    4.524035] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524038] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    4.524041] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    4.524045] sof-audio-pci 0000:00:0e.0: error: tplg component load failed -22
[    4.524056] sof-audio-pci 0000:00:0e.0: error: failed to load DSP topology -22
[    4.524058] sof-audio-pci 0000:00:0e.0: ASoC: failed to probe component -22
[    4.524102] sof_pcm512x sof_pcm512x: ASoC: failed to instantiate card -22
[    4.524175] sof_pcm512x: probe of sof_pcm512x failed with error -22
[    4.525866] asix 1-4.1:1.0 eth0: register 'asix' at usb-0000:00:15.0-4.1, ASIX AX88772B USB 2.0 Ethernet, 00:0e:c6:ca:ff:9d
[    4.525911] usbcore: registered new interface driver asix
[    4.529964] asix 1-4.1:1.0 enx000ec6caff9d: renamed from eth0
[    5.151684] Generic FE-GE Realtek PHY r8169-200:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-200:00, irq=IGNORE)
[    5.254256] r8169 0000:02:00.0 enp2s0: Link is Down
[    5.260406] Generic FE-GE Realtek PHY r8169-300:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-300:00, irq=IGNORE)
[    5.364034] r8169 0000:03:00.0 enp3s0: Link is Down
[    9.272623] r8169 0000:03:00.0 enp3s0: Link is Up - 1Gbps/Full - flow control off
[    9.272639] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[   10.389247] rfkill: input handler disabled
[   13.381128] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

[dmesg_tag-v1.4.2.log](https://github.com/thesofproject/sof/files/4212166/dmesg_tag-v1.4.2.log)

dengyangchao, 2020-02-17:
Sof-dev + tag v1.3 also can reproduce the issue

platform: APL UP2 with PCM512X in I2S mode
Firmware: 5dd9aac https://github.com/thesofproject/sof/releases
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: sof-apl-pcm512x.tplg

wwittbrx, 2020-02-17:
Did you remember to set up ACPI scripts on this platform? Don't know if this is still required but might be worth a check
https://github.com/thesofproject/acpi-scripts

dengyangchao, 2020-02-17:
@wwittbrx Yes, already added the acpi scripts.

dengyangchao, 2020-02-17:
Looking at the CI history, I found that the issue started to appear after the January 24th kernel update
http://sof-tcr.sh.intel.com/#/campaigns

dengyangchao, 2020-02-18:
Below are some info i tried
1. Sof-dev Feb 17th(b2d98237) +master Feb 17th(b206b7a8) ,  issue can't be reproduced.
2. Sof-dev Jan 23rd(37a07941)   + tag_v1.4.2   ,    issue can't be reproduced.
3. Sof-dev Jan 24th(9b1e6574)   + tag_v1.4.2    ,   issue can reproduce

slawblauciak, 2020-02-18:
That FW version is likely not compatible with the kernel/driver, to name one difference, there were many changes in the topologies and their ABI.

slawblauciak, 2020-02-18:
1.3 is likely incompatible with the top of the kernel, there were quite a few changes in the topologies I believe. As for 1.4.2, it will have to be verified.

mengdonglin, 2020-02-24:
> 2. Sof-dev Jan 23rd(37a07941)   + tag_v1.4.2   ,    issue can't be reproduced.
> 3. Sof-dev Jan 24th(9b1e6574)   + tag_v1.4.2    ,   issue can reproduce
@dengyangchao can you bi-sect the difference between the 2 kernel?

dengyangchao, 2020-02-24:
> > 1. Sof-dev Jan 23rd(37a07941)   + tag_v1.4.2   ,    issue can't be reproduced.
> > 2. Sof-dev Jan 24th(9b1e6574)   + tag_v1.4.2    ,   issue can reproduce
> >    @dengyangchao can you bi-sect the difference between the 2 kernel?

@mengdonglin I tried, but the code seems re-base, can't find the correct order of commit

lyakh, 2020-03-11:
This is a known incompatibility. We've replaced the kernel support for Up^2 / apl-pcm512, it's now incompatible with the original topology. We've updated the topology too. So "both new" and "both old" work, but a mix of old and new doesn't.

plbossart, 2020-03-11:
Ack, the old topology only worked with the machine driver code that was not upstreamed. Now that we did all the cleanups and upstreamed the machine driver, the new topology is required.

wenqingfu, 2020-03-17:
> This is a known incompatibility. We've replaced the kernel support for Up^2 / apl-pcm512, it's now incompatible with the original topology. We've updated the topology too. So "both new" and "both old" work, but a mix of old and new doesn't.

close as won't fix

closed 2020-03-17 by wenqingfu