thesofproject/sof #2393: [BUG][BYT][Nocodec] Pause-release aplay cause "ipc error for 0x60070000 size 12"

dengyangchao, 2020-02-17:
**Describe the bug**
Pause-release aplay cause "ipc error for 0x60070000 size 12"

**To Reproduce**
1.Reboot system via "sudo reboot"
2.Aplay via command "aplay -Dhw:0,1 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i"
3.Pause-release aplay.
4.Repeat step 3.

**Reproduction Rate**
3/3
1st reproduced at 8th
2nd reproduced at 14th
3rd reproduced at 10th

**Expected behavior**
Pause-release success and no error

**Impact**
Occur ipc error

**Environment**
Platform: BYT with  nocodec
Firmware: b206b7a8 https://github.com/thesofproject/sof/commits/master
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-byt-nocodec.tplg same as firmware

**Dmesg**
```
[   79.428500] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   79.428518] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   79.494922] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 4
[   79.494932] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   79.495076] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   79.495127] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 13317 DAI position 164966
[   79.928622] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   79.928639] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   80.097853] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 13317 DAI position 164966
[   80.097874] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 3
[   80.097891] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   80.099566] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   80.185445] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 4
[   80.185455] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   80.185625] sof-audio-acpi 80860F28:01: error: ipc error for 0x60070000 size 12
[   80.185639]  Media Playback 1: ASoC: trigger FE cmd: 4 failed: -16
[   80.185649] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 3
[   80.185654] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   80.185765] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   85.994685] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 4
[   85.994696] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[   85.997182] sof-audio-acpi 80860F28:01: error: ipc error for 0x60070000 size 12
[   85.997199]  Media Playback 1: ASoC: trigger FE cmd: 4 failed: -16
[   85.997213] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 3
[   85.997220] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   85.997358] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[   85.997374] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[   85.997380] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   85.997545] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   85.997577] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0

```

**sof-error-trace**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         WAIT          80079505.260417  80079504.000000       src/lib/wait.c:39  	ewt
    0      1          DMA          80079513.281250         8.020833 src/drivers/dw/dma.c:437 	dw_dma_stop() error: dma 1 channel 1 timeout
    0      1           SA          80079528.541667        15.260417      src/lib/agent.c:56  	validate(), ll drift detected, delta = 36080
    0      1          DMA          80165735.989583     86207.445312 src/drivers/dw/dma.c:275 	dw_dma_start() error: dma 1 channel 1 not ready ena 0x2 status 0x3
    0      1         PIPE          80165750.104167        14.114583 src/audio/pipeline.c:631 	pipeline_trigger() error: ret = -16, host->comp.id = 19, cmd = 3
    0      1          IPC          80165761.770833        11.666667    src/ipc/handler.c:511 	ipc: comp 19 trigger 0x70000 failed -16
    0      1         WAIT          80410002.291667    244240.515625       src/lib/wait.c:39  	ewt
    0      1       BUFFER          80410010.312500         8.020833 rc/trace/dma-trace.c:73  	trace_work() error: dma_copy_to_host_nowait() failed
    0      1          DMA          85973520.781250   5563510.500000 src/drivers/dw/dma.c:275 	dw_dma_start() error: dma 1 channel 1 not ready ena 0x3 status 0x3
    0      1         PIPE          85973534.895833        14.114583 src/audio/pipeline.c:631 	pipeline_trigger() error: ret = -16, host->comp.id = 19, cmd = 3
    0      1          IPC          85973546.562500        11.666667    src/ipc/handler.c:511 	ipc: comp 19 trigger 0x70000 failed -16
    0      1         COMP 3.20     85976171.927083      2625.364502 rc/audio/component.c:167 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
    0      1         COMP 3.21     85976184.791667        12.864583 rc/audio/component.c:167 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
    0      1         COMP 1.4      85976197.656250        12.864583 rc/audio/component.c:167 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
    0      1         COMP 1.3      85976210.416667        12.760417 rc/audio/component.c:167 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
    0      1         COMP 1.9      85976234.375000        23.958334 rc/audio/component.c:167 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
```

[dmesg.log](https://github.com/thesofproject/sof/files/4212250/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/4212253/logger.log)
[sof-error-trace.log](https://github.com/thesofproject/sof/files/4212254/sof-error-trace.log)

dengyangchao, 2020-02-18:
HW:0,0 also can reproduce the issue too, and even 20191203 sof-dev+master, issue also can be reproduced.
```
**** List of PLAYBACK Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: Low Latency (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: Media Playback 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

lgirdwood, 2020-02-18:
@dengyangchao related to #2400 ? Can you bisect.

dengyangchao, 2020-02-19:
> @dengyangchao related to #2400 ? Can you bisect.

@lgirdwood It's hard to bisect, i tried and even at date 20191001 master also can reproduce the issue

lgirdwood, 2020-02-19:
@dengyangchao ok, btw @ktrzcinx has some DMA and SSP fixes in the pipeline that may fix this. These fixes currently blocked on CI issue on BDW that @fredoh9 and @xiulipan are fixing.

dengyangchao, 2020-03-17:
Issue can't be reproduced, close.

closed 2020-03-17 by dengyangchao