thesofproject/sof #2400: [BUG][CML]unable to install hw params via speaker pipeline

aiChaoSONG, 2020-02-18:
**Describe the bug**
run aplay on speaker pipeline (hw:0,5), you will see "unable to install hw params" error.

**To Reproduce**
run `aplay -Dhw:0,5 -f dat -r 48000 -i -vvv -c 2 ./somewave.wav`

**Reproduction Rate**
100% on helios

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {b2d98237}
    * SOF: {53d35e49}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML with rt1011 and rt5682 codecs}
```
test@helios:~/chao/testsuite$ aplay -Dhw:0,5 -f dat -r 48000 -i -vvv -c2 ./llwav.wav
Playing WAVE './llwav.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Warning: rate is not accurate (requested = 44100Hz, got = 48000Hz)
         please, try the plug plugin 
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
```
```
[ 1022.892390] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1022.892628] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[ 1022.892638] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[ 1022.892643] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[ 1022.892652]  Speakers: ASoC: hw_params FE failed -22
[ 1022.892669] sof-audio-pci 0000:00:1f.3: pcm: free stream 5 dir 0
[ 1022.892950] sof-audio-pci 0000:00:1f.3: pcm: close stream 5 dir 0
```
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP 7.38    1278773821.145833              nan      src/audio/dai.c:243 	dai_verify_params() error: pcm channels parameter does not match hardware channels
    0      1         COMP 7.38    1278773826.979167         5.833333      src/audio/dai.c:380 	dai_params(): pcm params verification failed.
    0      1         PIPE         1278773833.385417         6.406250 src/audio/pipeline.c:391 	pipeline_params() error: ret = -22, host->comp.id = 34
    0      1          IPC         1278773840.625000         7.239583    src/ipc/handler.c:302 	ipc: pipe 7 comp 34 params failed -22

```

aiChaoSONG, 2020-02-18:
bisect, and de73e60a8 cause this issue.

lgirdwood, 2020-02-18:
@bkokoszx looks like maybe a duplicate of #2379 too.

bkokoszx, 2020-02-18:
@lgirdwood 
I've just added few debug logs to FW.
In above case, we try to make ```aplay``` with ```-c2``` (channels parameter set to 2). But, when we configure dai and send ```dai_set_config()``` to FW, we set: ```ssp->params.tdm_slots``` to 4.
```
 0      2          DAI 1.1      28490742.812500        12.708333 ers/intel/cavs/ssp.c:610    ssp_set_config(), config->ssp.tdm_slots = 4
```
There is a mismatch between our hardware parameters and pcm parameters. As I understand, now FW works properly and returns an error when there is a mismatch between pcm and hardware parameters (maybe I am not aware of something?). In my opinion we should just change channels in topology:

``` diff
diff --git a/tools/topology/sof-cml-rt1011-rt5682.m4 b/tools/topology/sof-cml-rt1011-rt5682.m4
index 0201829e..100c42f4 100644
--- a/tools/topology/sof-cml-rt1011-rt5682.m4
+++ b/tools/topology/sof-cml-rt1011-rt5682.m4
@@ -58,13 +58,13 @@ ifelse(SOF_ABI_VERSION_3_9_OR_GRT, `1',
        SSP_CONFIG(DSP_A, SSP_CLOCK(mclk, 24000000, codec_mclk_in),
                SSP_CLOCK(bclk, 4800000, codec_slave),
                SSP_CLOCK(fsync, 48000, codec_slave),
-               SSP_TDM(4, 25, 3, 15),
+               SSP_TDM(2, 25, 3, 15),
                SSP_CONFIG_DATA(SSP, 1, 24, 0, 0, 10)))',
 `DAI_CONFIG(SSP, 1, 6, SSP1-Codec,
         SSP_CONFIG(DSP_A, SSP_CLOCK(mclk, 24000000, codec_mclk_in),
                 SSP_CLOCK(bclk, 4800000, codec_slave),
                 SSP_CLOCK(fsync, 48000, codec_slave),
-                SSP_TDM(4, 25, 3, 15),
+                SSP_TDM(2, 25, 3, 15),
                 SSP_CONFIG_DATA(SSP, 1, 24)))')

 DEBUG_END
```
But I do not understand how it worked before?

wwittbrx, 2020-02-18:
@aiChaoSONG Is any special setup required for this? I tried installing the newest kernel (b2d9823) but Ubuntu fails at boot and the platform turns off. Older kernel (v5.5) boots just fine.

aiChaoSONG, 2020-02-18:
@wwittbrx No special setup is needed. if the kernel you build fails, you can use ci daily kernel: https://sof-ci-storage.sh.intel.com/daily/. or try to rebuild with the kconfig we use: https://sof-ci-storage.sh.intel.com/daily/20200218/linux-b2d98237-2/config-daily-default-20200218-0.txt

wwittbrx, 2020-02-18:
I'm getting the same result with daily kernel, looks like a platform issue, I will reinstall Ubuntu to start fresh and retry.

bkokoszx, 2020-02-19:
@lgirdwood 
Could you please refer to comment: https://github.com/thesofproject/sof/issues/2400#issuecomment-587457169 ?

lgirdwood, 2020-02-19:
@bkokoszx This can be fixed with the multi-dai config (it basically extends your recent pipeline params work) so that the DAI can also be updated to use a new config if the pipeline cannot be configured to do the conversion.
In the short term we could add a converter in the pipeline that can convert 2c to 4chan until multi DAi is ready.

Liviali155, 2020-02-20:
On TGL HDA and CML Mantis also can reproduce this issue when arecord (0,0) using "-fs32_le" or "-fs24_le"
***To Reproduce***
1.sudo reboot
2."arecord -Dhw:0,0 -r48000 -c2 -fs24_le test.wav -vv -i"  or "arecord -Dhw:0,0 -r48000 -c2 -fs32_le test.wav -vv -i 

[image]

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP 2.11    135730527.239583 135730528.000000 udio/eq_iir/eq_iir.c:859 	eq_iir_prepare(), No proc func
    0      1         PIPE         135730538.593750        11.354167 src/audio/pipeline.c:499 	pipeline_prepare() error: ret = -22, dev->comp.id = 9
    0      1          IPC         135730550.520833        11.927083    src/ipc/handler.c:311 	ipc: pipe 2 comp 9 prepare failed -22
````
</details>

<details>
 <summary>dmesg_error.log</summary>

````
[ 1109.059544] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[ 1109.059551] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[ 1109.059552] sof-audio-pci 0000:00:1f.3: periods:4
[ 1109.059587] sof-audio-pci 0000:00:1f.3: stream_tag 2
[ 1109.059602] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1109.060141] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[ 1109.060155] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[ 1109.060162] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[ 1109.060173]  HDA Analog: ASoC: hw_params FE failed -22
[ 1109.060222] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1109.060360] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1109.060369] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[ 1109.060392] sof-audio-pci 0000:00:1f.3: hda_link_hw_free: link_dev is not assigned
[ 1109.060870] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 1
[ 1109.321297] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1109.321336] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1109.821321] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1109.821361] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1111.395431] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[ 1111.439864] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1111.440081] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1111.440191] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 1111.440195] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[ 1111.440197] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1111.440234] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[ 1111.440240] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
[ 1111.440243] sof-audio-pci 0000:00:1f.3: New DSP state 4 substate 0
````
</details>

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 247af133 https://github.com/thesofproject/sof/commits/master
Kernel: 6af9a26f https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

Platform: CML Mantis with onboard codec ALC3204 in HDA mode
Firmware: 247af133 https://github.com/thesofproject/sof/commits/master
Kernel: 6af9a26f https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware

[dmesg0220.log](https://github.com/thesofproject/sof/files/4230276/dmesg0220.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/4230277/sof_trace.log)

bkokoszx, 2020-02-20:
@lgirdwood 
Thanks, I understand, but do we really need here, in that special case, that the SSP be configured with 4 channels (not 2; codec required this)?

The easiest way returning to previous state is to remove checking channel parameter in dai_verify() function.

Due to channel converter- should it be implemented as separate component or just static dai function? In 2->4 conversion case, should we duplicate first and second channel?

lgirdwood, 2020-02-20:
I would make it a separate component. This way it can be used anywhere and also allows certain post processing on some channels and not others.

bkokoszx, 2020-02-21:
@lgirdwood
I have one more thought - why can't just use mux/demux with special config (1 source/sink)? I think that demux/mux can be use as channel converter in some way

lgirdwood, 2020-02-21:
@bkokoszx ok, if that will do it it will be fine until multi DAI is merged.

plbossart, 2020-02-21:
> @bkokoszx ok, if that will do it it will be fine until multi DAI is merged.

there are no plans to support the multi-dai for now, and I don't get why the demux would be used in 1source 1 sink mode. for SoundWire we need 1src 2 sinks?

bkokoszx, 2020-02-24:
@plbossart 
It's because in that topology SSP has been configured with ```ssp->params.tdm_slots=4```, but when we try to make ```aplay``` with argument ```-c2``` (channels) we configure pipeline with pcm channel parameter equals to 2. There is a mismatch between hardware and pcm configuration and we need to make conversion from 2 to 4 inside the pipeline. As I understand we are able to use DEMUX in one sink/source mode and make that conversion using that component.

bkokoszx, 2020-02-24:
@lgirdwood @plbossart 
Sorry, my bad. I was not aware of variables like: ```config->ssp.tx_slots;``` and ```config->ssp.rx_slots;```, that specify which slots should be filled. When we try to play 2 channel stream and configure ssp with tdm_slots equal to 4 and tx_slots equal to e.g. 0b0011 everything should be correct. We do not need any DEMUX here. I will make PR resolving this issue today/tomorrow.

Liviali155, 2020-02-24:
Error not seen with 20200224  on tgl hda
Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 1d273c3a https://github.com/thesofproject/sof/commits/master
Kernel: 1d2ad939 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

bkokoszx, 2020-02-24:
@Liviali155 @aiChaoSONG 
Could you check whether issue can be reproduce with PR #2430 ?

aiChaoSONG, 2020-02-25:
@bkokoszx issue fixed by #2430

Liviali155, 2020-02-25:
> @Liviali155 @aiChaoSONG
> Could you check whether issue can be reproduce with PR #2430 ?

Error not seen with latest [sof/dev(8bf1a297)](https://github.com/thesofproject/linux/commit/8bf1a297a8b3deda946ee2f9aebab1843e03f155) and [master(8b595069)](https://github.com/thesofproject/sof/commit/8b595069dc23990b09b6cbe686f8f870234aba39) on 
CML laptop with onboard codec ALC3204 in HDA mode
TGL-U RVP with onboard codec ALC711 in HDA mode

Not need  #2430

bkokoszx, 2020-02-26:
I've made reopen. As I understand correctly, #2430 is required for SSP, not for HDA.

aiChaoSONG, 2020-02-26:
@bkokoszx #2430 surely fix this issue. The HDA platform mentioned above is not the same root cause as this one. and even without #2430, the issue on HDA platform cannot reproduced with the latest sof. 
Let's keep this open for a while, and close this after #2430 merged.

aiChaoSONG, 2020-02-28:
close for #2430 merged.

closed 2020-02-28 by aiChaoSONG