thesofproject/sof #2402: [BUG] Stream stall ASRC capture pipeline.

jdrzaszcz, 2020-02-18:
**Describe the bug**
When running simultaneous capture and playback on external loopback (same SSP) with ASRC component on capture pipeline there is stream stall on capture pipeline.
The test applies to 2 channels and other elements of the audio format do not matter.

**Topology**
```
                +------------------+
    +------+    | +---+  +-------+ |
    | Host |----->|Buf|->|SSP Dai|------------------+
    +------+    | +---+  +-------+ |                |
                +------------------+                |
                                                    |
                +---------------------------------+ |
    +------+    | +---+  +----+  +---+  +-------+ | |
    | Host |<-----|Buf|<-|Asrc|<-|Buf|<-|SSP Dai|<--+
    +------+    | +---+  +----+  +---+  +-------+ |
                +---------------------------------+
```
**To Reproduce**
Run one the test 19_02_TestAsrcCaptureSamplingRateChange on https://gitlab.devtools.intel.com/cavs_scripts-py/cavs_scripts-py/tree/task-jdrzaszx-ASRC_component_test python test branch

**Reproduction Rate**
About 50 %

**Environment**
Sof master branch b206b7a8d63c5c940c3c2ad17d3f4030e3705810 commit.

**Example log for WHL**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
[0m    0      2          DMA            437359.041667    437359.031250 src/drivers/intel/cavs/hda-dma.c:405 	hda-dmac: 4 channel 6 -> get[0m
[0m    0      2          DMA            437541.083333       182.041672 src/drivers/intel/cavs/hda-dma.c:587 	hda-dmac: 4 channel 6 -> config[0m
[0m    0      2          DMA            437607.958333        66.875000 src/drivers/intel/cavs/hda-dma.c:462 	hda-dmac: 4 channel 6 -> start[0m
[0m    0      2          DMA            437619.458333        11.500000 src/drivers/intel/cavs/hda-dma.c:327 	hda-dmac: 4 channel 6 -> enable[0m
[0m    0      2      unknown            437679.916667        60.458332 src/schedule/ll_schedule.c:300 	task add 0x9e0c0070 task->priority 4 start 500000 period 500000[0m
[0m    0      2      unknown            437693.416667        13.500000 src/schedule/ll_schedule.c:213 	num_tasks 2 total_num_tasks 2[0m
[0m    0      2           SA            438269.375000       575.958313      src/lib/agent.c:47  	perf plat last 24097 peak 24097 cpu last 57760, peak 4295025056[0m
[0m    0      2      unknown            438285.166667        15.791667 src/schedule/ll_schedule.c:173 	perf plat last 553 peak 553 cpu last 2746, peak 4294970042[0m
[0m    0      2          IPC            460345.625000     22060.458984    src/ipc/handler.c:1110	ipc: hdr 0x30010000 tx (76) > rx (28)[0m
[0m    0      2          IPC            460355.708333        10.083333    src/ipc/handler.c:1117	ipc: pipe 1 comp 0 -> new (type 1)[0m
[0m    0      2         COMP            460377.333333        21.625000     src/audio/host.c:324 	host_new()[0m
[0m    0      2          DMA            460420.500000        43.166668 src/drivers/intel/cavs/hda-dma.c:712 	hda-dmac :5 -> probe[0m
[0m    0      2          DMA            460519.125000        98.625000        src/lib/dma.c:122 	dma_get() ID 5 sref = 1 busy channels 0[0m
[0m    0      2          IPC            463048.833333      2529.708252    src/ipc/handler.c:1152	ipc: pipe 1 buffer 2 -> new (0x80 bytes)[0m
[0m    0      2       BUFFER            463061.375000        12.541667   src/audio/buffer.c:66  	buffer_new()[0m
[0m    0      2       BUFFER            463072.750000        11.375000   src/audio/buffer.c:25  	buffer_alloc()[0m
[0m    0      2          IPC            465608.916667      2536.166748    src/ipc/handler.c:1110	ipc: hdr 0x30010000 tx (80) > rx (28)[0m
[0m    0      2          IPC            465619.041667        10.125000    src/ipc/handler.c:1117	ipc: pipe 1 comp 1 -> new (type 2)[0m
[0m    0      2         COMP            465634.583333        15.541667      src/audio/dai.c:130 	dai_new()[0m
[0m    0      2          DAI            465693.708333        59.125000        src/lib/dai.c:58  	dai_get 1.0 new sref 1[0m
[0m    0      2        POWER            465709.416667        15.708333 src/platform/intel/cavs/lib/pm_runtime.c:243 	dis-dwdma-clk-gating index 0 GPDMA_CLKCTL 00000001[0m
[0m    0      2          DMA            465810.458333       101.041664        src/lib/dma.c:122 	dma_get() ID 0 sref = 1 busy channels 0[0m
[0m    0      2          IPC            467733.625000      1923.166626        src/ipc/ipc.c:328 	ipc: comp sink 2, source 0  -> connect[0m
[0m    0      2         PIPE            467744.916667        11.291667 src/audio/pipeline.c:75  	pipeline: connect comp 0 and buffer 2[0m
[0m    0      2          IPC            469578.333333      1833.416626        src/ipc/ipc.c:354 	ipc: comp sink 1, source 2  -> connect[0m
[0m    0      2         PIPE            469588.916667        10.583333 src/audio/pipeline.c:75  	pipeline: connect comp 1 and buffer 2[0m
[0m    0      2          IPC            472353.333333      2764.416748    src/ipc/handler.c:1110	ipc: hdr 0x30010000 tx (76) > rx (28)[0m
[0m    0      2          IPC            472363.541667        10.208333    src/ipc/handler.c:1117	ipc: pipe 2 comp 5 -> new (type 1)[0m
[0m    0      2         COMP            472383.416667        19.875000     src/audio/host.c:324 	host_new()[0m
[0m    0      2          DMA            472427.416667        44.000000        src/lib/dma.c:122 	dma_get() ID 4 sref = 2 busy channels 1[0m
[0m    0      2          IPC            474776.541667      2349.125000    src/ipc/handler.c:1152	ipc: pipe 2 buffer 7 -> new (0x180 bytes)[0m
[0m    0      2       BUFFER            474790.166667        13.625000   src/audio/buffer.c:66  	buffer_new()[0m
[0m    0      2       BUFFER            474801.416667        11.250000   src/audio/buffer.c:25  	buffer_alloc()[0m
[0m    0      2          IPC            477301.625000      2500.208252    src/ipc/handler.c:1110	ipc: hdr 0x30010000 tx (96) > rx (28)[0m
[0m    0      2          IPC            477311.708333        10.083333    src/ipc/handler.c:1117	ipc: pipe 2 comp 4 -> new (type 19)[0m
[0m    0      2         COMP            477326.333333        14.625000 src/audio/asrc/asrc.c:258 	asrc_new()[0m
[0m    0      2         COMP            477339.583333        13.250000 src/audio/asrc/asrc.c:267 	asrc_new(), source_rate=8000, sink_rate=0, asynchronous_mode=1, operation_mode=1[0m
[0m    0      2          IPC            479532.875000      2193.291748    src/ipc/handler.c:1152	ipc: pipe 2 buffer 8 -> new (0x80 bytes)[0m
[0m    0      2       BUFFER            479547.000000        14.125000   src/audio/buffer.c:66  	buffer_new()[0m
[0m    0      2       BUFFER            479558.250000        11.250000   src/audio/buffer.c:25  	buffer_alloc()[0m
[... 363 lines cut ...]
[0m    0      2          IPC           4418689.083333    242243.703125    src/ipc/handler.c:484 	ipc: comp 5 -> trigger cmd 0x40000[0m
[0m    0      2         PIPE 4.9       4418700.416667        11.333333 src/audio/pipeline.c:612 	pipeline_trigger()[0m
[0m    0      2         COMP 4.5       4418710.333333         9.916667     src/audio/host.c:272 	host_trigger()[0m
[0m    0      2          DMA           4418722.083333        11.750000 src/drivers/intel/cavs/hda-dma.c:462 	hda-dmac: 4 channel 0 -> start[0m
[0m    0      2          DMA           4418733.458333        11.375000 src/drivers/intel/cavs/hda-dma.c:327 	hda-dmac: 4 channel 0 -> enable[0m
[0m    0      2         COMP 4.4       4418792.541667        59.083332 src/audio/asrc/asrc.c:345 	asrc_trigger()[0m
[0m    0      2         COMP 4.6       4418803.541667        11.000000      src/audio/dai.c:503 	dai_comp_trigger(), command = 1[0m
[0m    0      2         COMP 4.6       4418813.333333         9.791667      src/audio/dai.c:514 	dai_comp_trigger(), START[0m
[0m    0      2          DAI 1.0       4418823.791667        10.458333 src/drivers/intel/cavs/ssp.c:702 	ssp_trigger() cmd 1[0m
[0m    0      2          DAI 1.0       4418835.333333        11.541667 src/drivers/intel/cavs/ssp.c:631 	ssp_start()[0m
[0m    0      2      unknown           4418900.041667        64.708336 src/schedule/ll_schedule.c:300 	task add 0xbe0603c0 task->priority 0 start 0 period 1000[0m
[0m    0      2      unknown           4418914.833333        14.791667 src/schedule/ll_schedule.c:213 	num_tasks 3 total_num_tasks 3[0m
[... cut ...]
[0m    0      2          DAI 1.0       4423177.625000        10.208333 src/drivers/intel/cavs/ssp.c:702 	ssp_trigger() cmd 1[0m
[0m    0      2          DAI 1.0       4423189.625000        12.000000 src/drivers/intel/cavs/ssp.c:631 	ssp_start()[0m
[0m    0      2      unknown           4423591.750000       402.125000 src/schedule/ll_schedule.c:300 	task add 0xbe060300 task->priority 0 start 0 period 1000[0m
[0m    0      2      unknown           4423606.833333        15.083333 src/schedule/ll_schedule.c:213 	num_tasks 4 total_num_tasks 4[0m

```

singalsu, 2020-02-18:
Thanks for reporting! I'll try repeat this here with 16 kHz or 8 kHz DMIC capture pipeline.

lgirdwood, 2020-02-21:
@singalsu any update ?

mengdonglin, 2020-02-24:
@jdrzaszcz which platform is used to reproduce this issue?

jdrzaszcz, 2020-02-24:
@mengdonglin I checked it on APL and WHL RVP

lgirdwood, 2020-02-24:
@jdrzaszcz btw, which buffer(s) are overrun/underun here ? I wonder if we need more buffering around the ASRC to deal with drift.

jdrzaszcz, 2020-02-24:
@lgirdwood As far as I know there is no overrun/underun here only strem stall on capture pipeline and I don't see any overrun/underun in logs.

lgirdwood, 2020-02-24:
@jdrzaszcz can you point out the error in the logs. I cant see anything after a quick check. I would expect to see a scheduler error if the pipeline is not run (stalled ?).

singalsu, 2020-02-24:
I'm not able to capture at 24 kHz via ALSA (Warning: rate is not accurate (requested = 24000Hz, got = 22050Hz)). Do you know @lgirdwood if I can modify arecord or ALSA/ASoC kernel to achieve that?

singalsu, 2020-02-24:
I'm using this topology (modified sof-apl-asrc-pcm512x.m4 - changed asrc-volume-capture to asrc-capture):

[image]

I'm not able to repeat the stall issue by capturing alternatively 16 -> 32/48 kHz. I can't do 24 kHz due to apparent ALSA limitation.

I'll still tweak the DMIC16k configration to 8 kHz (the first capture in log above) and see if it triggers something. I have never tested ASRC with 8 kHz configured DAI.

singalsu, 2020-02-24:
With 8 kHz DMIC DAI and 1st capture attempt 8 -> 8 kHz I got this warn (comp_update_buffer_produce(), no bytes to produce, source->comp.id = 18, source->comp.type = 2, sink->comp.id = 15, sink->comp.type = 19) into trace. Component 18 is DMIC1.IN and component 15 is ASRC3.0. However the captured wav clip was still OK.

```
    0      2          IPC         212867555.260417       737.031250    src/ipc/handler.c:484 	ipc: comp 14 -> trigger cmd 0x40000
    0      2         PIPE 3.19    212867558.854167         3.593750 src/audio/pipeline.c:620 	pipeline_trigger()
    0      2         COMP 3.14    212867562.239583         3.385417     src/audio/host.c:272 	host_trigger()
    0      2          DMA         212867566.041667         3.802083 intel/cavs/hda-dma.c:462 	hda-dmac: 4 channel 1 -> start
    0      2          DMA         212867569.531250         3.489583 intel/cavs/hda-dma.c:327 	hda-dmac: 4 channel 1 -> enable
    0      2         COMP 3.15    212867606.979167        37.447918 rc/audio/asrc/asrc.c:342 	asrc_trigger()
    0      2         COMP 3.18    212867610.625000         3.645833      src/audio/dai.c:553 	dai_comp_trigger(), command = 1
    0      2         COMP 3.18    212867613.958333         3.333333      src/audio/dai.c:564 	dai_comp_trigger(), START
    0      2          DAI 2.1     212867622.864583         8.906250 rs/intel/cavs/dmic.c:1279	dmic_start(), dmic->fifo_b
    0      2          DAI 2.1     212867627.031250         4.166667 rs/intel/cavs/dmic.c:1305	dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2          DAI 2.1     212867631.666667         4.635417 rs/intel/cavs/dmic.c:1305	dmic_start(), mic_a = 0, mic_b = 0, fir_a = 0, fir_b = 0
    0      2  SCHEDULE_LL         212867636.562500         4.895833 hedule/ll_schedule.c:300 	task add 0xbe078348 task->priority 4 start 1000 period 1000
    0      2  SCHEDULE_LL         212867641.197917         4.635417 hedule/ll_schedule.c:213 	num_tasks 3 total_num_tasks 3
    0      2          DAI 2.1     212867645.208333         4.010417 rs/intel/cavs/dmic.c:1375	dmic_start(), done active_fifos = 1
    0      2  SCHEDULE_LL         212867652.135417         6.927083 hedule/ll_schedule.c:300 	task add 0xbe04ed00 task->priority 0 start 0 period 1000
    0      2  SCHEDULE_LL         212867656.510417         4.375000 hedule/ll_schedule.c:213 	num_tasks 4 total_num_tasks 4
    0      2          DMA         212867814.947917       158.437500 src/drivers/dw/dma.c:1036	dw_dma_avail_data_size() size is 0!
    0      2       BUFFER 3.17    212867821.510417         6.562500   src/audio/buffer.c:149 	comp_update_buffer_produce(), no bytes to produce, source->comp.id = 18, source->comp.type = 2, sink->comp.id = 15, sink->comp.type = 19
    0      2         COMP 3.14    212867831.250000         9.739583     src/audio/host.c:753 	host_copy(): no bytes to copy
    0      2  SCHEDULE_LL         213268889.791667    401058.531250 hedule/ll_schedule.c:104 	task complete 0xbe078348
    0      2  SCHEDULE_LL         213268893.437500         3.645833 hedule/ll_schedule.c:107 	num_tasks 3 total_num_tasks 3
    0      2          IPC         217974546.927083   4705653.500000    src/ipc/handler.c:484 	ipc: comp 14 -> trigger cmd 0x50000
    0      2         PIPE 3.19    217974550.885417         3.958333 src/audio/pipeline.c:620 	pipeline_trigger()
    0      2         COMP 3.14    217974554.322917         3.437500     src/audio/host.c:272 	host_trigger()
    0      2          DMA         217974558.333333         4.010417 intel/cavs/hda-dma.c:538 	hda-dmac: 4 channel 1 -> stop
    0      2         COMP 3.15    217974562.343750         4.010417 rc/audio/asrc/asrc.c:342 	asrc_trigger()
    0      2         COMP 3.18    217974565.833333         3.489583      src/audio/dai.c:553 	dai_comp_trigger(), command = 0
    0      2         COMP 3.18    217974569.166667         3.333333      src/audio/dai.c:612 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         217974572.291667         3.125000 src/drivers/dw/dma.c:415 	dw_dma_stop(): dma 0 channel 1 stop
    0      2          DAI 2.1     217974578.072917         5.781250 rs/intel/cavs/dmic.c:1411	dmic_stop(), dmic_active_fifos = 1
    0      2  SCHEDULE_LL         217974583.437500         5.364583 hedule/ll_schedule.c:378 	task cancel 0xbe078348
    0      2  SCHEDULE_LL         217974587.656250         4.218750 hedule/ll_schedule.c:378 	task cancel 0xbe04ed00
    0      2  SCHEDULE_LL         217974591.822917         4.166667 hedule/ll_schedule.c:245 	num_tasks 2 total_num_tasks 2
    0      2          IPC         217974756.927083       165.104172    src/ipc/handler.c:357 	ipc: comp 14 -> free
    0      2         PIPE 3.19    217974760.260417         3.333333 src/audio/pipeline.c:691 	pipeline_reset()

```

singalsu, 2020-02-24:
Still no stalls. I can get with 8 kHz DAI the above warning in about 10% of cases. The used conversion 8 -> X kHz does not seem to impact. The file length does not vary per same capture rate.

@jdrzaszcz I'm not able to access the repo for the script that was used. What kind of capture duration and repeat pattern does it use? I wonder I should do a lot of very short captures or alternatively let it suspend in between or something.

jdrzaszcz, 2020-02-25:
@singalsu I'm useing 3 seconds sine wave.

jdrzaszcz, 2020-02-25:
@lgirdwood Unfortunately there is no error in the log. I only have TimeoutError: Capture stream not running for 1 seconds in the test, and I have zeros recorded in the capture file.

singalsu, 2020-02-25:
@jdrzaszcz In your setup is the capture SSP a slave for an external clock? I don't have here a slave capture interface to test with. I've only tested with slave SSP for playaback (SPDIF out). If it was master, DMIC should be pretty similar as SSP. Though DMIC has a deeper FIFO and maybe that warning that I saw was due to the FIFO not reaching watermark fast enough with low capture rates.

@lgirdwood Can the limitation of no 24 kHz capture in ALSA be worked around? These issues happened in 8 -> 24 kHz and 16 -> 24 kHz capture and I'm not able to test them in Linux.

jdrzaszcz, 2020-02-25:
@singalsu  In my setup I have loopback on one SSP.

singalsu, 2020-02-25:
@jdrzaszcz Since I'm not able to replicate could you do a favor and get the firmware trace output for failing test with a modified asrc.c. Please change in line  839 the comp_dbg() to comp_info(), similarly line 785 comp_cl_dbg() to comp_info(). It would show how many frames the ASRC can copy every period and how the DAI skew tracking is doing. (Or is there nowadays a way to enable verbose trace only for one component?)

If ASRC does not report any copied frames then trace also at line 813 the frames_src and frames_snk variables.

singalsu, 2020-02-25:
> @singalsu In my setup I have loopback on one SSP.

Is the SSP clocked internally by SoC or by external non-synchronized clock?

jdrzaszcz, 2020-02-25:
SSP is clocked internally by SoC.

jdrzaszcz, 2020-02-25:
@singalsu I made the changes, test result in attachment. 
[2020_02_25_11_11_19.zip](https://github.com/thesofproject/sof/files/4249421/2020_02_25_11_11_19.zip)

singalsu, 2020-02-25:
Strange, it looks like the third run when capturing 24k -> 24k lasted for 3s and still the wav file was empty. Maybe the output wasn't flushed. But there was no asrc_reset() and asrc_free() in the end and no reported output frames after last skew report.

I wonder if ASRC got stuck into a busy loop. I remember fixed such issue in push mode with a check and emergency bail out. Maybe that can happen in pull mode loop too. I need to review the algo code and see if that's possible. Thanks!

singalsu, 2020-02-26:
@jdrzaszcz Can you please retest with PR #2443 . I was able to replicate the quality issue here in testbench run. But since I wasn't able to replicate the stall in testbench nor in my UP2 setup I'm not sure this helps with stall. But I think the copy() frame counts control could have requested a conflicting amount to input and output in pull mode. The PR also gives more flex for push mode where also an unnecessary constraint was placed.

jdrzaszcz, 2020-02-27:
@singalsu Since PR #2443 has been merged i tested ASRC on capture pipline with master branch: 109450e6fdaa9308732e0358dc10680e673f7d58 commit. Unfortunately there is still stream stall on capture pipeline.

singalsu, 2020-02-27:
@jdrzaszcz Thanks for testing! I'll continue trying to replicate the issue by forcing more drift etc in real device and testbench.

Did you run the THD+N test, were the low performing capture conversions improved? If there was no improvement then the tests there did not trigger the issues those I found yesterday in testbench. The corruption was so bad that there was every now and then (about every 100 ms) a visible small sawtooth edge in the sine waveform.

singalsu, 2020-02-27:
The DAI skew factor looks very normal in your test runs for 8->24, 16->24, and 24->24 kHz. After filtering it deviates very little from 1.0 for every copy(). So the problem is not skew related. The plot is extract from DMA trace.

[image]

singalsu, 2020-02-27:
@jdrzaszcz I found a potential end of data hazard from pull mode copy() functions. The function could get stuck if the end of input is reached too early. Could you please test if this patch would impact the freeze?

[asrc_patch.zip](https://github.com/thesofproject/sof/files/4263077/asrc_patch.zip)

lgirdwood, 2020-02-27:
```diff
From 8caf8712701fa25d76fce44c66cfe53620fd62d7 Mon Sep 17 00:00:00 2001
From: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
Date: Thu, 27 Feb 2020 19:27:40 +0200
Subject: [PATCH] ASRC: Exit pull mode loop if reached end of input in copy()

This patch fixes a potential input data corruption and fail
to exit the filter filter loop.

---
 src/audio/asrc/asrc_farrow.c | 48 +++++++++++++++++-------------------
 1 file changed, 23 insertions(+), 25 deletions(-)

diff --git a/src/audio/asrc/asrc_farrow.c b/src/audio/asrc/asrc_farrow.c
index 038186054102..89a54acd3963 100644
--- a/src/audio/asrc/asrc_farrow.c
+++ b/src/audio/asrc/asrc_farrow.c
@@ -1232,21 +1232,20 @@ enum asrc_error_code asrc_process_pull16(struct comp_dev *dev,
 	/* Run state machine until number of output samples are written */
 	while (index_output_frame < output_num_frames) {
 		if (src_obj->time_value_pull < TIME_VALUE_ONE) {
-			/* Consume one input sample */
-			if (src_obj->io_buffer_idx != write_index) {
-				asrc_write_to_ring_buffer16(src_obj,
-							    input_buffers,
-							    src_obj->io_buffer_idx);
-				src_obj->io_buffer_idx++;
+			/* Consume one input sample if not end reached */
+			if (src_obj->io_buffer_idx == write_index)
+				break;
 
-				/* Wrap around */
-				if (src_obj->io_buffer_idx >=
-				    src_obj->io_buffer_length &&
-				    src_obj->io_buffer_mode == ASRC_BM_CIRCULAR)
-					src_obj->io_buffer_idx = 0;
+			asrc_write_to_ring_buffer16(src_obj, input_buffers,
+						    src_obj->io_buffer_idx);
+			src_obj->io_buffer_idx++;
 
-				(*input_num_frames)++;
-			}
+			/* Wrap around */
+			if (src_obj->io_buffer_mode == ASRC_BM_CIRCULAR &&
+			    src_obj->io_buffer_idx >= src_obj->io_buffer_length)
+				src_obj->io_buffer_idx = 0;
+
+			(*input_num_frames)++;
 
 			/* Update time as Q5.27 */
 			src_obj->time_value = (((int64_t)TIME_VALUE_ONE -
@@ -1336,20 +1335,19 @@ enum asrc_error_code asrc_process_pull32(struct comp_dev *dev,
 	while (index_output_frame < output_num_frames) {
 		if (src_obj->time_value_pull < TIME_VALUE_ONE) {
 			/* Consume input sample */
-			if (src_obj->io_buffer_idx != write_index) {
-				asrc_write_to_ring_buffer32(src_obj,
-							    input_buffers,
-							    src_obj->io_buffer_idx);
-				src_obj->io_buffer_idx++;
+			if (src_obj->io_buffer_idx == write_index)
+				break;
 
-				/* Wrap around */
-				if (src_obj->io_buffer_idx >=
-				    src_obj->io_buffer_length &&
-				    src_obj->io_buffer_mode == ASRC_BM_CIRCULAR)
-					src_obj->io_buffer_idx = 0;
+			asrc_write_to_ring_buffer32(src_obj, input_buffers,
+						    src_obj->io_buffer_idx);
+			src_obj->io_buffer_idx++;
 
-				(*input_num_frames)++;
-			}
+			/* Wrap around */
+			if (src_obj->io_buffer_mode == ASRC_BM_CIRCULAR &&
+			    src_obj->io_buffer_idx >= src_obj->io_buffer_length)
+				src_obj->io_buffer_idx = 0;
+
+			(*input_num_frames)++;
 
 			/* Update time as Q5.27 */
 			src_obj->time_value = (((int64_t)TIME_VALUE_ONE -
-- 
2.20.1
```

jdrzaszcz, 2020-02-28:
@singalsu Hi, i checked it with your path and there is still a stream stall. The reproduction rate is even higher than without it. Example logs in attachment. 
[2020_02_28_07_49_21.zip](https://github.com/thesofproject/sof/files/4265640/2020_02_28_07_49_21.zip)
[2020_02_28_07_51_20.zip](https://github.com/thesofproject/sof/files/4265641/2020_02_28_07_51_20.zip)

singalsu, 2020-02-28:
Thanks for testing, I really appreciate your help! 

Strange, here that exit condition triggers only in the end of playback. My capture loop script captures 100 times 8->8/16/32 kHz and not a single stall ever. Though I'm capturing from DMIC and not from SSP.

Can you revert the patch and go to git master and error traces to pull16() to see where it gets stuck. I can't think of anything else now. Error trace in mailbox should be better than DMA trace since it does not leave last items unflushed.

[asrc_pull16_debug.zip](https://github.com/thesofproject/sof/files/4266021/asrc_pull16_debug.zip)

jdrzaszcz, 2020-02-28:
I Reverted the changes and apply new patch.
Logs in attachment.
[2020_02_28_10_11_17.zip](https://github.com/thesofproject/sof/files/4266245/2020_02_28_10_11_17.zip)
[2020_02_28_10_11_31.zip](https://github.com/thesofproject/sof/files/4266246/2020_02_28_10_11_31.zip)

singalsu, 2020-02-28:
@jdrzaszcz I'm sorry, there's no new traces. It seems like the test was run for 32 bit capture while I patched the 16 bit data function. Here's added traces to pull32() to get the traces.

[asrc_pull16_pull32_debug.zip](https://github.com/thesofproject/sof/files/4266347/asrc_pull16_pull32_debug.zip)

jdrzaszcz, 2020-02-28:
@singalsu Oh, sorry i didn't notice, that it was connected with bit rate.

I rerun the tests with ASRC on capture and in some cases I didn't got any logs. 
I attached logs of all test cases.
[ASRC_capture.zip](https://github.com/thesofproject/sof/files/4266579/ASRC_capture.zip)

singalsu, 2020-02-28:
There's three cases with FAIL and these pX traces present:

- 19_01_TestAsrcCapture16000Hz16b16b2ch

- 19_01_TestAsrcCapture24000Hz16b16b2ch

- 19_01_TestAsrcCapture48000Hz32b32b2ch

Two first of them have "p3" as last entry, no exit "p4". One has "p0" as last entry. Is it so that the test system does not capture the mailbox error trace? I've found with Linux driver that the DMA trace to not flush always the latest output so it's not sure the p3 and p0 are the last whereabouts of ASRC. Do you know if with slim driver there's similar issue.

But anyway I'll debug now FIR coefficients update right after "p3" if there could be hazards. 

But

jdrzaszcz, 2020-02-28:
On slim driver we are using the same tool as on Linux driver.

singalsu, 2020-02-28:
OK, yes it looks like dma_trace_1_sink.log wraps on longer cases so it must be the etrace.

After "p3" there's straightforward deterministic loop code with intrinsics optimizations. I can't spot anything hazardous there. What if there would be 64 bit read/write align fault trap or panic in FW, would it show up in the log files with slim driver?

jdrzaszcz, 2020-02-28:
As far as I know when we have panic, all data is dumped from the memory window and save to file.

paulstelian97, 2020-02-28:
@jdrzaszcz I have seen that code but that doesn't work for hangs. This one is an instance of hanging so the automatic flush on panic isn't going to happen.

lgirdwood, 2020-02-28:
@singalsu if your suspicious of the intrinsics you could replace with a small C version or partially comment out sections and see if we are still hanging.

singalsu, 2020-03-02:
I'll test here if it's possible to run the generic C as xt-xcc build in realtime, if yes I'll make patch to test. Good idea!

singalsu, 2020-03-02:
Unfortunately even 16 kHz capture is too slow with generic C on cAVS1.5 (APL). I suppose cAVS1.8 is not significantly faster.

Would you @jdrzaszcz be OK for another added traces test? Since some traces seem to be left out, I'd add even more and finer grained traces into pull mode functions to narrow the potentially freezing functions.

jdrzaszcz, 2020-03-02:
@singalsu Ok, add more traces and then i will rerun the tests.

singalsu, 2020-03-02:
@jdrzaszcz Thanks! Here's one more to try with more "pX" output to trace. 
[more_asrc_debug.zip](https://github.com/thesofproject/sof/files/4275447/more_asrc_debug.zip)

jdrzaszcz, 2020-03-03:
@ Hi, I have run tests with ASRC on capture with your patch and I always have stream stall.
I attached logs of all test cases.

[ASRC_capture.zip](https://github.com/thesofproject/sof/files/4279623/ASRC_capture.zip)

singalsu, 2020-03-03:
It's quite interesting that every case failed this time. At least now I have a lot of cases to study, thanks!

singalsu, 2020-03-03:
Only the directories 19_01_TestAsrcCapture8000Hz16b16b2ch, 24b32b2ch, and 32b32b2ch contained the traces.  Two of them exits normally with "p6" the copy() filter core while one has as last item "p5". That does not make sense since it must follow "p0" or "p6". So the trace cannot be complete.

It's not possible to study why all other tests failed this time. The trace outputs contain only the header line. What might wrong in the test?

```
$ ls -lR 19_01_TestAsrcCapture* | grep dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu  37548 maalis  3 07:27 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu  32162 maalis  3 07:27 dma_trace_1_sink.log
-rw-r--r-- 1 singalsu singalsu  36940 maalis  3 07:28 dma_trace_1_sink.log

$ ls -lR 19_01_TestAsrcCapture* | grep dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:27 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:27 dma_trace_1.log
-rw-r--r-- 1 singalsu singalsu    101 maalis  3 07:28 dma_trace_1.log

```
Also how can we know the problem is not in the host component? It reports in every trace "host_copy(): no bytes to copy only some 450 us after host trigger and 340 us after DAI trigger start. Could it stall in the slim driver setup?

Also there's "dw_dma_avail_data_size() size is 0!" 290 us after DAI trigger start. and "	comp_update_buffer_produce(), no bytes to produce, source->comp.id = 6, source->comp.type = 2, sink->comp.id = 4, sink->comp.type = 19" after 350 us. Component 6 is DAI and component 4 is ASRC. These warnings happen pretty soon after pipeline start.

Does someone there know what to check if these host and DAI warnings are normal?

singalsu, 2020-03-19:
There's a fix proposal in PR #2587

singalsu, 2020-03-20:
@jdrzaszcz PR #2587 is now merged. Can you please re-test?

jdrzaszcz, 2020-03-23:
I have re-tested this issue with master branch and can't reproduce.

closed 2020-03-23 by jdrzaszcz

singalsu, 2020-03-23:
Great! Also thanks for a lot of help from you with this issue!