thesofproject/sof #2404: [BUG][CML Mantis]I/O error when do Capture-pause-release test  using (0,0)

Liviali155, 2020-02-18:
***Describe the bug***
I/O error when do Capture-pause-release test using (0,0)

***To Reproduce***
1.sudo reboot
2. ./sof-test-pause-release.sh -t pcm0c (script as attachment)

[image]

***Reproduction Rate***
10/10

***Expected behavior***
There is no I/O error

***Actual result***
I/O error
[image]

<details>
 <summary>dmesgerror.log</summary>

````
[ 7739.251053] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7739.251076] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7739.732448] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[ 7739.732453] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 7739.732726] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 7739.733241] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[ 7739.733245]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
[ 7739.733251] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
[ 7739.733755] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[ 7739.733760] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[ 7739.733975] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 7739.733977]  HDA Analog: ASoC: trigger FE cmd: 4 failed: -22
[ 7739.750774] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7739.750796] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7740.240197] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[ 7740.240209] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 7740.240448] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 7740.240462] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[ 7740.240471] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 7740.240837] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 7740.240856] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[ 7740.240876] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG    
````
</details>

***Environment***
Platform: CML Mantis with onboard codec ALC3204 in HDA mode
Firmware: 53d35e49 https://github.com/thesofproject/sof/commits/master
Kernel: b2d98237 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware

[dmesg0218.log](https://github.com/thesofproject/sof/files/4218828/dmesg0218.log)
[sof_trace0218.log](https://github.com/thesofproject/sof/files/4218831/sof_trace0218.log)
[sof-test-pause-release.txt](https://github.com/thesofproject/sof/files/4218835/sof-test-pause-release.txt)

Liviali155, 2020-02-18:
Seems a Firmware regression issue
I changed firmware to 20200110(commit:c6baa979),error not seen

Liviali155, 2020-02-18:
Bi-sect result:
caused by Firmware commit 91fb594b[](https://github.com/thesofproject/sof/commit/91fb594bd9455fcf8415a588cf45e6294ac46e0e)

Liviali155, 2020-02-18:
Related #2280

lgirdwood, 2020-02-18:
@Liviali155 is this fatal ? Does audio continue or do you need to restart aplay ?
@slawblauciak This looks like a race that's easy to reproduce with slower clock ?

Liviali155, 2020-02-19:
> is this fatal ? Does audio continue or do you need to restart aplay

After error  occurred  arecord stop,but if arecord again can work normally

lgirdwood, 2020-02-21:
@slawblauciak @mmaka1 can we make the 400M clock the default for CML etc, the low power clock can then be optionally selected by user who need it until the switching is dynamic at runtime.

slawblauciak, 2020-02-25:
Hmm, the FW log doesn't appear complete. I can see many pause/release operations being done from the dmesg, but only one in the FW trace. Is the trace you provided cut in any way? Does it always end up like that?

paulstelian97, 2020-02-25:
When doing sof-logger is the -l parameter offered? etrace has a tendency of being incomplete.

Liviali155, 2020-02-25:
I reproduced the issue again,this time I got the sof_errortrace,but seems I didn't catch the complete sof_trace used command "sudo sof-logger -l /etc/sof/sof-*.ldc -t -o sof_trace0225.log" 

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         397519970.520833 397519968.000000 intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.15    397519990.572917        20.052084      src/audio/dai.c:632 	dai_report_xrun() error: overrun due to no data available
    0      1         COMP 2.15    397520006.927083        16.354166      src/audio/dai.c:778 	comp_overrun() error: sink->free = 768, copy_bytes = 0
    0      1         PIPE         397520128.020833       121.093750 src/audio/pipeline.c:767 	pipeline_copy() error: ret = -61, start->comp.id = 15, dir = 0
    0      1         PIPE 2.16    397520148.229167        20.208334 src/audio/pipeline.c:934 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP 2.15    398270254.791667    750106.562500 rc/audio/component.c:158 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1         PIPE         398270275.000000        20.208334 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 2
    0      1          IPC         398270295.312500        20.312500    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x60000 failed -22
    0      1         COMP 2.15    398270656.718750       361.406250 rc/audio/component.c:134 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1         PIPE         398270676.979167        20.260416 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 3
    0      1          IPC         398270696.302083        19.322916    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x70000 failed -22
    0      1         COMP 2.15    399030622.447917    759926.125000 rc/audio/component.c:158 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1         PIPE         399030643.489583        21.041666 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 2
    0      1          IPC         399030663.072917        19.583334    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x60000 failed -22
    0      1         COMP 2.15    399031854.635417      1191.562500 rc/audio/component.c:134 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1         PIPE         399031875.260417        20.625000 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 3
    0      1          IPC         399031894.739583        19.479166    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x70000 failed -22
    0      1         COMP 2.10    399665198.750000    633304.000000 rc/audio/component.c:167 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP 2.11    399665240.625000        41.875000 rc/audio/component.c:167 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
````
</details>

[dmesg0225.log](https://github.com/thesofproject/sof/files/4250194/dmesg0225.log)
[sof_errortrace0225.log](https://github.com/thesofproject/sof/files/4250195/sof_errortrace0225.log)
[sof_trace0225.log](https://github.com/thesofproject/sof/files/4250196/sof_trace0225.log)

juimonen, 2020-02-25:
@Liviali155 can you also try to reproduce this with patch in: https://github.com/thesofproject/linux/pull/1689? I'm wondering if the codec gets into some strange state because we are switching it on/off like crazy...

Liviali155, 2020-02-26:
> @Liviali155 can you also try to reproduce this with patch in: [thesofproject/linux#1689](https://github.com/thesofproject/linux/pull/1689)? I'm wondering if the codec gets into some strange state because we are switching it on/off like crazy...

Tried with thesofproject/linux#1689,log as attachment,seems still didn't catch the complete sof_trace
<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA          90495923.802083  90495920.000000 intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.15     90495943.802083        20.000000      src/audio/dai.c:632 	dai_report_xrun() error: overrun due to no data available
    0      1         COMP 2.15     90495960.156250        16.354166      src/audio/dai.c:778 	comp_overrun() error: sink->free = 768, copy_bytes = 0
    0      1         PIPE          90496080.833333       120.677086 src/audio/pipeline.c:767 	pipeline_copy() error: ret = -61, start->comp.id = 15, dir = 0
    0      1         PIPE 2.16     90496100.937500        20.104166 src/audio/pipeline.c:934 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP 2.15     91249801.770833    753700.812500 rc/audio/component.c:158 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1         PIPE          91249821.875000        20.104166 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 2
    0      1          IPC          91249842.031250        20.156250    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x60000 failed -22
    0      1         COMP 2.15     91251055.468750      1213.437500 rc/audio/component.c:134 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1         PIPE          91251076.510417        21.041666 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 3
    0      1          IPC          91251095.833333        19.322916    src/ipc/handler.c:511 	ipc: comp 9 trigger 0x70000 failed -22
    0      1         COMP 2.10     91884250.364583    633154.500000 rc/audio/component.c:167 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP 2.11     91884292.343750        41.979168 rc/audio/component.c:167 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
````
</details>

[dmesg1689.log](https://github.com/thesofproject/sof/files/4253093/dmesg1689.log)
[sof_errortrace1689.log](https://github.com/thesofproject/sof/files/4253095/sof_errortrace1689.log)
[sof_trace1689.log](https://github.com/thesofproject/sof/files/4253096/sof_trace1689.log)

juimonen, 2020-02-26:
@Liviali155 ok thanks, at least the error trace seems to be consistent so xrun (no data), recover fail, and pipeline stop.

tlauda, 2020-03-03:
@lgirdwood @Liviali155 Basically the same problem as with #2467. If L1SEN is disabled, then it looks like some kind of pause/release sequence problem with this HDA codec. In the FW log we can clearly see, that the new data hasn't arrived on HDA link after releasing the stream. I'm suggesting codec issue, since we haven't seen this kind of error on CI for a long long time (since the HDA DAI sequence has been fixed in kernel).

lgirdwood, 2020-03-04:
+ @kv2019i @ranj063 @plbossart fyi - what are we doing here on the host side for HDA pause/release. My understanding is that the HDA link may power OFF (stream STOP only ?) after a timeout period to save power (i.e. codec then needs to be brought back up etc).

Liviali155, 2020-03-05:
On CFL-S RVP also can reproduce this issue
1.cd sof-test(as attachment)
2.cd test-case
3.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
4.check-pause-resume.sh -c 100 -m capture

I/O error when pause/release arecord(0,0),dmesg show "ipc error for 0x60060000 size "
Platform: CFL-S RVP with onboard codec ALC700 in HDA mode

<details>
 <summary>dmesg_error.log</summary>

````
    da-4 kernel: [ 3202.720799] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
jf-cfl-rvp-hda-4 kernel: [ 3202.720841] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
jf-cfl-rvp-hda-4 kernel: [ 3203.050245] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
jf-cfl-rvp-hda-4 kernel: [ 3203.050256] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
jf-cfl-rvp-hda-4 kernel: [ 3203.050647] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.051241] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
jf-cfl-rvp-hda-4 kernel: [ 3203.051259]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.051269] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
jf-cfl-rvp-hda-4 kernel: [ 3203.051843] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
jf-cfl-rvp-hda-4 kernel: [ 3203.051852] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
jf-cfl-rvp-hda-4 kernel: [ 3203.052167] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.052176]  HDA Analog: ASoC: trigger FE cmd: 4 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.052314] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
jf-cfl-rvp-hda-4 kernel: [ 3203.052322] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
jf-cfl-rvp-hda-4 kernel: [ 3203.052650] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.053245] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
jf-cfl-rvp-hda-4 kernel: [ 3203.053255]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.053265] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
jf-cfl-rvp-hda-4 kernel: [ 3203.053845] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
jf-cfl-rvp-hda-4 kernel: [ 3203.053854] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
jf-cfl-rvp-hda-4 kernel: [ 3203.054127] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.054135]  HDA Analog: ASoC: trigger FE cmd: 4 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.220797] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
jf-cfl-rvp-hda-4 kernel: [ 3203.220838] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
jf-cfl-rvp-hda-4 kernel: [ 3203.658219] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
jf-cfl-rvp-hda-4 kernel: [ 3203.658230] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
jf-cfl-rvp-hda-4 kernel: [ 3203.658598] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.659198] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
jf-cfl-rvp-hda-4 kernel: [ 3203.659209]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.659220] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
jf-cfl-rvp-hda-4 kernel: [ 3203.659751] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
jf-cfl-rvp-hda-4 kernel: [ 3203.659758] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
jf-cfl-rvp-hda-4 kernel: [ 3203.660080] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
jf-cfl-rvp-hda-4 kernel: [ 3203.660088]  HDA Analog: ASoC: trigger FE cmd: 4 failed: -22
jf-cfl-rvp-hda-4 kernel: [ 3203.660215] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
jf-cfl-rvp-hda-4 kernel: [ 3203.660222] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
jf-cfl-rvp-hda-4 kernel: [ 3203.660586] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 

jf-cfl-rvp-hda-4 kernel: [ 3203.661173] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg
````
</details>

<details>
 <summary>sof_errortrace.log</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         4003043952.083333              nan intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.15    4003043972.760417        20.677084      src/audio/dai.c:642 	dai_report_xrun() error: overrun due to no data available
    0      1         COMP 2.15    4003043989.114583        16.354166      src/audio/dai.c:919 	comp_overrun() error: sink->free = 768, copy_bytes = 0
    0      1         PIPE         4003044112.760417       123.645836 src/audio/pipeline.c:789 	pipeline_copy() error: ret = -61, start->comp.id = 15, dir = 0
    0      1         PIPE 2.16    4003044133.072917        20.312500 src/audio/pipeline.c:961 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP 2.15    4003802801.510417    758668.437500 rc/audio/component.c:160 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
[... cut ...]
    0      1         COMP 2.10    4005197918.645833    630765.500000 rc/audio/component.c:169 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP 2.11    4005197960.468750        41.822918 rc/audio/component.c:169 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4291401/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4291402/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4291404/slogger.txt)
[sof-test.zip](https://github.com/thesofproject/sof/files/4291418/sof-test.zip)

tlauda, 2020-03-10:
@Liviali155 Please verify with #2512.

Liviali155, 2020-03-10:
> @Liviali155 Please verify with #2512.

Tried with PR #2512 ,error not seen
Platform: CML laptop with onboard codec ALC3204 in HDA mode
               TGL-U RVP with onboard codec ALC711 in HDA mode

tlauda, 2020-03-11:
Fixed with #2512.

closed 2020-03-11 by tlauda