thesofproject/sof #2418: [BUG] No sound on i7-10510U running 5.5.3x

chrysophylax, 2020-02-20:
**Describe the bug**
Failure to load drivers on i7-10510U leads to only Dummy Output being available.

What have you tried to diagnose or workaround this issue?
Running very recent kernel (5.5.3.x)

**To Reproduce**
Boot Intel ClearLinux 5.5.3 on a Lenovo S540 13IML with an i7-10510U CPU. No sound.

**Reproduction Rate**
Every time.

**Expected behavior**
Working sound card output selectable in GNOME or through pactl.

**Impact**
Showstopper for Linux deployment on this SKU.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: N/A
2) Name of the topology file
    * Topology: Unfortunately not sure where this is stored on clearlinux
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo Ideapad S540 13IML with GenuineIntel i7-10510U running ClearLinux 32310 on kernel 5.5.3 

I've cross-filed the bug with the Intel distro guys & gals over here as well https://github.com/clearlinux/distribution/issues/1771

[dmesg.txt](https://github.com/thesofproject/sof/files/4233037/dmesg.txt)
[verbose-lspci.txt](https://github.com/thesofproject/sof/files/4233038/verbose-lspci.txt)

plbossart, 2020-02-20:
You don't have a firmware file in /lib/firmware/intel/sof as shown in 

````
[    8.320505] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cml.ri failed with error -2
[    8.320511] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cml.ri failed err: -2
[    8.321348] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
````

ahkok, 2020-02-20:
Yes, we need to package this and include it.

chrysophylax, 2020-02-20:
Thanks for the explanations both of you.

@plbossart I went to check [releases](https://github.com/thesofproject/sof/releases) but I can' t see any sof-cml.ri file available there. Should I be looking elsewhere?

plbossart, 2020-02-20:
sof-cml.ri should be a symlink to the latest firmware. We don't encode the version in the driver.

chrysophylax, 2020-02-20:
Progress!
I added `sof-cnl-v1.4.2.ri` to /etc/firmware/intel/sof on ClearLinux and created a symlink 
```
drwxr-xr-x 2 root root 4.0K Feb 21 00:14 .
drwxr-xr-x 3 root root 4.0K Feb 21 00:14 ..
lrwxrwxrwx 1 root root   17 Feb 21 00:14 sof-cml.ri -> sof-cnl-v1.4.2.ri
-rw-r--r-- 1 root root 280K Feb 21 00:14 sof-cnl-v1.4.2.ri
```
And the error about sof-cml.ri went away in dmesg.

However, now it can' t find a topology file...

```
[   13.259559] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   13.259874] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   13.259888] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   13.260306] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   13.260318] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   13.260515] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   13.260732] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops)
[   13.269322] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   13.269326] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   13.280677] probe of ehdaudio0D0 returned 1 after 37 usecs
[   13.287659] probe of ehdaudio0D2 returned 1 after 126 usecs
[   13.293675] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   13.293679] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   13.389469] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[   13.389473] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[   13.389705] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   13.392585] calling  skl_hda_audio_init+0x0/0x1000 [snd_soc_skl_hda_dsp] @ 498
[   13.392877] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-hda-generic-2ch.tplg failed with error -2
[   13.392881] sof-audio-pci 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-hda-generic-2ch.tplg failed err: -2
[   13.393811] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -2
[   13.394404] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -2
[   13.395779] initcall skl_hda_audio_init+0x0/0x1000 [snd_soc_skl_hda_dsp] returned 0 after 3113 usecs
```

Where can I download this `sof-hda-generic-2ch.tplg` file?

lgirdwood, 2020-02-21:
@chrysophylax can you check here https://github.com/thesofproject/sof-bin
This will be the fw, topology and debug file repo. It will be rebased over next few days as we are all aligning on layout, but you should be able to find everything.

ahkok, 2020-02-21:
Are these releases going to be tagged, and signed?

ahkok, 2020-02-21:
Looking at one of the branches, I see lots and lots of firmware files (great) but it's entirely unclear which ones are needed - are you going to at least add a sane, simple `Makefile` so that `make DESTDIR=<root> install` can just do the right thing and install the binary files into the right place for most Linux developers?

I can write it.... if you need.

chrysophylax, 2020-02-21:
@lrgirdwood it seems you rebased before I had time to pull down the tplg file. Any other place on the internet where it' s available?

lgirdwood, 2020-02-24:
@chrysophylax appologies, updated now - we are using a branch per release. please checkout stable- v1.4.2. It may still churn a bit.

chrysophylax, 2020-02-24:
Great, thanks for the update @lgirdwood . I just downloaded https://github.com/thesofproject/sof-bin/blob/stable-v1.4.2/lib/firmware/intel/sof-tplg-v1.4.2/sof-hda-generic-2ch.tplg and will try it out on the laptop later today.

chrysophylax, 2020-02-25:
So it works fine for playback, 2ch sound is correctly being played. On the other hand, there is no microphone input available. Per spec, https://psref.lenovo.com/syspool/Sys/PDF/IdeaPad/ideapad_S540_13IML/ideapad_S540_13IML_Spec.PDF there should be some kind of microphone on the laptop.

plbossart, 2020-02-25:
> So it works fine for playback, 2ch sound is correctly being played. On the other hand, there is no microphone input available. 

Can you paste the results of alsa-info.sh somewhere and provide a link? same for the dmesg log.

chrysophylax, 2020-02-27:
Clear Linux updated kernel version to 5.5.6 and we started getting kernel oopsies during boot that prevented loading of the desktop environment. Switching back to 5.5.5 and it worked. The inconvenience to my SO was 'fixed' by removing both the firmware and the topology files. Did something change between versions? We were using the 1.4.2 tplg and firmware

lgirdwood, 2020-02-27:
@chrysophylax any chance you can paste the dmesg showing the oops. To workaround boot crash in the past, I've had to manually prevent SOF modules being loaded at boot and then load afterwards by hand (i.e. tmp moving SOF module dir to ~/ for boot and then restore + modprobe)

chrysophylax, 2020-02-27:
Sure, I'll try to capture them later during the day when I have access to the device again.

plbossart, 2020-02-27:
I don't see anything critical between v5.5.5 and v5.5.6 for the sound subsystem

````
git log v5.5.5..v5.5.6 --oneline -- sound
478c9a918596 ASoC: Intel: consistent HDMI codec probing code
2264e85ec867 ALSA: usb-audio: add quirks for Line6 Helix devices fw>=2.82
2dd9cca2d9db ALSA: hda - Add docking station support for Lenovo Thinkpad T420s
7dabba5b77f4 ALSA: hda/hdmi - add retry logic to parse_intel_hdmi()
24c259557c45 ASoC: SOF: Intel: hda: Fix SKL dai count
72cd0385276c ASoC: wm_adsp: Correct cache handling of new kernel control API
41160b7bdbc3 ALSA: usb-audio: add implicit fb quirk for MOTU M Series
10dc28b1cdc7 ALSA: usb-audio: unlock on error in probe
9664fb1a6686 ASoC: SOF: Intel: hda-dai: fix compilation warning in pcm_prepare
76fccd309761 ASoC: atmel: fix build error with CONFIG_SND_ATMEL_SOC_DMA=m
148e2930ba82 ASoC: txx9: Remove unused rtd variable
2cfa8f60b79c ALSA: usb-audio: Add boot quirk for MOTU M Series
b8f775d0eed8 ASoC: Intel: sof_rt5682: Ignore the speaker amp when there isn't one.
79861be01c5b ALSA: sh: Fix compile warning wrt const
5e329c90de80 ALSA: hda/realtek - Apply mic mute LED quirk for Dell E7xx laptops, too
1f85ea289449 ALSA: sh: Fix unused variable warnings
92f2b0478613 ASoC: soc-topology: fix endianness issues
961034d5787d ALSA: usx2y: Adjust indentation in snd_usX2Y_hwdep_dsp_status
2f6e40d606c6 ASoC: SOF: Intel: hda: solve MSI issues by merging ipc and stream irq handlers
3a33c6004143 ASoC: intel: sof_rt5682: Add support for tgl-max98357a-rt5682
c27bcb3d09e7 ASoC: intel: sof_rt5682: Add quirk for number of HDMI DAI's
1dc2cef4cbaa ALSA: ctl: allow TLV read operation for callback type of element in locked case
````

dmesg and/or bisect would help, thanks

chrysophylax, 2020-02-27:
This only happens on 5.5.6-914 contra previous 5.5.5. It will just stay hung with a black screen until power-cycled by button no matter how much time elapses. Here' s the oops and [dmesg](https://gist.github.com/chrysophylax/10e4e4ec8b2254e8600a6ad9d6bf155e) 
```
Feb 27 21:49:40 tantalum kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
Feb 27 21:49:40 tantalum kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
Feb 27 21:49:40 tantalum kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Feb 27 21:49:40 tantalum kernel: BUG: unable to handle page fault for address: 0000138800000000
Feb 27 21:49:40 tantalum kernel: #PF: supervisor read access in kernel mode
Feb 27 21:49:40 tantalum kernel: #PF: error_code(0x0000) - not-present page
Feb 27 21:49:40 tantalum kernel: PGD 0 P4D 0 
Feb 27 21:49:40 tantalum kernel: Oops: 0000 [#1] SMP NOPTI
Feb 27 21:49:40 tantalum kernel: CPU: 3 PID: 188 Comm: kworker/3:1 Not tainted 5.5.6-914.native #1
Feb 27 21:49:40 tantalum kernel: Hardware name: LENOVO 81XA/LNVNB161216, BIOS CLCN27WW 11/20/2019
Feb 27 21:49:40 tantalum kernel: Workqueue: events sof_probe_work [snd_sof]
Feb 27 21:49:40 tantalum kernel: RIP: 0010:strlen+0x0/0x20
Feb 27 21:49:40 tantalum kernel: Code: 48 89 f9 74 09 48 83 c1 01 80 39 00 75 f7 31 d2 44 0f b6 04 16 44 88 04 11 48 83 c2 01 45 84 c0 75 ee 31 ff 41 89 f8 >
Feb 27 21:49:41 tantalum kernel: RSP: 0018:ffffad3bc06afcd0 EFLAGS: 00010206
Feb 27 21:49:41 tantalum kernel: RAX: ffff8cc9ac1039e8 RBX: ffff8cc97f37ba28 RCX: 0000000000000000
Feb 27 21:49:41 tantalum kernel: RDX: 0000000000000cc0 RSI: 0000138800000000 RDI: 0000138800000000
Feb 27 21:49:41 tantalum kernel: RBP: ffffad3bc06afcf8 R08: 0000000000000000 R09: 0000000000000000
Feb 27 21:49:41 tantalum kernel: R10: 0000000000000000 R11: 0000000000000000 R12: 0000138800000000
Feb 27 21:49:41 tantalum kernel: R13: ffffffffc0852ec0 R14: ffff8cc8f74f60b0 R15: 0000000000000cc0
Feb 27 21:49:41 tantalum kernel: FS:  0000000000000000(0000) GS:ffff8ccb4c4c0000(0000) knlGS:0000000000000000
Feb 27 21:49:41 tantalum kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 27 21:49:41 tantalum kernel: CR2: 0000138800000000 CR3: 000000015da0a004 CR4: 00000000003606e0
Feb 27 21:49:41 tantalum kernel: Call Trace:
Feb 27 21:49:41 tantalum kernel:  ? devm_kstrdup+0x22/0x80
Feb 27 21:49:41 tantalum kernel:  snd_soc_register_dai+0x70/0x190 [snd_soc_core]
Feb 27 21:49:41 tantalum kernel:  snd_soc_add_component+0xe4/0x2d0 [snd_soc_core]
Feb 27 21:49:41 tantalum kernel:  snd_soc_register_component+0x3e/0x60 [snd_soc_core]
Feb 27 21:49:41 tantalum kernel:  devm_snd_soc_register_component+0x54/0xa0 [snd_soc_core]
Feb 27 21:49:41 tantalum kernel:  sof_probe_work+0xf0/0x260 [snd_sof]
Feb 27 21:49:41 tantalum kernel:  ? __switch_to+0x386/0x420
Feb 27 21:49:41 tantalum kernel:  ? __switch_to_asm+0x34/0x70
Feb 27 21:49:41 tantalum kernel:  ? __schedule+0x29f/0x6f0
Feb 27 21:49:41 tantalum kernel:  ? push_to_pool+0x31/0xb0
Feb 27 21:49:41 tantalum kernel:  process_one_work+0x19b/0x370
Feb 27 21:49:41 tantalum kernel:  worker_thread+0x4b/0x3b0
Feb 27 21:49:41 tantalum kernel:  kthread+0x101/0x140
Feb 27 21:49:41 tantalum kernel:  ? process_one_work+0x370/0x370
Feb 27 21:49:41 tantalum kernel:  ? kthread_park+0xa0/0xa0
Feb 27 21:49:41 tantalum kernel:  ret_from_fork+0x1f/0x40
Feb 27 21:49:41 tantalum kernel: Modules linked in: snd_soc_hdac_hdmi snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_soc_dmic snd_sof_pc>
Feb 27 21:49:41 tantalum kernel:  int340x_thermal_zone pinctrl_cannonlake pinctrl_intel intel_hid ac intel_pmc_core sparse_keymap int3400_thermal acpi_therm>
Feb 27 21:49:41 tantalum kernel: CR2: 0000138800000000
Feb 27 21:49:41 tantalum kernel: ---[ end trace d2a01d09f5c80df2 ]---
Feb 27 21:49:41 tantalum kernel: probe of 0000:00:13.0 returned 1 after 633290 usecs
```

plbossart, 2020-02-27:
ACK, I can reproduce this error on 5.5.6, probably a bad merge happened, or missing patches in another series.

Thanks @chrysophylax for bringing this to our attention, we currently don't test -stable and that's a gap. @xiulipan @wenqingfu FYI.

plbossart, 2020-02-27:
Bisected to 
'ASoC: SOF: Intel: hda: Fix SKL dai count'

Gah, this is embarrassing. The patch that added the new DAI is missing in stable...
Anyways, we should really not use macros for array elements  but ARRAY_SIZE.

plbossart, 2020-02-27:
temporary work-around until we fix this with the -stable folks:

````
git cherry-pick ac42b142cd7652b3c9740cf81b8d8a3965e7a1e3
[detached HEAD 9bbb3e73cc5f] ASoC: SOF: Intel: hda: Add iDisp4 DAI
 Author: Sathyanarayana Nujella <sathyanarayana.nujella@intel.com>
 Date: Mon Dec 16 15:53:06 2019 -0800
 1 file changed, 4 insertions(+)
````

@chrysophylax do you have an email I can use for a Reported-by tag - assume you don't mind?

chrysophylax, 2020-02-28:
@plbossart billy@redpill-linpro.com dot com should be fine

tlauda, 2020-03-18:
@chrysophylax Issue to be closed?

tlauda, 2020-03-19:
Fixed in kernel.

ahkok, 2020-03-19:
@tlauda which exact kernel version has the fix? At least list the upstream commit ID, please.

tlauda, 2020-03-19:
@ahkok My mistake.