thesofproject/sof #2426: [FEATURE] Impulse Noise reduction as a post processing in DSP

sathyap-chrome, 2020-02-24:
**Is your feature request related to a problem? Please describe.**
on CML platform, we have seen a noise introduced by RT5682 ( in attachement). This noise is an impulsive and frequent than a 1 time noise.
  The idea of this feature is to see if we can identify and eliminate (or minimise) such noise

**Describe the solution you'd like**
  The solution is explained in the attachment. i would ask my colleague to even share the code snippet he tried on a matlab project on existing data ( not live).

Kindly review the document and solution and provide valuable suggestions on
a. if SOF has a solution available / under works for such a scenario
b. how can this solution be included as a process block in a pipeline for us to try in different topology
c. what could be effort on such an implementation.

Sample i/p and outputs as here:
[Impuse_Noise_Results.zip](https://github.com/thesofproject/sof/files/4243825/Impuse_Noise_Results.zip)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Screenshots are part of pdf attached.
[Audio_Headset_Noise_Cancellation.pdf](https://github.com/thesofproject/sof/files/4243751/Audio_Headset_Noise_Cancellation.pdf)

the original issue filed is here:
https://github.com/thesofproject/linux/issues/1604

lgirdwood, 2020-02-24:
@singalsu can you review and comment. Thanks

juimonen, 2020-02-24:
@sathyap-chrome couple of naive question: 
1) your original signal levels are quite low compared to peaks... so what happens if you have very high levels in the original signals? 
2) peaks seems to be really sharp (like only 1 sample long), so is a simple low pass causing audible artefacts?

a) Anyway we have general fir with possibility for arbitrary response, I don't think we have anything adaptive atm.
b) we have "process" component definition for effects in SOF (currently we have fir, iir etc). And the parameters can be set from toppology as binary blobs
c) effort is not too bad if you have working code ready (and you can't use existing fir or iir):
1) convert code to fixed point
2) define new process/effect class to driver, firmware & topology, you can see existing fir/iir for details

@singalsu can provide more comments and suggestions.

singalsu, 2020-02-24:
To me this looks like rather bad HW design (on CML). The codec power and audio bus needs to have proper filtering components resulting from signal integrity simulations. Has this error been forwarded to HW team?

singalsu, 2020-02-24:
The presentation doesn't look solid. The spectral leakage slide about time shift effect to DFT with boxcar window has nothing to do with described simple time domain procedure. Or maybe there's connection but I don't see such with just the slides. The human vocal tract also generates impulses and noises. Such proposed strong non-linear processing will degrade severely performance for speech recognition, the more the lower the sample rate is. I wanted to express this critique for the work before proceeding with concealing likely HW mistakes this way.

sathyap-chrome, 2020-02-25:
@singalsu Thanks for the feedback Seppo. We are working with codec and hw team to identify a solution. I will share more details as they are more clear. 

Idea is to see if we can find a sw wa as CML is almost ready for shipping. Note that the issue was first seen with 44.1 , but later is reproduced with 48k too (but a little hard to)

@juimonen Can you share some sample topology / snippet on how to add the process component ?
 i want to check if fir and iir can help here.

ShriramShastry, 2020-02-25:
@juimonen 
 1. _your original signal levels are quite low compared to peaks... so what happens if you have very high levels in the original signals?_
[Sriram] , Yes - as per slide number 7 (page-7) signal levels is low because it has only silence  and Channel-2 has impulse noise.  From subplot 3, we can see that Impulse noise is filtered out with silence left on the line. You can play "pOrigL_R.wav" and  "filter_R.wav" to notice difference.

In case of very heavy levels, slide 8 and 9 has audio and impulse noise. Post filtering , you can see only audio only i.e.  without impulse noises.

 2. _peaks seems to be really sharp (like only 1 sample long), so is a simple low pass causing audible artefacts?_
[Sriram ]:Yes, Peaks can be heard when audio is played off-line. Post filtering , audio signal has only silence ( no impulse noise)
 
 _a) Anyway we have general fir with possibility for arbitrary response, I don't think we have anything adaptive atm_.
 [Sriram] : What do you mean by anything adaptive atm ? Can you please share more information. 

 _b) we have "process" component definition for effects in SOF (currently we have fir, iir etc). And the parameters can be set from toppology as binary blobs_
 [Sriram] : It would be helpful if you can share a demo example.

 _c) effort is not too bad if you have working code ready (and you can't use existing fir or iir):_
 [Sriram] : Yes, I agree. I have converted the  Matlab simulation work into c - routines 
 _1. convert code to fixed point_
 [Sriram] : Done
 _2. define new process/effect class to driver, firmware & topology, you can see existing fir/iir for details_
 [Sriram] : Sure, I will go through existing iir/fir to get better understanding.

Please find the attachment for matlab code simulation with audio signal from the observed issue .

@singalsu : _The spectral leakage slide about time shift effect to DFT with boxcar window has nothing to do with described simple time domain procedure.
 Or maybe there's connection but I don't see such with just the slides._
[Sriram] : Sorry for the confusion. The picture was added under assumption section.Yes, it has nothing to do with example. 
Note : From slide number 4 to 10 is relevant to current issue. All plots are derived from the real data derived from issue description/attachment "Fail_5.wav"  (https://partnerissuetracker.corp.google.com/issues/145640941)

_The human vocal tract also generates impulses and noises. Such proposed strong non-linear processing will degrade severely performance for speech recognition
 Slide_
 [Sriram] : Impulse filtration would be helpful here as well. Implementation should be done very carefully to avoid any sort of degradation.

 
Thank you 

[get_audio.txt](https://github.com/thesofproject/sof/files/4248344/get_audio.txt)

singalsu, 2020-02-25:
@ShriramShastry Thanks for explanation and shared code!

We could try a brick-wall low-pass FIR as first test. I can prepare for you a topology to test with. Which SOF base topology are you using and which capture PCM is impacted by this issue?

What are the min/max capture sample rates you are using and what is your required audio bandwidth for the application? The FIR won't remove as much impulse noise as your procedure but on the other hand it won't degrade audio (except the band limiting) when the issue doesn't happen.

sathyap-chrome, 2020-02-25:
@singalsu the topology is [sof-cml-rt5682.m4](https://github.com/thesofproject/sof/blob/master/tools/topology/sof-cml-rt5682-kwd.m4) and 
the PCM is PCM0

singalsu, 2020-02-25:
Here's a quick test with FIR and IIR filters. The filters have cut-off at 7 kHz (assume speech bandwidth @ 48 kHz) and about 40 dB attenuation. Original waveform and spectra:

[image]

IIR filtered
[image]

FIR filtered
[image]

The problem with impulse noise is that it has all frequencies. These likely bit errors in audio bus are perfect impulses and equally strong in all frequencies. E.g. in FIR case the filtering for > 8 kHz is quite effective but the remaining part in the speech band is still pretty annoying.

[try_iir_fir.zip](https://github.com/thesofproject/sof/files/4249641/try_iir_fir.zip)

singalsu, 2020-02-25:
The enhanced sof-cml-rt5682 topology with FIR low-pass is now in [RFC] PR  #2434 . It should give similar result to simulation above. I don't have this device so I can't test.

plbossart, 2020-02-25:
@sathyap-chrome I have never seen a product compensate for a broken hardware link with a filter. That makes no sense at all. Fix the hardware!

sathyap-chrome, 2020-02-26:
@plbossart completely agree. we have been pushing for Hw fix only. this was a thought if for some reason, hw fix is not feasible ( currently few CML platforms are approaching production)

closed 2025-02-13 by singalsu