thesofproject/sof #2429: [BUG] [1.4.2] Profile not loading in Pulseaudio

danielcherubini, 2020-02-24:
**Describe the bug**
There's two bugs here I think, 
1. Pulseaudio won't load the profile `HiFi: Play HiFi quality Music (priority 8000, available: no)` I need to manually enable it and then Audio works fine
2. Restarting, or shutting down will result in a kernel panic, rolling back to 1.4.1 will resolve this issue.

This happens on Arch 5.4, 5.5, and 5.6 on a Lenovo C940

**To Reproduce**
It happens on every boot, and on every shutdown/restart

**Reproduction Rate**
100% of the time

**Expected behavior**
To not have to enable the profile after boot, and to be able to restart/shutdown

**Impact**
Bug 1 is an inconvenience, bug 2 is a showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: 1.4.2
2) Name of the topology file
    * Topology: n/a (whatever is installed with the basic package)
3) Name of the platform(s) on which the bug is observed.
    * Platform: Arch Linux

**Screenshots or console output**
```➜ pacmd list-cards
1 card(s) available.
    index: 0
	name: <alsa_card.pci-0000_00_1f.3-platform-skl_hda_dsp_generic>
	driver: <module-alsa-card.c>
	owner module: 6
	properties:
		alsa.card = "0"
		alsa.card_name = "sof-hda-dsp"
		alsa.long_card_name = "LENOVO-81Q9-LenovoYogaC940_14IIL-LNVNB161216"
		alsa.driver_name = "snd_soc_skl_hda_dsp"
		device.bus_path = "pci-0000:00:1f.3-platform-skl_hda_dsp_generic"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "34c8"
		device.product.name = "Smart Sound Technology Audio Controller"
		device.string = "0"
		device.description = "Smart Sound Technology Audio Controller"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	profiles:
		HiFi: Play HiFi quality Music (priority 8000, available: no)
		off: Off (priority 0, available: unknown)
	active profile: <HiFi>
	sinks:
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_5__sink/#2: Smart Sound Technology Audio Controller HDMI3/DP3 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_4__sink/#3: Smart Sound Technology Audio Controller HDMI2/DP2 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_3__sink/#4: Smart Sound Technology Audio Controller HDMI1/DP1 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp__sink/#5: Smart Sound Technology Audio Controller Speaker + Headphones
	sources:
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_5__sink.monitor/#2: Monitor of Smart Sound Technology Audio Controller HDMI3/DP3 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_4__sink.monitor/#3: Monitor of Smart Sound Technology Audio Controller HDMI2/DP2 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_3__sink.monitor/#4: Monitor of Smart Sound Technology Audio Controller HDMI1/DP1 Output
		alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp__sink.monitor/#5: Monitor of Smart Sound Technology Audio Controller Speaker + Headphones
		alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp__source/#6: Smart Sound Technology Audio Controller Headphones Stereo Microphone
		alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_6__source/#7: Smart Sound Technology Audio Controller Digital Microphone
	ports:
		[Out] HDMI3: HDMI3/DP3 Output (priority 700, latency offset 0 usec, available: no)
			properties:
				
		[Out] HDMI2: HDMI2/DP2 Output (priority 600, latency offset 0 usec, available: no)
			properties:
				
		[Out] HDMI1: HDMI1/DP1 Output (priority 500, latency offset 0 usec, available: no)
			properties:
				
		[Out] Speaker: Speaker (priority 100, latency offset 0 usec, available: unknown)
			properties:
				
		[Out] Headphones1: Headphones (priority 200, latency offset 0 usec, available: no)
			properties:
				
		[In] Headphones2: Headphones Stereo Microphone (priority 200, latency offset 0 usec, available: no)
			properties:
				
		[In] Mic: Digital Microphone (priority 100, latency offset 0 usec, available: no)
			properties:			
```

```➜ dmesg | grep audio                                
[    3.109667] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.109846] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.109860] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.110077] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    3.110085] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.110246] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.643428] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.647915] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.647917] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.741712] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    3.741714] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    3.816147] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    3.816149] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    3.816448] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    3.829743] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    3.829748] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.830644] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.830646] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.831448] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.831450] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.832232] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.832234] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.833051] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.833053] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.833054] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.833056] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.833057] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.833059] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.833827] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.833829] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.834731] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.834733] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.835572] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.843248] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    3.859602] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC298: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    3.859605] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.859606] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    3.859607] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    3.859608] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    3.859609] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x18
[    3.902811] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    3.902815] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[   11.352998] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   11.439831] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   21.180721] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   21.273898] sof-audio-pci 0000:00:1f.3: firmware boot complete```

danielcherubini, 2020-02-24:
One of the things i notice, is that the profile HiFi now says it's not available.. what are the criteria for that?

juimonen, 2020-02-25:
@danielcherubini the dmesg looks ok even though there are these "no reply expected" errors, they should not be fatal... can you paste here the oops dmesg you get with restart/shutdown (if you can get it to some file). If you can also find out the kernel version that would be nice...

If sof is crashing on boot you might not get alsa card at all, I think the profile could be unavailable if Pulseaudio can't open/probe the pcm properly (but now I guessing here also).

Duologic, 2020-02-28:
I think the kernel panic is as described in https://bugs.archlinux.org/task/65635
The panic happened on kernel 5.5.6, rolling back to 5.5.5 with sof-firmware 1.4.2 resulted in the 'unavailable' profile.

I didn't do more research and downgraded to kernel 5.5.2 with sof-firmware 1.4.1 (also alsa-lib 1.2.1.2) and the [CarbonX1 fix](https://gist.github.com/hamidzr/dd81e429dc86f4327ded7a2030e7d7d9) to get my working audio system back.

plbossart, 2020-02-28:
> I think the kernel panic is as described in https://bugs.archlinux.org/task/65635
> The panic happened on kernel 5.5.6, rolling back to 5.5.5 with sof-firmware 1.4.2 resulted in the 'unavailable' [...]

we provided a fix for 5.5-stable yesterday, see https://github.com/thesofproject/sof/issues/2418#issuecomment-592234791

danielcherubini, 2020-02-28:
fantastic thanks.

tlauda, 2020-03-10:
@danielcherubini Can we close this?

danielcherubini, 2020-03-10:
So I have reformatted and rebuilt the system from the ground up. The kernel panic is gone. But the  profile not being available is still there.