thesofproject/sof #2435: [BUG][CML][WOV] hw params ipc error when run arecord via wov pipeline

aiChaoSONG, 2020-02-26:
**Describe the bug**
if we run arecord on wov pipeline immediately after the first run(first run ended by CTRL+C or -d parameter), it is very easy to get "unable to install hw params". but if we wait for some time and then initiate the second run, everything's fine. once this error happens, wait for sometime(10s), we can still run arecord and trigger recording normally.

**To Reproduce**
- reboot device
- run `arecord -Dhw:0,8 -M -N  -r16000 -c2 -fs16_le -i -vvv -d 5 ./wov.wav` and make some sound trigger recording (first run)
- once first run ended, run `arecord -Dhw:0,8 -M -N  -r16000 -c2 -fs16_le -i -vvv -d 5 ./wov.wav` again immediately. (recommend to use UP ARROW to recall the last command), you should see the error.

**Reproduction Rate**
very easy to reproduce.

**Impact**
abnormal behavior on wov

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {8d47294d}
    * SOF: {a6726cf2}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML with rt1011 & rt5682 codec}

[dmesg_ipc_error_wov.log](https://github.com/thesofproject/sof/files/4253776/dmesg_ipc_error_wov.log)
[elogger_ipc_error.log](https://github.com/thesofproject/sof/files/4253777/elogger_ipc_error.log)
[tlogger_ipc_error.log](https://github.com/thesofproject/sof/files/4253778/tlogger_ipc_error.log)

keyonjie, 2020-02-26:
This could be the regression by merging PR#2228? @mrajwa @bkokoszx just got chance to review the PR source, the removing of the line "return PPL_STATUS_PATH_STOP;" making me nervous.

lgirdwood, 2020-02-26:
@aiChaoSONG teh pipeline and IPC code run far faster than 10s, can you try this again with smaller delays and let us know when the problem starts. This way will help us identify if its more FW or driver related.

bkokoszx, 2020-02-26:
@keyonjie @aiChaoSONG 
I've checked firmware before PR #2228 on commit: https://github.com/thesofproject/sof/tree/b206b7a8d63c5c940c3c2ad17d3f4030e3705810

I still have reproduction.

bkokoszx, 2020-02-26:
@keyonjie @aiChaoSONG 
I think the issue is caused by to small host buffer size. I cannot reproduce issue when I force biggest buffer size:
```arecord -Dhw:0,8 -M -N -r16000 -c2 -fs16_le -i -vvv -d 5 --buffer-size 67200 ./wov.wav```
Can you check?

lgirdwood, 2020-02-26:
@bkokoszx good find, perhaps we need to enforce the minimum host buffer size via topology (otherwise we can block on userspace freeing up buffer space as it reads)

aiChaoSONG, 2020-02-27:
@keyonjie @lgirdwood @bkokoszx Sorry for the misleading, we do need to specify a larger buffer-size. with a larger buffer-size, issue not reproduced.

closed 2020-02-27 by aiChaoSONG