thesofproject/sof #2438: [BUG][BUG][UCM]DMIC will capture empty sound after turn down volume to zero then turn it up.

ClarexZhou, 2020-02-26:
**Describe the bug**
System boot up or wake up from s3, then do arecord via DMIC(RT715). Turn down the DMIC capture volume  to zero then turn it up, DMIC will capture empty sound.

**To Reproduce** 
1. Boot up system 
2. Terminal 1: 'arecord -Dhw:0,4 -f dat 1.wav -vv -i'
3. Terminal 2: 'alsamixer', then turn down the PGA5.0 5 volume till zero then turn it up.

**Reproduced rate**
4/5
 
**Expected result**
DMIC capture normally.

**Actual result**
DMIC capture empty sound after turn up input volume from zero.

**Environment**
kernel: https://github.com/thesofproject/linux/tree/release/sdw_cml_hotfix commit: 38d0dcd5
FW: https://github.com/thesofproject/sof/commits/cml-010-drop-stable commit: 5e5f69e 
TPLG: Same with FW branch+PR#2431, sof-cml-rt711-rt1308-mono-rt715.tplg rename to sof-cml-rt711-rt1308-rt715.tplg 
Platform: CML Laptop with ALC711+ ALC1308+ ALC715 in SDW mode

No error message in dmesg.
**error sof logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA          17794084.322917  17794084.000000      src/lib/agent.c:55  	validate(), ll drift detected, delta = 25509
    0      1       VOLUME 5.19     30862011.354167  13067927.000000 udio/volume/volume.c:285 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 5.19     30862868.020833       856.666687 udio/volume/volume.c:285 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 5.19     30862878.489583        10.468750 udio/volume/volume.c:285 	volume_set_chan: Limited request 0 to VOL_MIN.
    0      1       VOLUME 5.19     32648222.760417   1785344.250000 udio/volume/volume.c:285 	volume_set_chan: Limited request 0 to VOL_MIN.
```
[dmesg.txt](https://github.com/thesofproject/sof/files/4255204/dmesg.txt)
[errorSofLogger.txt](https://github.com/thesofproject/sof/files/4255206/errorSofLogger.txt) 
[fullSofLogger.txt](https://github.com/thesofproject/sof/files/4255207/fullSofLogger.txt)

ClarexZhou, 2020-02-26:
Error not seen when using tplg without PR #2431, and same cml-010-drop-stable commit: 5e5f69e FW. 
Error not seen when using tplg with PR #2431 , but using Master branch FW, commit a6726cf2.

libinyang, 2020-02-26:
The PR add 2 "switch" controls in the tplg, one is for headset MIC, the other is for DMIC. User can use the "switch" controls to mute the capture pipeline (the volume pipeline with 'switch' support). The driver need the switch control because the driver need the 'switch' controls to support MIC LED.

juimonen, 2020-02-26:
so we have couple of patches in master that will probably affect this:

related PR's
https://github.com/thesofproject/sof/pull/2196
62dbfcff5cd20f0de688d817c8a070ad94e4c3c5

https://github.com/thesofproject/sof/pull/2109
e525c177b5dda8fc4c19b422026fa5f90d913522
901a963a678cbb1b3eb6e53ab29d80f82cd488d2
dcf722f826a405e4ae87b0d3ad87df1f5f0eeeb5
9d4ca10639a637905bdf76a0cb3677bcaf775f71

so we should try out these on top of the 010-drop-stable

juimonen, 2020-02-26:
https://github.com/thesofproject/sof/pull/2441
@ClarexZhou can you try if above PR helps?

lgirdwood, 2020-02-26:
@ClarexZhou does this bug occur in master ?

lgirdwood, 2020-02-26:
@singalsu any comments ?

ClarexZhou, 2020-02-27:
> @ClarexZhou does this bug occur in master ?

Error not occurred in master

ClarexZhou, 2020-02-27:
> #2441
> @ClarexZhou can you try if above PR helps?
Can't apply this patch on cml-010-drop-stable directly for merge failed.

ClarexZhou, 2020-02-27:
Error also can be reproduced via rt711 headset capture.

juimonen, 2020-02-27:
@ClarexZhou can you just try the PR 2441 itself? it is on top of 010-drop-stable already...

libinyang, 2020-02-27:
agree with Jaska, especially for 62dbfcf. The description is very similar with our issue.

ClarexZhou, 2020-02-27:
Error not seen with PR #2441  on  cml-010-drop-stable

ClarexZhou, 2020-02-28:
Error is fixed on cml-010-hot-fix branch. Close

closed 2020-02-28 by ClarexZhou