thesofproject/sof #2445: [TGL][BUG][SDW]" Unable to install hw params:" display when captured via arecord0,4(dmic).

sinahuang, 2020-02-27:
**Describe the bug**
" Unable to install hw params:" display when captured via arecord0,4(dmic).

**To Reproduce** 
1. Boot up system..
2.  Arecord -Dhw:0,4 -r16000 -c4 -fs16_le 00.wav -vv -i

**Reproduced rate**
100%
 
**Expected result**
Arecord should work successfully.

**Actual result**
Dmesg shows "error: ipc error for 0x60010000 size 20" and arecord cannot work.

```
ubuntu@ubuntu-NUC7i7BNH:~$ arecord -Dhw:0,4 -r16000 -c4 -fs16_le 00.wav -vv -i 
Recording WAVE '00.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Channels 4
arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 64
CHANNELS: 4
RATE: 16000
PERIOD_TIME: 124500
PERIOD_SIZE: 1992
PERIOD_BYTES: 15936
PERIODS: (4 5)
BUFFER_TIME: 499500
BUFFER_SIZE: 7992
BUFFER_BYTES: 63936
TICK_TIME: 0
ubuntu@ubuntu-NUC7i7BNH:~$ 
```

```
[  118.168393] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  118.168395] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e40
[  118.168396] sof-audio-pci 0000:00:1f.3: periods:4
[  118.168425] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  118.168430] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  118.168705] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  118.168708] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  118.168709] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  118.168711]  DMIC16kHz: ASoC: hw_params FE failed -22
[  118.168714] sof-audio-pci 0000:00:1f.3: pcm: free stream 4 dir 1
[  118.168842] sof-audio-pci 0000:00:1f.3: pcm: close stream 4 dir 1
[  118.666865] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG

```
**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:8d47294d
FW:  https://github.com/thesofproject/sof/commits/master commit: a6726cf
Tplg: Same with FW branch, sof-tgl-rt711-i2s-rt1308.tplg 
Platform:TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode

[dmesg.log](https://github.com/thesofproject/linux/files/4253341/dmesg.log)
[sof_trace.log](https://github.com/thesofproject/linux/files/4253342/sof_trace.log)

lgirdwood, 2020-02-27:
@sinahuang @ClarexZhou I cant see any issue in the FW log. Can you run again and extra the FW trace for the failed operation. I would expect to see an error message in trace. Also please attache error trace log.

sinahuang, 2020-02-27:
FW bisect result：
Regression caused by kernel commit：https://github.com/thesofproject/sof/commit/de73e60a8c627debeb93f040d7aa3d1cfc2b04aa
@bkokoszx please check it.

lgirdwood, 2020-02-27:
@bkokoszx fyi - bisected to https://github.com/thesofproject/sof/commit/de73e60a8c627debeb93f040d7aa3d1cfc2b04aa

bkokoszx, 2020-03-02:
@sinahuang @lgirdwood 
With @emilchudzik we have tried to set up TGL platform (we have been using https://github.com/thesofproject/kconfig/blob/master/kconfig-sof-default.sh), but we have some error in kernel.
[image]
Do you have any suspicion what might be a root cause?

@sinahuang 
Maybe could you provide kernel binary?

sinahuang, 2020-03-03:
Issue also can be reproduced on TGL-Y with codec ALC711 and  RT1308 in SDW mode.

Attached the kernel binary :http://sof-ci-storage.sh.intel.com/daily/20200303/linux-65f9d39d-1/linux-image-5.6.0-rc3-daily-default-20200303_5.6.0-rc3-topic-sof-dev-65f9d39d-1_amd64.deb

emilchudzik, 2020-03-03:
@plbossart We have enabled SDW and DMIC in BIOS.
Kernel binary:
http://sof-ci-storage.sh.intel.com/daily/20200302/linux-30887e1f-2/linux-image-5.6.0-rc3-daily-default-20200302_5.6.0-rc3-topic-sof-dev-30887e1f-2_amd64.deb

Full dmesg below:
[dmesg.txt](https://github.com/thesofproject/sof/files/4280148/dmesg.txt)

slawblauciak, 2020-03-03:
Just to make sure, https://github.com/thesofproject/sof/pull/2430 has been merged recently, is this still reproducible?

sinahuang, 2020-03-03:
Yes，issue still can be reproduced on the latest FW.

emilchudzik, 2020-03-03:
@plbossart could you explain us one question. Even if the platform support SDW or not, I have disabled support SDW in kernel config and disable SDW in BIOS. Why kernel still want to check SDW machine driver? It blocks further loading.
We'd like to disable SDW to omit issue with SDW machine driver and focus on DMIC and this bug.
Is it possible?

Kconfig that we tried today:
https://github.com/thesofproject/kconfig/blob/master/kconfig-sof-default.sh
without: hdaudio-codecs-defconfig and soundwire-defconfig

On our windows python tests SDW works.

plbossart, 2020-03-03:
> Kconfig that we tried today:
> https://github.com/thesofproject/kconfig/blob/master/kconfig-sof-default.sh
> without: hdaudio-codecs-defconfig and soundwire-defconfig

well then it's not the default...
if there is no SoundWire Kconfig we fall back to HDaudio. 
You must do something wrong, sorry.

emilchudzik, 2020-03-04:
You didn't read carefully our problem with SDW/HDA on TGL.
I ask you:
Why during loading linux, kernel still want to check SDW machine driver, even if I disable SDW support in the konfig. The same for HDA...

That's why I change default kconfig.
Because we want to debug DMIC issue ftom topic. -> without problems with SDW and HDA.
There is still problem:
[image]

lgirdwood, 2020-03-04:
@emilchudzik as a sanity check, can you delete your current SOF modules ```/lib/modules/version/kernel/sound/soc*``` and re-install. Sometimes old modules with previous kconfig options can persist in the modules dir and create issues. If this does not cure the issue can you raise this as a bug on the kernel, as it seems the Kconfig options from HDA/SDW are being ignored.

ClarexZhou, 2020-03-04:
This issue is only reproduced on TGL with codec ALC711 in SDW mode. DMIC capture works ok on TGL with codec ALC711 in HDA mode.

plbossart, 2020-03-04:
@emilchudzik I stand by my comment that if SoundWire is not enabled, you should not see any messages related to SoundWire.

Look at hda.c

````c
#if IS_ENABLED(CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE)
/* Check if all Slaves defined on the link can be found */
static bool link_slaves_found(struct snd_sof_dev *sdev,
			      const struct snd_soc_acpi_link_adr *link,
			      struct sdw_intel_ctx *sdw)
{
...
}

static int hda_sdw_machine_select(struct snd_sof_dev *sdev)
{
	...
		} else {
			dev_info(sdev->dev,
				 "No SoundWire machine driver found\n");
		}
	}

	return 0;
}
#else
static int hda_sdw_machine_select(struct snd_sof_dev *sdev)
{
	return 0;
}
#endif
````

If SoundWire was truly not supported in the kernel, you would land on the second fallback.

emilchudzik, 2020-03-05:
This is config part when I disable HDA and SDW:

```
CONFIG_SND_SOC_SOF_HDA_COMMON=m
# CONFIG_SND_SOC_SOF_HDA_LINK is not set
# CONFIG_SND_SOC_SOF_HDA_PROBES is not set
# CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1 is not set
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
# CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE_LINK is not set
CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE_LINK_BASELINE=m
```

This is what I found in linux repo:
[image]

As you see there is mismatch in naming, because in config there is added **_LINK**,

BDW this condition is always entered in first fallback
#if IS_ENABLED(CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE)
even if I disable or enable SDW support in: make menuconfig.
So I don't understand why :)

Doesn't matter...
As Clare wrote this bug reproduce with SDW enabled, so we haven't such setup working...

bkokoszx, 2020-03-06:
@lgirdwood @sinahuang 
Going back to the main thread, thanks for @ClarexZhou  for sharing remote TGL. 

In topology file ```sof-tgl-rt711-i2s-rt1308.m4``` we have ```dmic16k``` configured in stereo mode:
```
DAI_CONFIG(DMIC, 1, 4, dmic16k,
	   DMIC_CONFIG(1, 500000, 4800000, 40, 60, 16000,
		DMIC_WORD_LENGTH(s16le), 400, DMIC, 1,
		PDM_CONFIG(DMIC, 1, STEREO_PDM0)))
```
, so is it all right that we want here to record 4 channels? 
```arecord -Dhw:0,4 -r16000 -c4 -fs16_le 00.wav -vv -i```

If we want to record 2 channels we should make little change in topology file: 
``` diff
 PIPELINE_PCM_ADD(sof/pipe-volume-capture-16khz.m4,
-       5, 4, 4, s16le,
+       5, 4, 2, s16le,
        1000, 0, 0,
        16000, 16000, 16000)
```
and invoke ```arecord -Dhw:0,4 -r16000 -c2 -fs16_le 00.wav -vv -i``` (```arecord``` with ```-c2``` flag).

On the other hand, if we want to record 4 channels here, we should make following topology change:
``` diff
 DAI_CONFIG(DMIC, 1, 4, dmic16k,
           DMIC_CONFIG(1, 500000, 4800000, 40, 60, 16000,
                DMIC_WORD_LENGTH(s16le), 400, DMIC, 1,
-               PDM_CONFIG(DMIC, 1, STEREO_PDM0)))
+               PDM_CONFIG(DMIC, 1, FOUR_CH_PDM0_PDM1)))

 ifelse(HDMI, `1',
 `
```

I can make PR for this, but I do not know whether we want record 2 or 4 channels?

lgirdwood, 2020-03-06:
@bkokoszx this should be fixed in the multi DAI config (when its merged). I would keep it 4 chans on the DMIC, userspace alsa-plugins) can always convert to stereo if needed.

plbossart, 2020-03-06:
> This is config part when I disable HDA and SDW:
>
> ```
> CONFIG_SND_SOC_SOF_HDA_COMMON=m
> # CONFIG_SND_SOC_SOF_HDA_LINK is not set
> # CONFIG_SND_SOC_SOF_HDA_PROBES is not set
> # [...]

No, it's by design.

All platforms capable of enabling SoundWire select  SND_SOC_SOF_INTEL_SOUNDWIRE_LINK_BASELINE

there in menuconfig  you have a user visible switch

config SND_SOC_SOF_INTEL_SOUNDWIRE_LINK
	bool "SOF support fo

and that switch controls whether or not the code enables soundwire.

config SND_SOC_SOF_INTEL_SOUNDWIRE_LINK_BASELINE
	tristate
	select SND_SOC_SOF_INTEL_SOUNDWIRE if SND_SOC_SOF_INTEL_SOUNDWIRE_LINK
	help
	  This option is not user-selectable but automagically handled by
	  'select' statements at a higher level

So the code is correct to check for SND_SOC_SOF_INTEL_SOUNDWIRE

> BDW this condition is always entered in first fallback
> #if IS_ENABLED(CONFIG_SND_SOC_SOF_INTEL_SOUNDWIRE)
> even if I disable or enable SDW support in: make menuconfig.
> So I don't understand why :)

You probably have an error somewhere, the rules in Kconfig don't back your conclusions.
Also always look for the generated .config. not your initial one.

bkokoszx, 2020-03-08:
@sinahuang 
Can you check whether PR#2506 fixes this issue?

sinahuang, 2020-03-09:
Check this issue with PR#2506 on TGL-U RVP, issue can't be reproduced. PR#2506 seems can fix this issue.

bkokoszx, 2020-04-08:
@sinahuang @ClarexZhou 
I have to update PR #2506. Can you check whether it can be reproduce with updated PR.

Above PR generates to separate topology files: ```sof-tgl-rt711-i2s-rt1308-2ch.tplg``` and 
```sof-tgl-rt711-i2s-rt1308-4ch.tplg```.

If you want to run ```arecord -c4```, you should use ```sof-tgl-rt711-i2s-rt1308-4ch.tplg```.
If you want to run ```arecord -c2```, you should use ```sof-tgl-rt711-i2s-rt1308-2ch.tplg```.

sinahuang, 2020-04-08:
Issue can't be reproduced with PR https://github.com/thesofproject/sof/pull/2506 and  PR generates topology files.
if run arecord -c4 with  sof-tgl-rt711-i2s-rt1308-4ch.tplg, arecord can work normally.
 if run arecord -c2 with sof-tgl-rt711-i2s-rt1308-2ch.tplg, arecord can work normally.

PR #2506 (2020-04-08): tgl: use additional CHANNEL define in rt711 + rts1308 tplg
 This commit adds CHANNELS define in sof-tgl-rt711-i2s-rt1308.m4
 topology in order to properly set pipeline pcm channels and dmic
 configuration (STEREO_PDM0 or FOUR_CH_PDM0_PDM1 mode).

Fixes #2445.
closed 2020-04-08 by lgirdwood