thesofproject/sof #2452: [BUG][TGL]Then volume is small and have much noise when arecord via dmic using format s24_le

Liviali155, 2020-02-27:
**Describe the bug**
Then volume is small and have much noise when arecord via dmic (0,6) or (0,7) using format fs24_le compared with using format fs32_le/fs16_le

**To Reproduce**
1.sudo reboot
2.keep the volume of  arecord not change
3.arecord -Dhw:0,6  -r48000 -c4 -fs24_le   fs24_le.wav -vv -i
  arecord -Dhw:0,6  -r48000 -c4 -fs32_le   fs32_le.wav -vv -i
  arecord -Dhw:0,6  -r48000 -c4 -fs16_le   fs16_le.wav -vv -i
4. separately paplay   fs24_le.wav /fs32_le.wav/  fs16_le.wav

[image]

**Reproduction Rate**
10/10

**Expected behavior**
The volume of fs24_le.wav normally and have no noise

**Actual result**
The volume of fs24_le.wav is very small  and have much  noise compared  with /fs32_le.wav/  fs16_le.wav

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware: 247af13 https://github.com/thesofproject/sof/commits/master
Kernel: 6af9a26f https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

[dmesgvolume.log](https://github.com/thesofproject/linux/files/4230675/dmesgvolume.log)
[sof_tracevolume.log](https://github.com/thesofproject/linux/files/4230676/sof_tracevolume.log)
[fs16_le.zip](https://github.com/thesofproject/linux/files/4230682/fs16_le.zip)
[fs24_le.zip](https://github.com/thesofproject/linux/files/4230684/fs24_le.zip)
[fs32_le.zip](https://github.com/thesofproject/linux/files/4230685/fs32_le.zip)

singalsu, 2020-02-27:
@xiulipan Thanks for doing the raw format check! It looks like format converter issue indeed.

@Liviali155 Is the issue similar with gcc and xt-xcc built firmware? The format converter uses a different optimized code version when built with xt-xcc.

wwittbrx, 2020-02-27:
I have taken the fs24_le.zip from the original post and converted the file with the tool we use in slim driver tests. The result looks ok to me, the volume is ok and I don't hear more noise than in fs16 or fs32. I have attached the original and converted file. So to me this is not an issue and should be closed.

[fs24_le.zip](https://github.com/thesofproject/sof/files/4261395/fs24_le.zip)

singalsu, 2020-02-27:
I tried raw conversion with sox and gave correct looking result too. The output can be could be wav as well. Flac might be more robust format.

`sox --rate 48000 --bits 32 --channels 2 --encoding signed-integer fs24.raw fs24.flac vol 256`

Also confirm that the captured S24 raw was OK.

singalsu, 2020-02-27:
For @xiulipan 's recording:

`sox --rate 48000 --bits 32 --channels 4 --encoding signed-integer s24_le.raw s24_le.flac vol 256`

Though in about 0.45s - 0.75s time there's some "noise". Maybe it's unrelated?

kv2019i, 2020-02-27:
@singalsu wrote:
> `sox --rate 48000 --bits 32 --channels 2 --encoding signed-integer fs24.raw fs24.flac vol 256`

Or competing solution:
apt-get install ecasound
ecasound -f:s32_le,2,48000 -i fs24.raw -f:s16_le,2,48000 -o fs24-out.wav -ea:25600 -ev

You'll also get some simple amplitude statistics printed out when conversion is ready.

This uses same trick as Seppo with sox. Very few apps support ALSA's S24_LE directly. It's primarily intended to communicate with audio hardware, not for direct usage by applications (most applications will use S32_LE/S32_BE).

singalsu, 2020-02-27:
Nitpick, this doesn't lose precision:

`ecasound -f:s32_le,4,48000 -i s24_le.raw -f:s32_le,4,48000 -o fs24-out.wav -ea:25600 -ev -x`

(I typed and Kai instructed :^)

Liviali155, 2020-02-28:
I used format_converter.exe to convert fs24_le.wav,then used  Audacity to check,the wave is normal,closed it
[image]

closed 2020-02-28 by Liviali155

xiulipan, 2020-02-28:
@singalsu @wwittbrx @kv2019i Thanks for checking the issue, it seems the issus is caused by the s24_3le support is rare in real world.