thesofproject/sof #2465: [BUG][TGL]"loading /lib/firmware/intel/sof/sof-tgl.ri failed with error -22" display when system boot up successfully.

sinahuang, 2020-02-28:
**Describe the bug**
Can't load "sof-tgl.ri "files normally after system boot up successfully. Aplay -l display “nosoundcard”. 

**To Reproduce**
1.Run "./inst.sh 20200228-tgl"
2.Run "ll /lib/firmware/intel/sof*" to check the FW install normally.
3.Run "aplay -l" 

**Reproduction Rate**
5/5

**Expected behavior**
System should work normally.

```
[    3.454001] sof-audio-pci 0000:00:1f.3: HDA codec #2 probed OK: response: 80862812
[    3.454540] hdaudio ehdaudio0D2: loading codec module: hdaudio:v80862812r00100000a01
               
[    3.457255] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
[    3.457269] sof-audio-pci 0000:00:1f.3: loading firmware
[    3.459590] sof-audio-pci 0000:00:1f.3: loading /lib/firmware/intel/sof/sof-tgl.ri failed with error -22
[    3.459593] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-tgl.ri failed with error -22
[    3.459596] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-tgl.ri failed err: -22
[    3.459642] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -22
[    3.459844] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    3.459854] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    3.459859] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    3.460185] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -22
```
**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:0ce0ab9
FW:  https://github.com/thesofproject/sof/commits/master commit: 37fc1b6
Tplg: Same with FW branch, sof-tgl-rt711-i2s-rt1308.tplg 
Platform:TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode

Can't cat the sof trace.
[dmesg28.log](https://github.com/thesofproject/sof/files/4265378/dmesg28.log)

sinahuang, 2020-02-28:
FW bisect result：
Regression caused by FW commit：https://github.com/thesofproject/sof/commit/109450e6fdaa9308732e0358dc10680e673f7d58
@singalsu please help to check it.

juimonen, 2020-02-28:
@sinahuang the commit you pointed out above is changes to testbench script? I don't see how it could affect boot or anything?

sinahuang, 2020-02-28:
Closed it, for my wrong operation.

closed 2020-02-28 by sinahuang

singalsu, 2020-02-28:
Thanks, that would have been a weird one!